bdring/FluidNC #1192: No movement

Timacaro, 2024-04-24:
### Wiki Search Terms

Good evening,
I uploaded everything I need onto the ESP and its working all fine. Uploaded the 6_pack_2130_XYZABC.yaml and that is also now loaded. Still when trying to get 1 motor moving it does not happen. 

### Controller Board

6-pack

### Machine Description

I need to have 4 Y motors at the same time, 1 X and 1 Z motor to move. 

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack tmc_2130 XYZABC
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.3
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0
        spi_index: -1
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.250
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.6
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
        spi_index: -1
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.250
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.11
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
        spi_index: -1
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.250
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        
  a:
    steps_per_mm: 53.400
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.14
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15
        spi_index: -1
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.250
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false

  b:
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.2
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.19
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16
        spi_index: -1
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.250
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        
  c:      
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.25
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.22
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23
        spi_index: -1
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.250
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.36
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: i2so.24
  mist_pin: i2so.27
  delay_ms: 0

probe:
  pin: gpio.39
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

PWM:
  pwm_hz: 5000
  output_pin: gpio.14
  enable_pin: gpio.13
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.7.16 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:WARN: Cannot open configuration file:config.yaml]]
[MSG:INFO: Using default configuration]]
[MSG:INFO: Axes: using defaults]]
[MSG:INFO: Machine Default (Test Drive)]]
[MSG:INFO: Board None]]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Axis X (-1000.000,0.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO: Axis Y (-1000.000,0.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO: Axis Z (-1000.000,0.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO: Kinematic system: Cartesian]]
[MSG:INFO: Using spindle NoSpindle]]
[MSG:INFO: Connecting to STA SSID:Arizona]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
[MSG:INFO: Connecting...]]
[MSG:INFO: Connected - IP is 192.168.2.33]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]]
[MSG:INFO: SSDP Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]
```

### User Interface Software

WebUI

### What happened?

I try to understand how things are working with the pins and the 2130 SPI I have placed on the board. I did not change anything from that standard yaml file.

bdring, 2024-04-24:
Mitch just wrote a new wiki page with a flow chart to help with that. Give it a try.

http://wiki.fluidnc.com/en/support/setup/motion

MitchBradley, 2024-04-24:
```
[MSG:WARN: Cannot open configuration file:config.yaml]]
```
That means that FluidNC cannot find your config file.  If you uploaded it to a name other than "config.yaml", you must tell FluidNC the name by sending "$config/filename=YOUR_CONFIG_NAME.yaml" in FluidTerm, or else set it via WebUI 
[image]

MitchBradley, 2024-04-24:
> I need to have 4 Y motors at the same time, 

FluidNC does not directly support 4 motors on one axis.  The most is 2 motors per axis.  It might be possible to connect 2 motors to one driver and 2 motors to another driver.

Timacaro, 2024-04-25:
Good evening Mitch,
Thank you for your advice. I got something moving, but goes with a lot of noise and sometimes back and forward. It looks as if there is not enough power or maybe too much. The motor (got one connected at the moment) does not get warm. Where should I start with this.
Thank you for your time.

MitchBradley, 2024-04-25:
http://wiki.fluidnc.com/en/support/setup/motion

Timacaro, 2024-04-29:
Good evening,
Spend most of the weekend figuring out what is causing the motor not to move. It goes, slow, sometimes in different directions and makes a lot of noise. I went through the motor guide and still I have something that is not correct and maybe I do not understand everything correctly and being a Dutch guy that could be the case. 
Got a yaml file from Github. Got TMC2130 and motors that are 1.5 ampere, 24 volts, 2.3 ohm, 1,8 deg. In any case I tried to change the current on the stepper, which is now 1.5 v also tried to lock the motors but I can't get that to work only with using the $ME command. What am I missing? Can you help me please? Thank you in advance.

board: 6 Pack
name: 6 Pack tmc_2130
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 200
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      tmc_2130:
        cs_pin: i2so.3
        step_pin: i2so.2
        direction_pin: i2so.1
        disable_pin: I2SO.0
        spi_index: -1
        r_sense_ohms: 0.11
        run_amps: 0.75
        hold_amps: 0.25
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 100
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 200
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.32
      limit_pos_pin: NO_PIN
[... 196 lines cut ...]
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
coolant:
  flood_pin: i2so.24
  mist_pin: i2so.27
  delay_ms: 0
probe:
  pin: gpio.39
  check_mode_start: true
  toolsetter_pin: NO_PIN
macros: {}
start:
  must_home: false
PWM:
  pwm_hz: 5000
  output_pin: gpio.14
  enable_pin: gpio.13
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%

MitchBradley, 2024-04-29:
> It goes, slow, sometimes in different directions and makes a lot of noise.

The usual cause of "slow, different directions, and noisy" is a bad connection between the stepper driver and the stepper motor.  A stepper motor has two coils.  Motion happens when coils A and B are energized in sequence - A B A B A B etc. - with a particular timing pattern (phase) and polarity.  If there is a broken connection to one of the coils, the motor sees the pattern A _ A _ A _ instead.  Instead of moving smoothly, the motor will sometimes move slowly with jerking, sometimes move in one direction for awhile then switch directions, and sometimes just sit there shaking.

The other things that can cause this behavior, in addition to a broken wire or bad connection, are
* The motor is bad - one of the coils inside has failed.  That can be tested with a multimeter.  The resistance across coil A should be the same as the resistance across coil B - typically a few ohms.
* The stepper driver has failed and can no longer energize one of the coils.

bdring, 2024-04-29:
http://wiki.fluidnc.com/en/support/faq#motor-moves-erratically

Timacaro, 2024-04-29:
I measured the motor 2.5 ohms on both channels so that should be ok. Even put the two wires together and the motor was hard to move.  Changed to a different driver, problem is the same. 6Pack is in SPI mode, card on 5 V, TMC2130 on 1.5 volts. When I send a movement command I measure about 21 volts. Still the motor makes a loud noise and goes back and forward. Do you have any other idea?

bdring, 2024-04-29:
What do you have the Vcc set for? 

https://github.com/bdring/6-Pack_CNC_Controller/wiki/Stepper-Jumper-Motor-Setup#stepper-jumper-motor-setup

Timacaro, 2024-04-29:
On 5v

Timacaro, 2024-04-29:
And I just read it should be 3.3v.

Timacaro, 2024-04-29:
Not there is not movement at all.

MitchBradley, 2024-04-29:
You could try a different motor, or swapping that motor to a different stepper.

bdring, 2024-04-29:
It should be 3.3v. The TMC2130 can run at either voltage, but SPI returns data to the ESP32 so it must be 3.3v or it can break the ESP32.

It usually does not hurt in the short term, but definately leave it at 3.3v.

Can you show us your startup messages?

Timacaro, 2024-04-29:
Do you mean the yaml file or something else?

board: 6 Pack
name: 6 Pack tmc_2130
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 200
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      tmc_2130:
        cs_pin: i2so.3
        step_pin: i2so.2
        direction_pin: i2so.1
        disable_pin: I2SO.0
        spi_index: -1
        r_sense_ohms: 0.11
        run_amps: 0.75
        hold_amps: 0.25
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 100
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 200
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.32
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
[... 194 lines cut ...]
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
coolant:
  flood_pin: i2so.24
  mist_pin: i2so.27
  delay_ms: 0
probe:
  pin: gpio.39
  check_mode_start: true
  toolsetter_pin: NO_PIN
macros: {}
start:
  must_home: false
PWM:
  pwm_hz: 5000
  output_pin: gpio.14
  enable_pin: gpio.13
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%

Timacaro, 2024-04-29:
> You could try a different motor, or swapping that motor to a different stepper.

I did all of that already, no change.

MitchBradley, 2024-04-29:
Maybe you are trying to move faster than the motors can go.  I don't see any details above about what commands you are using to perform the motion, nor any information about your mechanical system.

Timacaro, 2024-04-29:
I am only sending commands through the web interface, using CNCj for example doesn't work either. I connected 1 X motor on X  to the 6PACK and that's it.

bdring, 2024-04-29:
Here is how to check your startup messages.

http://wiki.fluidnc.com/en/support/troubleshooting_config_files#check-the-startup-messages

Timacaro, 2024-04-29:
Ah you mean is this. Ok that is a new chapter then for me and I will attend to that tomorrow. Thank you for you help today. Time for me to go to bed.

[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.7.17 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:HOMECNC.yaml]
[MSG:INFO: Machine 6 Pack tmc_2130]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 6]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.2 Dir:I2SO.1 CS:I2SO.3 Disable:I2SO.0 Index:-1 R:0.110]
[MSG:INFO:  X Neg Limit gpio.33]
[MSG:INFO: Axis Y (50.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.5 Dir:I2SO.4 CS:I2SO.6 Disable:I2SO.7 Index:-1 R:0.110]
[MSG:INFO:  Y Neg Limit gpio.32]
[MSG:INFO: Axis Z (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.10 Dir:I2SO.9 CS:I2SO.11 Disable:I2SO.8 Index:-1 R:0.110]
[MSG:INFO:  Z Neg Limit gpio.35]
[MSG:INFO: Axis A (150.000,1110.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.13 Dir:I2SO.12 CS:I2SO.14 Disable:I2SO.15 Index:-1 R:0.110]
[MSG:INFO:  A Neg Limit gpio.34]
[MSG:INFO: Axis B (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.18 Dir:I2SO.17 CS:I2SO.19 Disable:I2SO.16 Index:-1 R:0.110]
[MSG:INFO:  B Neg Limit gpio.2]
[MSG:INFO: Axis C (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.21 Dir:I2SO.20 CS:I2SO.22 Disable:I2SO.23 Index:-1 R:0.110]
[MSG:INFO:  C Neg Limit gpio.25]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: A Axis driver test passed]
[MSG:INFO: B Axis driver test passed]
[MSG:INFO: C Axis driver test passed]
[MSG:INFO: feed_hold_pin gpio.36]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.13 Out:gpio.14 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant I2SO.24]
[MSG:INFO: Mist coolant I2SO.27]
[MSG:INFO: Probe gpio.39]
[MSG:INFO: Connecting to STA SSID:Arizona]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.2.33]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

MitchBradley, 2024-04-29:
> I am only sending commands through the web interface, using CNCj for example doesn't work either. I connected 1 X motor on X to the 6PACK and that's it.

If I were to tell you that I spoke to someone over the telephone, you would not know what I said.  Similarly, knowing that you sent commands through the Web interface and CNCjs doesn't help me know what the commands were.

Timacaro, 2024-04-30:
Good evening,
You are totally right. This is for example what I have send to the 6PACK 
$J=G91 G21 F1000 X10
ok
<Jog|MPos:12.945,0.000,0.000,0.000,0.000,0.000|FS:1000,0>

Still no movement by the way after I have changed to 3.3 volts. I reset, the motor will lock for a few seconds.

bdring, 2024-04-30:
Try sending $MD and $ME commands. Check the lock state after each command.

Timacaro, 2024-04-30:
No it does not lock.

bdring, 2024-04-30:
The each coil mounts to adjacent terminals.

[image]

Timacaro, 2024-04-30:
Ok, so I reconnected it to the picture above. It all stays the same, locks when I reboot for a few seconds and then you can move the stepper again. Do you have any other ideas?

bdring, 2024-04-30:
$ME does not lock the motors?

Timacaro, 2024-04-30:
No

bdring, 2024-04-30:
I am running out of ideas.

Try changing ` idle_ms: 250` in your config to 255

Timacaro, 2024-04-30:
Changed, no result. $ME has no effect. It still makes me think why the motor unlocks after reboot. Could that be a thing?

bdring, 2024-04-30:
You could try adding `:low` after `disable_pin: I2SO.0`

I doubt it will work, but it is the only thing I can think of.

Timacaro, 2024-04-30:
Unfortunately no. Ok thank you for your help with this. I am done and going to find another card for my project. Again thank you for your efforts.

Timacaro, 2024-04-30:
I think I found it, I let you know tomorrow if I am right. Hardware issue, but not the card.

MitchBradley, 2024-04-30:
Sometimes if it locks for a few seconds, then unlocks, it is because the driver overheated and shut down to protect itself.  I have also seen cases where the power supply shut down or glitched, causing the driver to shut down.

closed 2024-12-12 by bdring