bdring/FluidNC #1193: Unable to connect to Wifi

hms-11, 2024-04-25:
### Wiki Search Terms

Wifi Problem

### Controller Board

ESP32 Custom Board
Custom Board based on the 6 pack controller with I2S outputs and octocoupler inputs. 6 Axis plus other features. Full schematics can be found here:
https://oshwlab.com/coreyearl1985/esp32-external-cnc-driver-4-axis

### Machine Description

No machine yet, issue on initial setup

### Configuration file

```yaml
No config file, issue is on initial setup of a new controller.
```

### Startup Messages

```text
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.7.16 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:WARN: Cannot open configuration file:config.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID: HotchsIndoorWifi]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: No SSID]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.16 (wifi) '$' for help]
ok
[VER:3.7 FluidNC v3.7.16:]
[OPT:PHS]
[MSG:Machine: Default (Test Drive)]
[MSG:Mode=STA:SSID=:Status=Not connected:IP=0.0.0.0:MAC=0C-B8-15-E3-22-5C]
[MSG:]
[MSG:Mode=AP:SSID=FluidNC:IP=192.168.0.1:MAC=0C-B8-15-E3-22-5D]
ok
```

### User Interface Software

WebUI as well as FluidNC Web Installer

### What happened?

After successful initial successful install through FluidNC Web Installer wifi setup fails. I can successfully enter the SSID and Password through the terminal interface however the as can be seen in the startup message posted the controller fails to connect to the SSID. 
The password does not contain any special characters, just upper and lower case letters and numbers. 
I have successfully connected to the same network using a simple Arduino sketch for connecting to Wifi and the controller connects and gets an IP address. 
I have attempted to connect to multiple wifi networks with the same error.
When using the web installer terminal, I the controller can see the available networks and shows high signal strength when using the $Wifi/ListAPs command.

bdring, 2024-04-25:
Send `$List/APS` to show all access points and signal strengths

Send `$STA/SSID` to show the STA SSID you are requesting.

Paste the responses here, so we can compare them.

hms-11, 2024-04-25:
Thanks for the quick reply.
Here is $Wifi/ListAPS : 
Grbl 3.7 [FluidNC v3.7.16 (wifi) '$' for help]
$Wifi/ListAPS
{"AP_LIST":[
    "AP_LIST":[
      {"SSID":"HotchsIndoorWifi",
        "SIGNAL":"100",
        "IS_PROTECTED":"1"
      },
      {"SSID":"hotchs",
        "SIGNAL":"72",
        "IS_PROTECTED":"1"
      },
      {"SSID":"HotchsYardWifi",
        "SIGNAL":"52",
        "IS_PROTECTED":"1"
      },
      {"SSID":"HotchsIndoorWifi",
        "SIGNAL":"24",
        "IS_PROTECTED":"1"
      }
    ]
  }
  
ok

Here is $STA/SSID : 
$STA/SSID
$Sta/SSID= HotchsIndoorWifi
ok

hms-11, 2024-04-25:
I've tried multiple AP's with strengths from ~80 - 100 with no change. This same controller will connect to the same AP if using the following simple Arduino sketch:
```
#include<WiFi.h>

const char *ssid = "YourSSID";  
const char *password = "YourPassword";
        
void initWiFi() {
      WiFi.mode(WIFI_STA);
      WiFi.begin(ssid, password);
      Serial.print("Connecting to WiFi ..");
      while (WiFi.status() != WL_CONNECTED) {
        Serial.print('.');
        delay(1000);
      }
      Serial.println(WiFi.localIP());
    }
    
    void setup() {
      Serial.begin(115200);
      initWiFi();
      Serial.print("RRSI: ");
      Serial.println(WiFi.RSSI());
    }
    
    void loop() {
      // put your main code here, to run repeatedly:
    }
```

hms-11, 2024-04-25:
> You appear to have everything correct.
> 
> Have you tried the other APs with higher strength as a test?
> 
> It could be the controller is adding a little noise that makes it hard to connect.

I just noticed it listed the same access point twice, it's a mesh network. 
I've also tried the 'hotchs' AP to make sure it wasn't grabbing the low strength 'HotchsIndoorWifi' AP with the same result.
I've also tried it on another network with only a single AP with the same result.

MitchBradley, 2024-04-25:
Maybe the security settings are incompatible.  There is a configuration variable $Sta/MinSecurity with the following values

        "OPEN"
        "WEP"
        "WPA-PSK"
        "WPA2-PSK"         (default)
        "WPA-WPA2-PSK"
        "WPA2-ENTERPRISE"

hms-11, 2024-04-25:
Good thinking but unfortunately no cigar. 
I just checked and the AP is WPA/WPA2 - Personal so I switched MinSecurity to WPA-WPA2-PSK and no change.

hms-11, 2024-04-25:
I'll give that a shot. Might be a couple days before I get a chance to mess around with it and recompile but I'll update the issue with what I find. Thanks for the suggestion!

hms-11, 2024-04-25:
One more detail I forgot to add to the initial issue, I'll try the suggestion mentioned by @MitchBradley but I wanted to include this.
If I try and use the FluidNC AP and go through the setup procedure, when I try and search for Wifi Access Points to connect to I get "Failed: 406 Wrong data"
I've attached a screen shot. 
I've got a couple more of these boards, I'll try and upload to another one just to ensure this particular board doesn't have some sort of weird hardware issues and I will also try a generic ESP32 dev board as well to isolate the issue to my custom board. 
[image]

MitchBradley, 2024-04-25:
Re 406: https://github.com/bdring/FluidNC/issues/1168

hms-11, 2024-04-26:
Quick update.
I haven't had a chance to go through WiFiConfig::StartSTA() in detail to attempt to fix there but I had a couple minutes last night to play around with a couple other boards to test. 
I tested another of my custom boards, same issue. 
I also tested just a plain ESP32 DevKit and it also has the same issue, so whatever the problem is, it does not appear to be related to the face that I have a custom board in use.

hms-11, 2024-04-26:
Ok this is embarrassing, but I've solved the issue and the issue is actually that I'm a dummy.
I was inputting my wifi credentials like this
$Sta/SSID = HotchsIndoorWifi 
or
$Sta/SSID= HotchsIndoorWifi
Same with the password. I was adding spaces from the command to the inputting of data.
I just entered it through FluidTerm as 
$Sta/SSID=HotchsIndoorWifi 
I did the same with the password, no spaces. 
Connected successfully. 
So no actual issue, just a user inputting data incorrectly.