bdring/FluidNC #1196: Abnormal operating speed

DmiZar, 2024-04-29:
### Wiki Search Terms

 Jogging run at a very slow speed.

### Controller Board

MKS DLC-32

### Machine Description

XYZ 3-axis router machine

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: CNC3018
meta: (31.07.2023) by DmiZar

stepping: 
  engine: I2S_static
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 12
  
uart1: 
  txd_pin: gpio.25
  rxd_pin: gpio.33
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1

uart2: 
  txd_pin: gpio.0
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 57600
  mode: 8N1

uart_channel1: 
  uart_num: 2
# report_interval_ms: 200
# all_messages: false
  

i2so: 
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi: 
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard: 
  cs_pin: gpio.15
  card_detect_pin: gpio.39
  frequency_hz: 8000000
  
kinematics: 
  Cartesian: 

axes: 
  shared_stepper_disable_pin: I2SO.0
  shared_stepper_reset_pin: NO_PIN
  x: 
    steps_per_mm: 1600.000000
    max_rate_mm_per_min: 3000.000000
    acceleration_mm_per_sec2: 200.000000
    max_travel_mm: 268.000000
    soft_limits: true
    homing: 
      cycle: 2
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 2000.000000
      settle_ms: 100
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.36
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.000000
      tmc_2209:
        uart_num: 1
        addr: 0
        cs_pin: NO_PIN
        r_sense_ohms: 0.110
        run_amps: 1.300
        hold_amps: 0.500
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: NO_PIN

  y: 
    steps_per_mm: 1600.000000
    max_rate_mm_per_min: 3000.000000
    acceleration_mm_per_sec2: 200.000000
    max_travel_mm: 178.000000
    soft_limits: true
    homing: 
      cycle: 2
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 2000.000000
      settle_ms: 100
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.500000
      tmc_2209:
        uart_num: 1
        addr: 1
        cs_pin: NO_PIN
        r_sense_ohms: 0.110
        run_amps: 1.300
        hold_amps: 0.500
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: I2SO.5
        direction_pin: I2SO.6:low
        disable_pin: NO_PIN

  z: 
    steps_per_mm: 1600.000000
    max_rate_mm_per_min: 2000.000000
    acceleration_mm_per_sec2: 200.000000
    max_travel_mm: 90.000000
    soft_limits: true
    homing: 
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 2000.000000
      settle_ms: 100
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.000000
      tmc_2209:
        uart_num: 1
        addr: 2
        cs_pin: NO_PIN
        r_sense_ohms: 0.110
        run_amps: 1.300
        hold_amps: 0.500
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: I2SO.3
        direction_pin: I2SO.4
        disable_pin: NO_PIN

control: 
  safety_door_pin: gpio.23
  reset_pin: NO_PIN
  feed_hold_pin: gpio.18
  cycle_start_pin: gpio.19
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN

coolant: 
  flood_pin: gpio.5:low
  mist_pin: NO_PIN
  delay_ms: 0

probe: 
  pin: gpio.22
  toolsetter_pin: NO_PIN
  check_mode_start: true

macros: 
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 
  after_homing: 
  after_reset: 
  after_unlock: 

start: 
  must_home: true
  deactivate_parking: false
  check_limits: false

parking: 
  enable: true
  axis: Z
  target_mpos_mm: 0.000000
  rate_mm_per_min: 2000.000000
  pullout_distance_mm: 5.000000
  pullout_rate_mm_per_min: 500.000000

user_outputs: 
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000

#The output is on this pin. It is controlled via M62, M63, M64 and M65 commands 
#Beeper connector. Command:
#"M62 P0" turn beeper On.  M64 P0 - немедленно
#"M63 P0" turn beeper Off. M65 P0 - немедленно

#"G4 P0.5" wait up to 0.5 seconds

#M64 P0
#G4 P0.5
#M65 P0

  digital0_pin: I2SO.7 
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN

PWM: 
#Send M6 T0 to use the PWM spindle
  pwm_hz: 1000
  direction_pin: NO_PIN
  output_pin: gpio.32
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 2000
  spindown_ms: 2000
  tool_num: 0
  speed_map: 0=0.000% 0=30.000% 10000=100.000%
  off_on_alarm: true

Laser:
#Send M6 T100 to use the laser
#M3 is constant power and M4 is dynamic power mode (scales linearly with speed during accel/decel)
  pwm_hz: 1000
  output_pin: gpio.26  
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 100
  speed_map: 0=0.000% 255=100.000%
  off_on_alarm: true

arc_tolerance_mm: 0.002000
junction_deviation_mm: 0.010000
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
lanner_blocks: 16
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.7.17 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:INFO: Configuration file:My_MKS_DLC32_21_2209.yaml]]
[MSG:INFO: Machine CNC3018]]
[MSG:INFO: Board MKS-DLC32 V2.1]]
[MSG:INFO: UART1 Tx:gpio.25 Rx:gpio.33 RTS:NO_PIN Baud:115200]]
[MSG:INFO: UART2 Tx:gpio.0 Rx:gpio.4 RTS:NO_PIN Baud:57600]]
[MSG:INFO: uart_channel2 created]]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:gpio.39 freq:8000000]]
[MSG:INFO: Axis Y (-178.000,0.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     tmc_2209 UART1 Addr:1 CS:NO_PIN Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN R:0.110]]
[MSG:INFO:  Y Pos Limit gpio.35]]
[MSG:INFO: Axis Z (-90.000,0.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     tmc_2209 UART1 Addr:2 CS:NO_PIN Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN R:0.110]]
[MSG:INFO:  Z Pos Limit gpio.34]]
[MSG:INFO: X Axis driver test passed]]
[MSG:INFO: Y Axis driver test passed]]
[MSG:INFO: Z Axis driver test passed]]
[MSG:INFO: safety_door_pin gpio.23]]
[MSG:INFO: feed_hold_pin gpio.18]]
[MSG:INFO: cycle_start_pin gpio.19]]
[MSG:INFO: Kinematic system: Cartesian]]
[MSG:INFO: Using spindle PWM]]
[MSG:INFO: Flood coolant gpio.5:low]]
[MSG:INFO: Probe gpio.22]]
[MSG:INFO: Connecting to STA SSID:Keenetic-7527]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
[MSG:INFO: Connected - IP is 192.168.2.120]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]]
[MSG:INFO: SSDP Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]
ok
```

### User Interface Software

UGS, Candle

### What happened?

I repeated the problem No. 1122 "Abnormal operating speed" when the override feeds is not set to 100%. Jogging run at a very slow speed, the feed value is displayed as "nan". The sender of the G-code does not matter, I sent commands manually, and also used Candle and UGS.

### Other Information

<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:**128**,100,100|A:F>
<Jog|MPos:-0.118,0.000,0.000|FS:**nan**,0|WCO:0.000,0.000,0.000

bdring, 2024-04-29:
Can you manually show a simple way to recreate the problem in a console. I can't make the problem happen. Can you get the nan to happen without a jog command. The problem may occur before the jog command.

I don't think jogs are currently intended to be affected by overrides.

DmiZar, 2024-04-30:
Yes. I'll try to do this.

I supply power to the machine.
Grbl 3.7 [FluidNC v3.7.17 (wifi) '$' for help]
ok

?
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:F>

$H
[MSG:Homed:Z]
[MSG:Homed:XY]
ok

?
$J=G21G91X-10.000Y-10.000Z0.000F1000
ok

?
<Idle|MPos:-10.000,-10.000,0.000|FS:0.0|Ov:100,100,100|A:F>

All ok.

Next, in UGS I set the override to 120%.
?
<Idle|MPos:-10.000,-10.000,0.000|FS:0.0|Ov:**120**,100,100|A:F>

And repeat:
<Idle|MPos:-10.000,-10.0> $J=G21G91X-10.000Y-10.000Z0.000F1000
ok

?
<Jog|MPos:-10.317,-10.317,0.000|FS:**nan**,0>
ok

nan and a very low speed of movement.

bdring, 2024-05-01:
I can't repeat the problem with your example.

DmiZar, 2024-05-02:
The problem is very easy to repeat.
I took a new ESP32 Devkit V1 board and loaded it with firmware 3.7.17.
Then I uploaded the file fluidnc-config-files/official/MKS_DLC32_21_XYZ.yaml.
Launched WebUI.
Set the override to 110%.
Started a jog:
$J=G91 G21 F1000 X10
Controller response:
<Jog|MPos:10.003,0.000,0.000|FS:nan,0|Ov:110,100,100>.
As you can see, nan is available.

[image]

DmiZar, 2024-05-03:
Thank you for resolving this issue. I'll compile the firmware and test how it works.

MitchBradley, 2024-05-03:
You don't have to compile it yourself.  The Continuous Integration action that tests all Pull Requests now leaves a binary that you can download.  Go to https://github.com/bdring/FluidNC/actions/runs/8930764528 and scroll down to the bottom.

DmiZar, 2024-05-03:
I'm understood, thank you.

MitchBradley, 2024-05-04:
Did you test it?

DmiZar, 2024-05-05:
Yes, I checked the engraver's operation in different modes. Everything is working correctly.
Thank you, you have a great project!

commit 17271d30 (2024-05-05): Fix #1196 - abnormal operating speed (#1204)

The problem is that, if you issue a feedrate override when
there is no motion, it calls the planner recalculate code
which then incorrectly updates some planner blocks that are
not in use with bogus data.  Then when you issue a jog, it
uses the bogus data.

I think this is the correct fix but it hasn't been tested with
a lot of different motion situations.  It certainly fixes the
problem as stated, and seems to work for simple non-jog G1
commands.
closed 2024-05-05 by MitchBradley