bdring/FluidNC #1199: Problem: Installing the FluidNC on an esp32-s3

MiguelASTavares, 2024-04-30:
### Wiki Search Terms

esp32-s3

### Controller Board

6-pack

### Machine Description

n

### Configuration file

```yaml
n
```

### Startup Messages

```text
esptool.py --chip auto --baud 230400 dump_mem 0x3ff5a018 4 SecurityFuses.bin
esptool.py v3.1
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-S3(beta3)
Chip is ESP32-S3(beta3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 00:00:00:00:00:00
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 230400

A fatal error occurred: Invalid head of packet (0x69)
esptool failed
```

### User Interface Software

install scripts 

### What happened?

I was trying to install the FluidNC onto a esp32-s3 using the install scripts present in the release, however that gave me the error presented above. After that, I also try to compile the PlatformIO version both the [env:wifi_s3] and [env:noradio_s3]. In the build, it gave the following error:
 ```
In file included from /home/pm/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/hal/include/hal/uart_hal.h:30,
                 from FluidNC/src/Uart.cpp:12:
/home/pm/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/hal/esp32s3/include/hal/uart_ll.h: In function 'uint32_t uart_ll_get_baudrate(uart_dev_t*)':
/home/pm/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/hal/esp32s3/include/hal/uart_ll.h:181:37: error: no matching function for call to 'uart_clkdiv_reg_t(volatile uart_clkdiv_reg_t&)'
     uart_clkdiv_reg_t div_reg = hw->clkdiv;
```

Is there anything that I'm doing wrong? Is the support for the s3 still not completed?

bdring, 2024-04-30:
We do not support S3 yet.

MitchBradley, 2024-04-30:
Anything we support ends up being expensive for us because the number of people who ask questions far exceeds the number of people who contribute to the project.  Supporting S3 would double our release engineering and testing time, and would mean that we have to have more hardware for testing.  We are already overwhelmed with support questions, for example from people who blew up their ESP32 and asked a lot of questions, only to realize that they plugged it in backwards.

MiguelASTavares, 2024-05-06:
I have also tried to install the FluidNC on an ESP32-C3. The installer runs, but once it's in the console state it crashes with the error:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004d1f8
invalid header: 0xffffffff
invalid header: 0xffffffff
```

I also tried to install on a ESP32-S2, but the process stops with the secure boot msg. Do you have an alternative way to install in a ESP32-S2 with the secure boot active?

SoupySoups, 2024-05-12:
@MiguelASTavares have a look at the  [wip s3 branch](https://github.com/bdring/FluidNC/tree/S3_WIP)