bdring/FluidNC #1201: Error when re-initializing TMC 2209 stepper motors from version 3.7.15.

DmiZar, 2024-05-01:
### Wiki Search Terms

TMC2209

### Controller Board

MKS DLC-32

### Machine Description

CNC3018 3-axis engraver with TMС 2209 drivers connected according to a standard scheme to a free UART

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: CNC3018
meta: (31.07.2023) by DmiZar

stepping: 
  engine: I2S_static
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 12
  
uart1: 
  txd_pin: gpio.25
  rxd_pin: gpio.33
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1

uart2: 
  txd_pin: gpio.0
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 57600
  mode: 8N1

uart_channel1: 
  uart_num: 2
# report_interval_ms: 200
# all_messages: false
  

i2so: 
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi: 
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard: 
  cs_pin: gpio.15
  card_detect_pin: gpio.39
  frequency_hz: 8000000
  
kinematics: 
  Cartesian: 

axes: 
  shared_stepper_disable_pin: I2SO.0
  shared_stepper_reset_pin: NO_PIN
[... 297 lines cut ...]
$I
[VER:3.7 FluidNC 3.7.13:]
[OPT:PHS]
[MSG:Machine: CNC3018]
[MSG:Mode=STA:SSID=Keenetic-7527:Status=Connected:IP=192.168.2.120:MAC=0C-DC-7E-55-C9-FC]
ok
$MI
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
ok

Starting from version 3.7.15, reinitialization of stepper motors ocсurs with an error:
$I
[VER:3.7 FluidNC v3.7.15:]
[OPT:PHS]
[MSG:Machine: CNC3018]
[MSG:Mode=STA:SSID=Keenetic-7527:Status=Connected:IP=192.168.2.120:MAC=0C-DC-7E-55-C9-FC]
ok
$MI
[MSG:INFO: X Axis driver test passed]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x21 got 0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x21 got 0x0]
ok
On versions 3.7.16 and 3.7.17 the same thing, after which the stepper motors do not move and the controller needs to be restarted.

bdring, 2024-05-01:
It might be the `use_enable:` Try turning that off. Can you use the disable pin?

DmiZar, 2024-05-01:
I inverted the `use_enable:` value by setting it to `false`. Reinitialization completes without errors.
Thank you!