bdring/FluidNC #1205: Problem: Probing coordinates seem messed up

nicolas-lang, 2024-05-03:
### Wiki Search Terms

probing

### Controller Board

6-pack

### Machine Description

Gantry CNC XYYZ with external DM556T drivers, X and Y are belt driven, Z is a trapezoidal spindle. The router is a relais switched Kress.

### Input Circuits

```yaml
Touchplate on 6-pack 5v Opto input module. touchplate is 5v, tool is gnd.
Touchplate is a cooper plated FR4
```

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack Ext-DMT-556T XYYZ
meta: by Nicolas Lang
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 10
  dir_delay_us: 0
  disable_delay_us: 0

start:
  must_home: false
  deactivate_parking: false
  check_limits: true

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 66.423
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 844.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.25:low
      limit_all_pin: NO_PIN
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2:low
        direction_pin: I2SO.1:low
        disable_pin: I2SO.0:low

  y:
    steps_per_mm: 66.423
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 75.000
    max_travel_mm: 1025.000
    soft_limits: true
    homing:
      cycle: 2
[... 337 lines cut ...]
<Run|MPos:0.000,0.000,-4.212|FS:915,0>
<Run|MPos:0.000,0.000,-13.357|FS:1250,0>
<Run|MPos:0.000,0.000,-23.793|FS:1250,0>
<Run|MPos:0.000,0.000,-33.787|FS:806,0>
<Run|MPos:0.000,0.000,-38.748|FS:56,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,-39.000|FS:0,0>
G38.6 Z-40 F100 P2
[GC:G38.2 G54 G17 G21 G90 G94 M5 M9 T0 F100 S0]
<Run|MPos:0.000,0.000,-39.000|FS:100,0>
<Run|MPos:0.000,0.000,-39.783|FS:100,0>
<Run|MPos:0.000,0.000,-40.615|FS:100,0|WCO:0.000,0.000,-44.556>
<Run|MPos:0.000,0.000,-41.448|FS:100,0>
<Run|MPos:0.000,0.000,-41.499|FS:100,0|Pn:P>
<Run|MPos:0.000,0.000,-41.513|FS:85,0>
<Run|MPos:0.000,0.000,-41.529|FS:70,0|Pn:P>
[PRB:0.000,0.000,-41.497:1]
$J=G90 G21 F1000 Z3
[MSG:INFO: Probe offset applied:]
ok
<Idle|MPos:0.000,0.000,-41.733|FS:0,0|Pn:P|WCO:0.000,0.000,-43.497>
ok
<Jog|MPos:0.000,0.000,-41.732|FS:180,0|Pn:P|Ov:100,100,100>
<Jog|MPos:0.000,0.000,-41.368|FS:247,0>
<Idle|MPos:0.000,0.000,-40.496|FS:0,0>
```

nicolas-lang, 2024-05-09:
In summary my expectation is: 
- for a given physical location i would expect static machine coordinates, with the probing only affecting the current work coordinates.

This is not the case with my machine, so there must be some kind of issue.

Possible candidates for the issue:
- The touchplace has some kind of delayed electrical triggering (capacity, contact resistance, ground loop with toolhead etc.)
I will investigate this by trining to use a tool length sensor instead of a touchplate

- Software Issue
not sure how to debug this further, and right now i mainly would like to have confirmed if my expectation regading the designed behaviour of probing  is correct (see first sentence)

MitchBradley, 2024-05-09:
Try with v3.7.15 .  There was a change in the probing code since then that could possibly have affected the accuracy.

MitchBradley, 2024-05-09:
I did some tests on my laser gantry with various versions.  I don't see any probe consistency problems with any version, including the latest 3.7.17.

I changed my config file so that the x limit switch is the probe pin, and probed in the -x direction, using $j between probes to pull away.  In all cases, the probe position was repeatable within 1 step.  My x steps_per_mm is 80 so 1 step is 0.0125 mm.

```
ok
g38.6 x-20 f100 p2
[PRB:0.188,0.000,0.000:1]
[MSG:INFO: Probe offset applied:]
ok
<Idle|MPos:0.013,0.000,0.000|FS:0,0|Pn:P|WCO:445.475,197.825,0.000>
$j=g91 g21 x3 f1000
ok
<Idle|MPos:3.013,0.000,0.000|FS:0,0|Pn:P>
g38.6 x-20 f100 p2 
[PRB:0.188,0.000,0.000:1]
[MSG:INFO: Probe offset applied:]
ok
$j=g91 g21 x10 f1000
ok
g38.6 x-20 f100 p2  
[PRB:0.200,0.000,0.000:1]
[MSG:INFO: Probe offset applied:]
ok
<Idle|MPos:0.025,0.000,0.000|FS:0,0|Pn:P|WCO:445.488,197.825,0.000>
$j=g91 g21 x10 f1000
ok
g38.6 x-20 f100 p2  
[PRB:0.200,0.000,0.000:1]
[MSG:INFO: Probe offset applied:]
ok
<Idle|MPos:0.013,0.000,0.000|FS:0,0|Pn:P|WCO:445.488,197.825,0.000>
$j=g91 g21 x17 f1000
ok
g38.6 x-20 f100 p2  
[PRB:0.200,0.000,0.000:1]
[MSG:INFO: Probe offset applied:]
ok
```

MitchBradley, 2024-05-09:
Perhaps you could check for mechanical problems.  One way to do that would be to move back and forth, ending at the same position, and use a dial indicator or some other precise measuring tool to check the repeatability.

nicolas-lang, 2024-05-09:
Hi, thanks for your comments it's much apreciated.
Especially your confirmation that the software side seems to work correctly removes a lot of variables from the testing... 

I am quite sure to have no mechnical issue, as the same hardware has been working reliably with the previous controller (grbl with some arduino based shield) for 3+ years

Right now the main thing I'll be testing will probably be the electrical setup. Previously I was using a 24V touch plate circuit with optocoupler and a external pull-up. My new setup uses only one of the 5V Inputs of the optocoupler shield of the 6 pack, maybe this suffers from noise or capacity issues.... 

I'll do some more testing tomorrow and will update or close this issue afterwards.

Thanks a Lot, Nicolas

MitchBradley, 2024-05-09:
Okay, but yesterday I had a mechanical problem with a machine that has been working correctly for 25 years.  Turns out that a dovetail saddle that holds the spindle on the Z axis had slipped by 9 mm.  "Used to work" is evidence but not a proof.

MitchBradley, 2024-05-09:
You could also try I2SO_STATIC mode.  It is supposed to switch to that for probing but maybe something is wrong.

nicolas-lang, 2024-05-26:
I am mostly sure it was a strange effect that was caused by the 5V IO shield in combination with a badly designed touch plate (a double sided unetched circuit board blank)  that acted like a plate-capacitior delaying the probe activation causing the drift.

I haven't had much time to confirm/test but it seems like adding pullup + transistor and switching to 24V for the touch plate resolved the issue...

closed 2024-05-26 by nicolas-lang