bdring/FluidNC #1206: Problem: TMC driver not detected after reinitialize, only one axis driver test passed at the beginning.

HiroKwok, 2024-05-03:
### Wiki Search Terms

https://github.com/bdring/FluidNC/issues/1201
https://github.com/bdring/FluidNC/issues/1008
https://github.com/bdring/FluidNC/issues/1001

### Controller Board

[image]
[image]

### Machine Description

CorXY Laser Cutting Machine , TMC2209 × 3 

### Configuration file

```yaml
board: KIDIN-NC MAIN BOARD V1
name: KIDIN-NC LASER CUTTING MACHINE
meta: "update for 3.7.1"

stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0

kinematics:
  corexy:
  
uart1:
  txd_pin: gpio.16
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1
  
user_outputs:
  digital0_pin: gpio.15
  
axes:
  shared_stepper_disable_pin: gpio.2:high
  
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 4000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: true

    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
      feed_mm_per_min: 400.000
      seek_mm_per_min: 2400.000
      settle_ms: 200
    
    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 5.00
      tmc_2209:
        uart_num: 1
        addr: 3
        r_sense_ohms: 0.110
        run_amps: 0.500
        hold_amps: 0.050
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: gpio.14
        direction_pin: gpio.27:low:pu
        disable_pin: NO_PIN
    motor1:
      null_motor:

  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 4000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 200
    soft_limits: true

    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: true
      feed_mm_per_min: 400.000
      seek_mm_per_min: 2400.000
      settle_ms: 200
    
    motor0:
      limit_neg_pin: gpio.39:low
      hard_limits: false
      pulloff_mm: 5.00
      tmc_2209:
        uart_num: 1
        addr: 0
        r_sense_ohms: 0.110
        run_amps: 0.500
        hold_amps: 0.250
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: gpio.26
        direction_pin: gpio.25:low
        disable_pin: NO_PIN
    motor1:
      null_motor:
      
  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 4000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 30

    homing:
      cycle: 0
      mpos_mm: 0
      positive_direction: true
      feed_mm_per_min: 400.000
      seek_mm_per_min: 600.000
      settle_ms: 200
    
    motor0:
      limit_neg_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 4.00
      tmc_2209:
        uart_num: 1
        addr: 2
        r_sense_ohms: 0.110
        run_amps: 0.500
        hold_amps: 0.050
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: gpio.33
        direction_pin: gpio.32
        disable_pin: NO_PIN

    motor1:
      null_motor:
 
      
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: gpio.35
  
probe:
  pin: gpio.13:low
  toolsetter_pin: NO_PIN
  check_mode_start: false

start:
  must_home: false

Laser:
  pwm_hz: 5000
  output_pin: gpio.17:low:pu
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4

[MSG:INFO: FluidNC v3.7.4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:KD_NC_MAIN.yaml]
[MSG:INFO: Using default speed map]
[MSG:INFO: Machine KIDIN-NC LASER CUTTING MACHINE]
[MSG:INFO: Board KIDIN-NC MAIN BOARD V1]
[MSG:INFO: UART1 Tx:gpio.16 Rx:gpio.4 RTS:NO_PIN Baud:115200]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.35 freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: User Digital Output:0 on Pin:gpio.15]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.2]
[MSG:INFO: Axis X (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 UART1 Addr:3 Step:gpio.14 Dir:gpio.27:low:pu Disable:NO_PIN R:0.110]
[MSG:INFO:  X Neg Limit gpio.36:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-200.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 UART1 Addr:0 Step:gpio.26 Dir:gpio.25:low Disable:NO_PIN R:0.110]
[MSG:INFO:  Y Neg Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-30.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 UART1 Addr:2 Step:gpio.33 Dir:gpio.32 Disable:NO_PIN R:0.110]
[MSG:INFO:  Z Neg Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO: X Axis driver test passed]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x21 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x21 got 0x0x0]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.17:low:pu Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Probe Pin: gpio.13:low]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.4 (wifi) '$' for help]
$MI
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x21 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x21 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x21 got 0x0x0]
ok
```

### What happened?

I am designing a laser cutting machine based on ESP32.  I met a problem , steppers (XYZ) drive by TMC2209,  but everytime only one axis can be detected at the beginning. If I plug in all the TMC2209 modules any axis can not be detected （Axis TMC driver not detected - expected 0x0x21 got 0x0x0）.Even have one axis can be detected at the first time, after reinitialize ($MI) all the axis can not be detected.

If I plug in one TMC2209 module, only one axis can be detected (at the first init). More then two TMC2209 modules plug in all the axis can not be detected.

bdring, 2024-05-03:
Show a photo of the controller. How do you set the address on the drivers?

HiroKwok, 2024-05-03:
[image]

I try to change the resistance between Urat1 TX and RX ，from original 1kΩ to 330 Ω， it seem more than one aixs can be detected now （even $MI）. Yesterday  I burned a TMC2209 module due to an operating error. So I can't test it completely today.I will test it again when the new module arrives.

HiroKwok, 2024-05-03:
> Show a photo of the controller. How do you set the address on the drivers?

0d285e970fc850917478bad4ccb39b4
I try to change the resistance between Urat1 TX and RX ，from original 1kΩ to 330 Ω， it seem more than one aixs can be detected now （even $MI）. Yesterday I burned a TMC2209 module due to an operating error. So I can't test it completely today.I will test it again when the new module arrives.

HiroKwok, 2024-05-03:
> Show a photo of the controller. How do you set the address on the drivers?

[image]

I try to change the resistance between Urat1 TX and RX ，from original 1kΩ to 330 Ω， it seem more than one aixs can be detected now （even $MI）. Yesterday I burned a TMC2209 module due to an operating error. So I can't test it completely today.I will test it again when the new module arrives.

bdring, 2024-05-03:
How do you set the address??

HiroKwok, 2024-05-06:
> How do you set the address??

[image]

I set the address via jumpers.

Finally, I adjusted the resistance to 100Ω, and the problem seemed to be solved, and all three axes could be recognized normally.

closed 2025-01-21 by bdring