bdring/FluidNC #1218: Problem: GPIO2 use for limit_neg_pin can't pu 3.3v only 0.31v

Yaque, 2024-05-10:
### Wiki Search Terms

FluidNC Axes Setup 
ESP32 Pin Reference
GPIO_NUM_2
Some dev boards have an LED on this. It does not work well as an input if this is the case because the LED affects the voltage on the pin.
It is a strapping pin. It must be either left unconnected/floating, or driven Low, in order to enter the serial bootloader [(read more)](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html?highlight=gpio2)
It works well as the spindle pin. The LED gives a nice indication of the spindle on and PWM level.

### Controller Board

midtbot

### Machine Description

midtbot

### Configuration file

```yaml
name: "midtbot stepstick"
board: "midtbot"

kinematics:
  midtbot:

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 8000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 100
      feed_mm_per_min: 500.000
      seek_mm_per_min: 2000.000
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.2:low:pu
      pulloff_mm: 1.000
      stepstick:
        direction_pin: gpio.26:high
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 8000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 100
      feed_mm_per_min: 500.000
      seek_mm_per_min: 2000.000
      positive_direction: false

    motor0:
      limit_neg_pin: gpio.4:low:pu
      pulloff_mm: 1.000
      stepstick:
        direction_pin: gpio.25:low
        step_pin: gpio.14
    motor1:
      null_motor:

  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 5
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: true

    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.27
        min_pulse_us: 1000
        max_pulse_us: 2000
      
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
```

### Startup Messages

```text
$G
$SS
[MSG:INFO: FluidNC v3.0.x (noGit)  (noGit)]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]]
[MSG:INFO: Local filesystem type is spiffs]]
[MSG:INFO: Configuration file:midtbot_stepstick.yaml]]
[MSG:INFO: Machine midtbot stepstick]]
[MSG:INFO: Board midtbot]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Shared stepper disable gpio.13]]
[MSG:INFO: Axis X (100.000,1100.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.26 Disable:NO_PIN]]
[MSG:INFO:  X Neg Limit gpio.2:low:pu]]
[MSG:INFO:   Motor1]]
[MSG:INFO: Axis Y (100.000,1100.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     stepstick Step:gpio.14 Dir:gpio.25:low Disable:NO_PIN]]
[MSG:INFO:  Y Neg Limit gpio.4:low:pu]]
[MSG:INFO:   Motor1]]
[MSG:INFO: Axis Z (-5.000,0.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     rc_servo Pin:gpio.27 Pulse Len(1000,2000 period:1048575)]]
[MSG:INFO:     Update timer for rc_servo at 20 ms]]
[MSG:INFO:   Motor1]]
[MSG:INFO: Kinematic system: midtbot]]
[MSG:INFO: Using spindle NoSpindle]]
[MSG:INFO: Connecting to STA SSID:MPL]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
[MSG:INFO: Connected - IP is 192.168.100.164]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]]
[MSG:INFO: SSDP Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:X|WCO:0.000,0.000,0.000>
```

### What happened?

limit_neg_pin: gpio.2:low:pu  only 0.31V  homing x can't work   , change gpio2 for direction_pin , gpio can 3.1V ,  gpio26 can 3.1

MitchBradley, 2024-05-10:
What do you expect us to do about it?

Yaque, 2024-05-11:
> What do you expect us to do about it?

to set  limit_neg_pin: gpio.2:low:pu 

pin  voltage   only   0.31V    , but  pin  voltage  3.3V  is  rihgt

limit switch error

homing X  can't  working

bdring, 2024-05-11:
Can you add an external pullup resistor of a low value like 1k?

Yaque, 2024-05-12:
> Can you add an external pullup resistor of a low value like 1k?

thank! it's working.

closed 2024-05-12 by Yaque