bdring/FluidNC #1221: Problem: FluidNC 6XCNC Controller - strange LED lights and steppers suddenly not working

shenzhenadam, 2024-05-14:
### Wiki Search Terms

Could not find anything in wiki that matches my issue

### Controller Board

6X CNC controller

### Machine Description

1. Custom built CNC machine - XYYZ
2. 4 stepper drivers using Shenli SL2690A
3. 2 Power supplies: Meanwell LRS-350-24
4. 4 Steppers: Shenli SL60S21C0C104PJ3-0830 1.8Deg/STEP DC 3.0A

### Input Circuits

```yaml
I doubt that there is an issue here as the machine/6X was working perfectly the day before.
```

### Configuration file

```yaml
Hello, I have a new issue with the FluidNC 6XCNC Controller. While yesterday I did my first test run on the wood plank: https://youtu.be/z4_9rOci9T8?si=AOMqgIGy_-KCQBn9 - and everything went well ...
And today I was about to start a new job, but nothing worked correctly - The Z axis did not move, the Y axis would move but the two steppers not in sync, and the X axis would move - but I was not able to home any of the axis. Also the LEDs were not lighting up correctly. I have not touched anything since that job yesterday.

I am using UGS via USB cable and I am not using the FluidNC WebUI

I have since detached the controller from the steppers and machine and have again clean installed the latest firmware and uploaded the most recent 6x_default.yaml from github to see if my custom yaml was corrupted somehow overnight. And then testing again to see what happens with the lights - it looks strange still: see image & video here: https://imgur.com/a/fluidnc-6xcnc-contoller-issue-iXHO72h
Is this a component failure or something in the firmware corrupted? I am getting no error codes. Is there something I can do by terminal to see what's going on - any other suggestion please.

Please see attached yaml file I was using yesterday which worked fine with my machine.

  y:
    steps_per_mm: 2560.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 600.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 300.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.26:high:pu
      hard_limits: false
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
    motor1:
      limit_neg_pin: NO_PIN    
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.33:high:pu
      hard_limits: false
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

  z:
    steps_per_mm: 2560.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 150.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 1.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.32:high
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15
        
 
        
       
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  
probe:
  pin: gpio.39:low
  toolsetter_pin: gpio.36:low
  
# Using MOSFETs (Check Spindle Pin Usage
coolant:
  flood_pin: gpio.12
  mist_pin: gpio.4
  delay_ms: 0

start:
  must_home: false
  
## Begin Huanyang  
# uart1:
  # txd_pin: gpio.15
  # rxd_pin: gpio.16
  # rts_pin: gpio.14
  # baud: 9600
  # mode: 8N1

# Huanyang:
  # uart_num: 1
  # modbus_id: 1
  # tool_num: 0
  # speed_map: 0=0% 0=25% 6000=25% 24000=100%
  # off_on_alarm: false

# #begin PWM
pwm:
  pwm_hz: 5000
  direction_pin: NO_PIN
  output_pin: gpio.13
  enable_pin: gpio.14
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 10000=100.000%
  off_on_alarm: false
  
# #begin Laser
# Laser:
  # pwm_hz: 5000
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true
  
# #begin 10V
# 10V:
  # forward_pin: gpio.15
  # reverse_pin: gpio.14
  # pwm_hz: 5000
  # output_pin: gpio.13
  # enable_pin: NO_PIN
  # direction_pin: NO_PIN
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0  
  # tool_num: 0
  # speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  # off_on_alarm: false
```

### Startup Messages

```text
Please note: This is taken from 6X after it has been removed from the machine and steppers. See explanation in "what happened"

Grbl 3.7 [FluidNC v3.7.17 (wifi) '$' for help]
$ss
[MSG:INFO: FluidNC v3.7.17 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:INFO: Configuration file:6x_default.yaml]]
[MSG:INFO: Machine 6x Default]]
[MSG:INFO: Board 6x]]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:254ms]]
[MSG:INFO: Axis count 6]]
[MSG:INFO: Axis X (150.000,450.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]]
[MSG:INFO:  X Neg Limit gpio.2:low:pu]]
[MSG:INFO: Axis Y (-150.000,150.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]]
[MSG:INFO:  Y Neg Limit gpio.26:low:pu]]
[MSG:INFO: Axis Z (-150.000,150.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]]
[MSG:INFO:  Z Neg Limit gpio.33:low]]
[MSG:INFO: Axis A (150.000,1110.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]]
[MSG:INFO:  A Neg Limit gpio.32:low]]
[MSG:INFO: Axis B (150.000,350.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]]
[MSG:INFO:  B Neg Limit gpio.35:low]]
[MSG:INFO: Axis C (150.000,350.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.21 Dir:I2SO.20 Disable:I2SO.23]]
[MSG:INFO:  C Neg Limit gpio.34:low]]
[MSG:INFO: Kinematic system: Cartesian]]
[MSG:INFO: PWM Spindle Ena:gpio.14 Out:gpio.13 Dir:NO_PIN Freq:5000Hz Period:8191]]
[MSG:INFO: Using spindle PWM]]
[MSG:INFO: Flood coolant gpio.12]]
[MSG:INFO: Mist coolant gpio.4]]
[MSG:INFO: Probe gpio.39:low]]
[MSG:INFO: Toolsetter gpio.36:low]]
[MSG:INFO: STA SSID is not set]]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]]
[MSG:INFO: AP started]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Captive Portal Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]
ok
```

### User Interface Software

UGS

### What happened?

Hello, I have a new issue with the FluidNC 6XCNC Controller. While yesterday I did my first test run on the wood plank: https://youtu.be/z4_9rOci9T8?si=AOMqgIGy_-KCQBn9 - and everything went well ...
And today I was about to start a new job, but nothing worked correctly - The Z axis did not move, the Y axis would move but the two steppers not in sync, and the X axis would move - but I was not able to home any of the axis. Also the LEDs were not lighting up correctly. I have not touched anything since that job yesterday.

I am using UGS via USB cable and I am not using the FluidNC WebUI

I have since detached the controller from the steppers and machine and have again clean installed the latest firmware and uploaded the most recent 6x_default.yaml from github to see if my custom yaml was corrupted somehow overnight. And then testing again to see what happens with the lights - it looks strange still: see image & video here: https://imgur.com/a/fluidnc-6xcnc-contoller-issue-iXHO72h
Is this a component failure or something in the firmware corrupted? I am getting no error codes. Is there something I can do by terminal to see what's going on - any other suggestion please.

Please see attached yaml file I was using yesterday which worked fine with my machine.

### Other Information

This shows the issue:
https://imgur.com/a/fluidnc-6xcnc-contoller-issue-iXHO72h
This shows the machine working perfectly the day before:
https://youtu.be/z4_9rOci9T8?si=AOMqgIGy_-KCQBn9

bdring, 2024-05-15:
I think you should have 2 files on your ESP32. 

config.yaml
io_test.nc

You can check with the `$localfs/list` command. If you don't have them let me know.

If you do, send `$config/filename=config.yaml` to set that as your config file. Then send `$localfs/run=io_test.nc`

It should blink all the LEDs. That will be a good first test to do.

shenzhenadam, 2024-05-15:
I don't have config.yaml or io_test.nc on my ESP32:
Grbl 3.7 [FluidNC v3.7.17 (wifi) '$' for help]
$localfs/list
[FILE: 6x_default.yaml|SIZE:5437]
[FILE: favicon.ico|SIZE:1150]
[FILE: index.html.gz|SIZE:117066]
[FILE: taig_probe.flnc|SIZE:1649]
[/littlefs/ Free:52.00 KB Used:140.00 KB Total:192.00 KB]

[image]

shenzhenadam, 2024-05-15:
Thanks .. Yes I found those already .. I'm doing that now .. please wait a sec ..

shenzhenadam, 2024-05-15:
Test of video here:
https://imgur.com/a/T8gJ21t

$localfs/run=io_test.nc
ok
[MSG:INFO: GCode Comment...IO Test Begin]
[MSG:INFO: Enabling all motors]
[MSG:INFO: Disabling all motors]
[MSG:INFO: Enabling all motors]
[MSG:INFO: Disabling all motors]
[MSG:INFO: Enabling all motors]
[MSG:INFO: Disabling all motors]
[MSG:INFO: GCode Comment...Tesing IO4]
[MSG:INFO: GCode Comment...Tesing IO12]
[MSG:INFO: GCode Comment...Tesing IO14 and IO15]
[MSG:INFO: GCode Comment...Test Complete]
[MSG:/littlefs/io_test.nc file job succeeded]

bdring, 2024-05-15:
The video suggests that motor outputs 3 and 4 are not working correctly. They both connect to the same component. It runs off the same signals as X and Y, so there is no evidence of an ESP32 or config problem.

The component is proven to be quite robust, but it is possible to damage it with ESD or overcurrent.

[image]
 
If you bought it from Elecrow or me I can replace it. If it happens again there is likely a problem on your end.

You might try changing the config to use motor outputs 5 and 6 for a while to see if they work before using a new one. 

[image]

shenzhenadam, 2024-05-15:
Thanks for that .. Yes I could try moving the motors to 5 & 6 as a temporary fix ... but later I will want to use those outputs for rotary axis A/B. Yes I purchased via Elecrow - that would be great to get a replacement - shall I send back this unit in return?
I will change the config to motors 5 & 6 tomorrow and test.
Many thanks!

bdring, 2024-05-17:
Your problem is quite odd. I am not sure I have heard of it before. The chips are still functioning as shift registers, but the output section is damaged. 

I suggest you continue to test it for a while to see if the new arrangement holds up.  I would hate to see a replacement controller fail the same way.

Elecrow just sold out, They should get some more in about 2 weeks. I have some in stock that can be sent if it is urgent.

shenzhenadam, 2024-05-27:
Hello .. Sorry for the delay .. The FluidNC controller has worked fine during a long 9 hour V-Carve - I will post a link to some video clips later.

shenzhenadam, 2024-05-27:
Here's a video clip - you can watch from 3:11 https://youtu.be/5iAweBlNY3M?si=lI8tQ7oZL3Ech0c6&t=191

shenzhenadam, 2024-06-17:
Hello, I have tried & tested several new jobs using the same axis with 1,2,5&6 (XYYZ)   - seems to be stable for now ... However I will be needing all 6 axis relatively soon .. Can you please indicate when you send out the replacement board - I can then send this board back to the factory. Many thanks!

bdring, 2024-06-18:
Contact me via DM on our Discord server.  I need some details from you about the order. 

https://discord.gg/FbmHW2CK

martinbircher, 2025-10-25:
Hi,
My search about problems with the similar V1 Jackpot CNC controller let me here. I think I killed one of the 74AHCT595PW,118 on that board. First the C axis and then the B axis. I could confirm that the ESP is not the problem by changing it. What I did (probably causing the problem was connecting a MKS SERVO42C board to it. The first mistake was connecting it under power which caused a voltage spike killing the outputs for C and then later the motor was stalling which killed also B.
Now when I looked at it with the oscilloscope I noticed that DIR and STP are constantly high.
I wonder if there is a way to protect the chip better?
My approach is now to try to replace the 74AHCT595PW,118 chip.