bdring/FluidNC #1223: [MSG:ERR: Command requires idle state]

alphaonex86, 2024-05-16:
### Wiki Search Terms

Most command give:
[MSG:ERR: Command requires idle state]
Include when I try home X
$Grbl/HomingCycleEnable=0
$Grbl/HardLimitsEnable=1
$Grbl/SoftLimitsEnable=0
What I can check to solve this?

### Controller Board

ESP32

### Machine Description

ESP32 with X/Y/Z

### Configuration file

```yaml
name: 1L_CM
board: ESP32 DEVKIT V1
axes:
  x:
    steps_per_mm: 160
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 2000
    homing:
      cycle: 2
      mpos_mm: 5
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: gpio.2:pu
      limit_pos_pin: gpio.4:pu
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.5
        direction_pin: gpio.13
        disable_pin: NO_PIN
  y:
    steps_per_mm: 160
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 5
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: gpio.14:pu
      limit_pos_pin: gpio.15:pu
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.16
        direction_pin: gpio.17
        disable_pin: NO_PIN
  z:
    steps_per_mm: 160
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 5
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: gpio.18:pu
      limit_pos_pin: gpio.19:pu
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.25
        direction_pin: gpio.26
        disable_pin: NO_PIN
start:
  check_limits: true
control:
  estop_pin: gpio.23
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
stepping:
  engine: RMT
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
i2c0:
   sda_pin: gpio.21
   scl_pin: gpio.22
oled:
  i2c_num: 0
  i2c_address: 60
  width: 128
  height: 64
  radio_delay_ms: 0
probe:
  pin: gpio.32:pu
  toolsetter_pin: no_pin
  check_mode_start: false
```

### Startup Messages

```text
[MSG:ERR: Command requires idle state]
```

### User Interface Software

Serial

### What happened?

Most command give:
[MSG:ERR: Command requires idle state]

### GCode File

$HX

MitchBradley, 2024-05-16:
That is not the startup messages.  That is the message after startup.  Search the wiki for "startup messages".  The New Issue template also tells you how to get the startup messages; read it carefully

bdring, 2024-05-17:
`[MSG:ERR: Command requires idle state]` is a clue. The cause of "not idle" is likely shown before this message. Read the startup messages.

If you find the problem please let us know so this issue can be closed.

alphaonex86, 2024-05-17:
Yes, ESP32 with hardware problem, same firmware/config file on another ESP32 work perfectly

closed 2024-05-17 by alphaonex86

MitchBradley, 2024-05-17:
The next time that you file an issue, please read the instructions about startup messages carefully and submit the full set of them.  If we had that, it would have been much easier to diagnose the problem.