bdring/FluidNC #1224: Problem: Homing X in >3.7.10 not working on ESP32

jkeating732, 2024-05-17:
### Wiki Search Terms

Not homing X axis

### Controller Board

MKS Tinybee v1.0 

### Machine Description

MPCNC Primo

### Configuration file

```yaml
board: MKS TinyBee V1.0 XXYYZ
name:
 
kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

#fan
# user_outputs:
#   digital0_pin: i2so.19

axes:
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 325.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      hard_limits: true
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0

    # use E0 driver for 2nd X axis motor
    motor1:
      limit_neg_pin: gpio.32
      hard_limits: true
      pulloff_mm: 2.000
      stepstick:
        # step_pin: I2SO.10
        # direction_pin: I2SO.11
        # disable_pin: I2SO.9
        step_pin: I2SO.4
        direction_pin: I2SO.5
        disable_pin: I2SO.3
[... 364 lines cut ...]
[MSG:INFO:  Y2 Neg Limit gpio.35]
[MSG:INFO: Axis Z (0.000,80.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.14 Disable:I2SO.12]
[MSG:INFO: macro0_pin gpio.4:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.2:pd Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:myWIFI]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.50.63]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.0 [FluidNC v3.0.x (noGit) (wifi) '$' for help]
ok
[VER:3.0 FluidNC v3.0.x (noGit):]
[OPT:PHS]
[MSG: Machine: ]
[MSG: Mode=STA:SSID=myWIFI:Status=Connected:IP=192.168.50.63:MAC=0C-8B-95-EF-B4-D0]
```

bdring, 2024-05-17:
Show what happens when you try to home with `$H` in the latest version of the firmware with '$message/level=debug' . Use fluidTerm or the web terminal

Use the web install or release files, so the revision is properly shown in the messages.

jkeating732, 2024-05-17:
I have to manually build and upload via PlatformIO because the upload_speed is different than the web will allow (115200). I could not do a $H because I don't have a Z stop in place yet. Below I uploaded 3.7.17 and ran

### Commands Ran
```
$message/level=debug
$HY
$HX
```

### Output
```

Grbl 3.0 [FluidNC v3.0.x (noGit) (wifi) '$' for help]

ok

ok
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.0.x (noGit)  (noGit)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ]
[MSG:INFO: Board MKS TinyBee V1.0 XXYYZ]
[MSG:INFO: UART1 Tx:gpio.17 Rx:gpio.16 RTS:NO_PIN Baud:1000000]
[MSG:INFO: uart_channel1 created at report interval: 75]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-325.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.33]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:  X2 Neg Limit gpio.32]
[MSG:INFO: Axis Y (-220.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:  Y Neg Limit gpio.22]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.11 Disable:I2SO.9]
[MSG:INFO:  Y2 Neg Limit gpio.35]
[MSG:INFO: Axis Z (0.000,80.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.14 Disable:I2SO.12]
[MSG:INFO: macro0_pin gpio.4:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.2:pd Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID: myWIFI]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.50.63]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.0 [FluidNC v3.0.x (noGit) (wifi) '$' for help]
ok
[VER:3.0 FluidNC v3.0.x (noGit):]
[OPT:PHS]
[MSG:Machine: ]
[MSG:Mode=STA:SSID= myWIFI:Status=Connected:IP=192.168.50.63:MAC=0C-8B-95-EF-B4-D0]
ok

Grbl 3.0 [FluidNC v3.0.x (noGit) (wifi) '$' for help]
$message/level=debug
ok
[MSG:DBG: WebSocket 0 from 192.168.50.56 uri /]
$HY
[MSG:DBG: Homing Cycle Y]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,0.000]
[MSG:DBG: Planned move to 0.000,242.000,0.000 @ 5000.000]
[MSG:DBG:  Y2 Neg Limit 1]
[MSG:DBG: Homing limited Y2]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Homing limited Y Y2]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from 0.000,24.010,0.000]
[MSG:DBG: Planned move to 0.000,22.010,0.000 @ 300.000]
[MSG:DBG:  Y2 Neg Limit 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: CycleStop Pulloff0]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from 0.000,22.010,0.000]
[MSG:DBG: Planned move to 0.000,24.210,0.000 @ 300.000]
[MSG:DBG:  Y2 Neg Limit 1]
[MSG:DBG: Homing limited Y2]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Homing limited Y Y2]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from 0.000,23.910,0.000]
[MSG:DBG: Planned move to 0.000,21.910,0.000 @ 300.000]
[MSG:DBG:  Y2 Neg Limit 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was 0.000,21.910,0.000]
[MSG:Homed:Y]
[MSG:DBG: mpos becomes 0.000,0.000,0.000]
[MSG:DBG: mpos transformed 0.000,0.000,0.000]
[MSG:DBG: Homing done]
ok
[MSG:INFO: uart_channel1 auto report interval set to 200 ms]
[MSG:INFO: uart_channel1 auto report interval set to 200 ms]
$HX
[MSG:DBG: Homing Cycle X]
[MSG:DBG: Homing nextPhase CycleDone]
[MSG:DBG: mpos was 0.000,0.000,0.000]
[MSG:Homed:]
[MSG:DBG: mpos becomes 0.000,0.000,0.000]
[MSG:DBG: mpos transformed 0.000,0.000,0.000]
[MSG:DBG: Homing done]
ok
```

MitchBradley, 2024-05-17:
> I have to manually build and upload via PlatformIO because the upload_speed is different than the web will allow (115200).

I don't understand that.  upload_speed only affects the step where platformio is installing the newly-compiled firmware onto the ESP32.  It does not affect the way that the binary is compiled so the usage behavior should be identical between a release binary and user-compiled binary.

If you do need to compile yourself, it is much better to use git clone to download the files, instead of getting a .zip file from github.  vscode knows how to do git clone with a click.  If it were possible to configure github so .zip source downloads were unavailable, I would do so, as they are a nuisance.

jkeating732, 2024-05-17:
I was using PlatformIO to upload the binary, in the platformio.ini file I need to set this line ```upload_speed = 115200``` so it can upload the image. If I use the web via https://installer.fluidnc.com it does not allow me to set the upload speed.

Is there a way that I'm not aware of to have PlatformIO upload a binary that isn't compiled? Happy to try if there is. :)

bdring, 2024-05-17:
It sounds like you have some non standard hardware issues. You should explain that setup to us.

MitchBradley, 2024-05-17:
PlatformIO uses the standard esptool program to upload - the same tool that is included in our release installer packages.  You could get a release package (for example https://github.com/bdring/FluidNC/releases/download/v3.7.17/fluidnc-v3.7.17-posix.zip) and use the script therein to do the install.  If you need to use a different binary than the one in the release package, just replace the file wifi/firmware.bin with the one you want.  If you need to change the upload baud rate, edit install-wifi.bat (windows version) or tools.sh (posix version).

closed 2025-01-21 by bdring