bdring/FluidNC #1225: Problem:

UrbanBobek, 2024-05-19:
### Wiki Search Terms

Homing/homing cycle/seek_mm_per_min

### Controller Board

6-pack, v1p5

### Machine Description

The axis I am trying to run is using an external stepper motor driver. The motor in a nema23 stepper. I am using it to drive a rotary axis, so the `steps_per_mm` are set to represent the rotation of the axis in degrees. I have the axis marked/connected as the Z axis.

  

### Configuration file

```yaml
board: 6 Pack
name: Strickar 
meta: 2024-05-19
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 1185.20
    max_rate_mm_per_min: 800.000
    acceleration_mm_per_sec2: 40.000
    max_travel_mm: 72.0
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.0
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.33
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.500
      stepstick:
        ms3_pin: i2so.3
        step_pin: I2SO.2
        direction_pin: I2SO.1:low
        disable_pin: I2SO.0
        
  y:
    steps_per_mm: 1538.5
    max_rate_mm_per_min: 500.000
    acceleration_mm_per_sec2: 20.000
    max_travel_mm: 62.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.0
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 0.5
      stepstick:
        ms3_pin: i2so.6
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
        
  z:
    steps_per_mm: 33.329
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 130.000
    max_travel_mm: 10000000.0
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.0
      feed_mm_per_min: 500.000
      seek_mm_per_min: 1400.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.0
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8
        
  a:
    steps_per_mm: 100
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 77.00
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.0
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    
    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.12
        min_pulse_us: 2200
        max_pulse_us: 500
        
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

start:
  must_home: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.7.17 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is spiffs]]
[MSG:INFO: Configuration file:config.yaml]]
[MSG:INFO: Machine Strickar ]]
[MSG:INFO: Board 6 Pack]]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]]
[MSG:INFO: Axis count 4]]
[MSG:INFO: Axis X (-72.000,0.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1:low Disable:I2SO.0]]
[MSG:INFO:  X Pos Limit gpio.33]]
[MSG:INFO: Axis Y (-62.000,0.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]]
[MSG:INFO:  Y Pos Limit gpio.34]]
[MSG:INFO: Axis Z (-10000000.000,0.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]]
[MSG:INFO:  Z Pos Limit gpio.35]]
[MSG:INFO: Axis A (0.000,77.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     rc_servo Pin:gpio.12 Pulse Len(2200,500 period:1048575)]]
[MSG:INFO:     Update timer for rc_servo at 20 ms]]
[MSG:INFO: Kinematic system: Cartesian]]
[MSG:INFO: Using spindle NoSpindle]]
[MSG:INFO: Connecting to STA SSID:strickarNET]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
[MSG:INFO: Connected - IP is 192.168.1.10]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]]
[MSG:INFO: SSDP Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]
```

### What happened?

The problem is with the homing speed. Currently the first search of the homing switch is painfully slow. Regardless of the value I set for the `seek_mm_per_min` the homing speed does no seem to change. Also, despite the `feed_mm_per_min` value being lower the `seek_mm_per_min`, the speed of second homing cycle is much faster.

The homing speed is slow regardless if I start the homing procedure from the WebUI or FluidTerm (with `$HZ`). 

I am attaching the debug messages when I run the homing procedure for the Z axis:
```
$HZ                                
[MSG:DBG: Homing Cycle Z]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,-9.991]
[MSG:DBG: Planned move to 0.000,0.000,1099990.000 @ 1400.000]
[MSG:DBG:  Z Pos Limit 1]
[MSG:DBG: Homing limited Z]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from 0.000,0.000,0.960]
[MSG:DBG: Planned move to 0.000,0.000,-0.040 @ 500.000]
[MSG:DBG:  Z Pos Limit 0]
[MSG:DBG: CycleStop Pulloff0]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from 0.000,0.000,-0.030]
[MSG:DBG: Planned move to 0.000,0.000,1.070 @ 500.000]
[MSG:DBG:  Z Pos Limit 1]
[MSG:DBG: Homing limited Z]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from 0.000,0.000,0.990]
[MSG:DBG: Planned move to 0.000,0.000,-0.010 @ 500.000]
[MSG:DBG:  Z Pos Limit 0]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was 0.000,0.000,0.000]
[MSG:Homed:Z]
[MSG:DBG: mpos becomes 0.000,0.000,0.000]
[MSG:DBG: mpos transformed 0.000,0.000,0.000]
[MSG:DBG: Homing done]
ok

```

bdring, 2024-05-19:
Try jogging the axis at the same speed as your seek rate. How does that compare to your homing speed.

BTW. Do you always use mm or sometimes inches?

Axes XYZ should be used for rotary axes especially if you use inches.

UrbanBobek, 2024-05-20:
I set the `axis/z/max_rate_mm_per_min` to `1400.000000` and jogged the axis. If I press the move button for the z axis on the webUI I can see that the `$J=G91 G21 F100 Z10` command is sent and the axis moves slowly, similarly to when the homing command is send. However, if I send, for example, the `Z100` command, the axis moves much faster.
I observe the same behaviour with the settings I provided above.

I am not sure if this is relevant, but when running the `Z<some_value>` command I can see in the terminal that the FS (feed speed?) is set to the value of  `axis/z/max_rate_mm_per_min`
```
<Run|MPos:0.000,0.000,87.461,0.000|FS:1400,0>
```
, but when I start the homing procedure, the FS is set to 858
```
[MSG:DBG: Planned move to 0.000,0.000,10999990.000 @ 1400.000]
<Idle|MPos:0.000,0.000,-9.991,0.000|FS:858,0>
<Home|MPos:0.000,0.000,-7.831,0.000|FS:858,0|WCO:0.000,-27.000,0.000,0.000>
<Home|MPos:0.000,0.000,-4.411,0.000|FS:858,0|Ov:100,100,100>
<Home|MPos:0.000,0.000,-0.990,0.000|FS:858,0>
```
If I move the Z axis on the webUI the FS is 100 (I assume because of the command  `$J=G91 G21 F100 Z10` )
```
<Jog|MPos:0.000,0.000,0.000,0.000|FS:100,0|Ov:100,100,100>
<Jog|MPos:0.000,0.000,-3.420,0.000|FS:100,0>
<Jog|MPos:0.000,0.000,-8.401,0.000|FS:100,0>
```
I should note again that the movement speed of homing and jogging via the webUI is similar (and not 8x greater).

I am not sure if I totally understand the question. For all axis except Z I am using mm. For Z I want the units to be degrees of rotation of the plate that is connected to the Z axis motor (via a 1:4 gear ratio), similarly to how your String art machine functioned (if I understood that setup correctly).

closed 2025-01-11 by bdring