bdring/FluidNC #1227: Problem: Web GUI  unresponsive, Pendant have a blacscreen

fvargasfrank, 2024-05-24:
### Wiki Search Terms

N/A

### Controller Board

6 Pack External XYYZ YL620

### Machine Description

CNC4Newbie

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYYZ YL620
meta:
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 133.450
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 1291
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 133.847
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 1312
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
    
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

  z:
    steps_per_mm: 200.236
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 150.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.32:low
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

probe:
  pin: gpio.36:low
  toolsetter_pin: NO_PIN
  check_mode_start: true

uart2:
  txd_pin: gpio.13
  rxd_pin: gpio.14
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

uart_channel2:
  uart_num: 2
  report_interval_ms: 75

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.39:low
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
        
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

start:
  must_home: false
  
coolant:
  flood_pin: gpio.25
  mist_pin: gpio.2
  delay_ms: 0

  
10V:
  output_pin: gpio.26 
  forward_pin: gpio.4  
  reverse_pin: gpio.16 
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 6000=0% 24000=100%
```

### Startup Messages

```text
$G
[MSG:ERR: mount_to_vfs failed code 0xffffffff]
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok

$SS
[MSG:INFO: FluidNC v3.7.17 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:INFO: Configuration file:config.yml]]
[MSG:INFO: Machine 6 Pack External XYYZ YL620]]
[MSG:INFO: Board 6 Pack]]
[MSG:INFO: UART2 Tx:gpio.13 Rx:gpio.14 RTS:NO_PIN Baud:1000000]]
[MSG:INFO: uart_channel2 created at report interval: 75]]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Axis X (0.000,1291.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]]
[MSG:INFO:  X Neg Limit gpio.34:low]]
[MSG:INFO: Axis Y (0.000,1312.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]]
[MSG:INFO:  Y Neg Limit gpio.35:low]]
[MSG:INFO:   Motor1]]
[MSG:INFO: Using spindle 10V]]
[MSG:INFO: Flood coolant gpio.25]]
[MSG:INFO: Mist coolant gpio.2]]
[MSG:INFO: Probe gpio.36:low]]
[MSG:INFO: Connecting to STA SSID:Not Found]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
[MSG:INFO: Connecting...]]
[MSG:INFO: Telnet started on port 23]]
ok
```

### User Interface Software

WebUI

### What happened?

I'm not sure what happened. After the machine completed a job, the pendant displayed "NC." However, the machine remained responsive. After I rebooted the Machine, The pendant has a black screen and the controller's web GUI is not coming up. The ESP board has a red light, but I'm not sure if it was always on.
power off for 10 minutes and unplug the controller 
GUI came back on (Black screen on the pendant)

[MSG:ERR: mount_to_vfs failed code 0xffffffff]

Try new sd card and same error new sd work fine on PC

Not sure if its realated to anything but the past couple of runs 
The machine would freeze in place for a few secounds and then continue to move

bdring, 2024-05-24:
Let's start with the pendant.

I'll assume the pendant is the FluidDial, if not let me know what it is. 

It appears to be broken. It should have some display regardless of the controller. Disconnect it from the controller and power it from the USB. Reply with what happens.

fvargasfrank, 2024-05-24:
Just connected it to usb c, windows says unrecognizable usb and it still shows black screen, probably went dead.

fvargasfrank, 2024-05-25:
The sd card works now with the old one, nothing new just left it off for the night, I ran a job and the machine still freezes.

https://github.com/bdring/FluidNC/assets/35478171/6593cdf6-1635-4204-8bd1-0756849ec38f

bdring, 2024-06-03:
Try changing this to I2S_STATIC in your config file.

```yaml
stepping:
  engine: I2S_STATIC
```

Send $I at the console and show me the response.

fvargasfrank, 2024-06-03:
[VER:3.7 FluidNC v3.7.17 (v3-7-17-017cf0b9):]
[OPT:MPHS]
[MSG:Machine: 6 Pack External XYYZ YL620]
[MSG:Mode=STA:SSID=Not Found:Status=Connected:IP=192.168.1.5:MAC=B0-A7-32-DD-6B-60]
ok

bdring, 2024-06-03:
Try running with I2S_STATIC and keep a FluidTerm connection open the whole time. If you get a lock up see if there are any debugging messages there.

The SSID=Not Found:Status=Connected is odd.

MitchBradley, 2024-06-03:
Aha, an out of memory condition.  That is hard to fix.  It can sometimes happen when there is a lot of other network traffic that ends up consuming a lot of memory inside the Espressif networking code.  There is not a lot we can do about it in the short term.  One thing that can help is to use a smaller SD card.  High capacity SD cards use a lot of memory because they have very large chunk sizes on the card.  Another thing that might help is to put FluidNC on a dedicated WiFi router, so it is not subjected to traffic from other things on the network.

MitchBradley, 2024-06-03:
On the topic of SD cards that work on PC but not on FluidNC: Modern high-capacity SD cards store data in large chunks, up to 64KBytes in size.  That chunk size affects the amount of memory that is necessary in order to read a file from the card, even if the file itself is small.  64KBytes is a small amount for a PC that has many gigabytes of memory, but it is a lot for an ESP32 that has a total of only 520 KBytes, of which 80% is already used for other purposes before any SD card accesses are attempted.

fvargasfrank, 2024-06-03:
I will get a 4gb sd card and go from there, will probably isolate the wifi by putting it on a private VLAN with only access to my phone and pc