bdring/FluidNC #1231: Problem with homing with 2 motor

JoanMOliver, 2024-05-27:
### Wiki Search Terms

Pull_off problems, doble motor per axis, homing problems, control direction,

### Controller Board

MKS TinyBee

### Machine Description

1.-CO2 Laser machine  
2.-One motor in axis X and  2 motors in Axis Y,
3.-External Drivers 
4.- Stepper Motors
5.- inductive Sensors 

### Configuration file

```yaml
board: Laser Dolivers
name: MKS TinyBee Laser 4
meta: Joan M Oliver 1 2024

start:
  must_home: false
  deactivate_parking: false
  check_limits: false

kinematics:
  Cartesian:

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  # shared_fault_pin: gpio.2
  # homing_runs: 2
  
  x:
    steps_per_mm: 160
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 1200.000
    max_travel_mm: 1250.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:high:pd
      hard_limits: true
      pulloff_mm: 10.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: NO_PIN

  y:
    steps_per_mm: 160
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 800.000
    max_travel_mm: 2500.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 150.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:high:pd
      hard_limits: true
      pulloff_mm: 10
      stepstick:
        step_pin: I2SO.4
        direction_pin: I2SO.5
        disable_pin: NO_PIN

    motor1:
      limit_neg_pin: gpio.22:high:pd
      hard_limits: true
      pulloff_mm: 10
      stepstick:
        step_pin: I2SO.7
        direction_pin: I2SO.8
        disable_pin: NO_PIN

#   z:
#    steps_per_mm: 320
#     max_rate_mm_per_min: 2000.000
#     acceleration_mm_per_sec2: 500.000
#     max_travel_mm: 52.000
#     soft_limits: true
#     homing:
#       cycle: 1
#       positive_direction: true
#       mpos_mm: 0.000
#       seek_mm_per_min: 1000.000
#       settle_ms: 500
#       seek_scaler: 1.100
#       feed_scaler: 1.100

#     motor0:
#       limit_neg_pin: gpio.35:high:pd
#       hard_limits: true
#       pulloff_mm: 5.00
#       stepstick:
#        step_pin: I2SO.10
#        direction_pin: I2SO.11
#        disable_pin: NO_PIN
 
#  a:
#    # E0
#    steps_per_mm: 157.750
#    max_rate_mm_per_min: 18000.000
#    acceleration_mm_per_sec2: 1500.000
#    max_travel_mm: 325.000
#    soft_limits: true
#    homing:
#      cycle: 1
#      positive_direction: false
#      mpos_mm: 0.000
#      feed_mm_per_min: 300.000
#      seek_mm_per_min: 5000.000
#      settle_ms: 500
#      seek_scaler: 1.100
#      feed_scaler: 1.100
#
#    motor0:
#      limit_neg_pin: gpio.34
#      hard_limits: false
#      pulloff_mm: 2.000
#      stepstick:
#        step_pin: I2SO.13
#        direction_pin: I2SO.14:low
#        disable_pin: NO_PIN

UART1:
  txd_pin: gpio.0
  rxd_pin: gpio.4
  rts_pin: gpio.13
  baud: 115200
  mode: 8N1

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
# safety_door_pin: gpio.21
  reset_pin: NO_PIN
#  feed_hold_pin: gpio.12
  cycle_start_pin: NO_PIN
  # EXP1 BTN_ENC
  # macro0_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
#  estop_pin: gpio.13:high:pu
```

### Startup Messages

```text
[MSG:RST]
[MSG:INFO: FluidNC v3.7.16 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:CNC_Dolivers_MKS_TinyBee.yaml]
[MSG:INFO: Machine MKS TinyBee Laser 4]
[MSG:INFO: Board Laser Dolivers]
[MSG:INFO: UART1 Tx:gpio.0 Rx:gpio.4 RTS:gpio.13 Baud:115200]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1250.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.33:pd]
[MSG:INFO: Axis Y (-2500.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.32:pd]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8 Disable:NO_PIN]
[MSG:INFO:  Y2 Neg Limit gpio.22:pd]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:gpio.16:low Out:gpio.17 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Flood coolant I2SO.3:low]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.16 (wifi) '$' for help]
```

### User Interface Software

WebUi, UGS the same problem with bouth 

### What happened?

 I have a problem with the Y axis when doing the homing. I have FluidNC mounted on an MKS TinyBee (I2S_STATIC) with two motors installed on this axis with a pull-off of 7mm. When I do the homing, there is no problem, both motors go to the homing position. Once they find the negative position sensor, both axes stop. Depending on which one arrives first, it waits for the other. But then, when they go to do the pull-off position, one motor goes in the correct direction inward 7mm, but the other motor goes in the opposite direction towards the sensor incorrectly. 

### Other Information

You will see that I tested it with version FluidNC v3.7.16, but I also tried it with version FluidNC v3.7.17 and got the same result. I chose to continue with version FluidNC v3.7.16 because I'm not sure if it could be a problem with the update.

JoanMOliver, 2024-05-28:
https://github.com/bdring/FluidNC/assets/111784083/c94391df-8be6-47c7-bd12-97662d9f420d

Here there are a video Showing the problem

bdring, 2024-05-28:
Are you 100% sure the switches are associated with the right motors? Many people have the motor0 switch on the Motor1 side by accident and the behavior is not right.

MitchBradley, 2024-05-28:
Your video will not play for me, nor will it download.  I tested 3.7.17 with my dual-y-motor laser gantry, on a Fysetc E4 controller, and squared homing worked perfectly.  If you send $message/level=debug , you will get lots of homing progress report messages that might help.

JoanMOliver, 2024-05-30:
Hello,

After updating the software to version 3.7.17 and conducting more tests on the software, I decided to check with the oscilloscope and found the problem. The signal for the direction was not turning due to a bad connection to the direction pin of the Y-axis Motor 0. I just need to apologize for the time I've made you lose as it was a hardware issue. Thank you very much for your help.

closed 2024-05-30 by MitchBradley