bdring/FluidNC #1232: Second limit switch Y axis is not working properly

FHS1959, 2024-05-27:
### Wiki Search Terms

I am using FYSETC E4 board, dual Y-axis motor with 2 end switches. the $limits reports that the switches are working correctly Motor0 is reported as Lower y and motor1 as upper Y, so far so good.
If homing Y-axis and the end switch of motor0 is triggered motor0 and motor1 are working simultaneous (stop, pull back, home again and pull back), even if end switch of motor1 is not triggered, If I trigger motor1 end switch, the motor stops, if the end switch is released, motor1 continues running in the same direction. Motor0 does not respond on the end switch of motor1.  

### Controller Board

FYSETC E4

### Machine Description

Building my own laser engraver, now testing Y-axis. X-axis is in the config file not yet included.

### Configuration file

```yaml
name: "Fysetc Ant CoreXY"
board: "Fysetc E4"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

uart1:
  txd_pin: gpio.22
  rxd_pin: gpio.21
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1

kinematics:
  corexy:

axes:
  shared_stepper_disable_pin: gpio.25

  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 1.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 3.500
      feed_scaler: 3.500

    motor0:
      limit_neg_pin: gpio.35:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209:
        uart_num: 1
        addr: 3
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.500
        microsteps: 16
        run_mode: stallguard
        homing_mode: stallguard
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        use_enable: false
        step_pin: gpio.33
        direction_pin: gpio.32:low
        disable_pin: NO_PIN

    motor1:
      limit_neg_pin: gpio.34:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209:
        uart_num: 1
        addr: 0
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.500
        microsteps: 16
        run_mode: stallguard
        homing_mode: stallguard
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        use_enable: false
        step_pin: gpio.14
        direction_pin: gpio.12:low
        disable_pin: NO_PIN

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

start:
  must_home: false
  deactivate_parking: false
  check_limits: false
```

### Startup Messages

```text
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.7.17 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine Fysetc Ant CoreXY]
[MSG:INFO: Board Fysetc E4]
[MSG:INFO: UART1 Tx:gpio.22 Rx:gpio.21 RTS:NO_PIN Baud:115200]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.25]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (1.000,301.000)]
[MSG:INFO:   Motor0]
[MSG:DBG: TMCStepper Library Ver. 0x703]
[MSG:INFO:     tmc_2209 UART1 Addr:3 CS:NO_PIN Step:gpio.33 Dir:gpio.32:low Disable:NO_PIN R:0.110]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2209 UART1 Addr:0 CS:NO_PIN Step:gpio.14 Dir:gpio.12:low Disable:NO_PIN R:0.110]
[MSG:INFO:  Y2 Neg Limit gpio.34]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Y Axis driver test passed]
[MSG:DBG: Y Axis Stallguard]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:DBG: Y2 Axis Stallguard]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:Telfy_11461]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.1.72]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.17 (wifi) '$' for help]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y2 Neg Limit 0]
[MSG:DBG: WebSocket 0 from 192.168.1.26 uri /]
[MSG:DBG: WebSocket 1 from 192.168.1.26 uri /]
```

### User Interface Software

WebUI and fluidterm

### What happened?

each motor of the Y-axis should perform his own homing procedure and squaring

### Other Information

Homing debug info
[MSG:DBG: Homing Cycle Y]
[MSG:DBG: Y Axis Stallguard]
[MSG:DBG: Y2 Axis Stallguard]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,100.000,0.000]
[MSG:DBG: Planned move to 0.000,-950.000,0.000 @ 800.000]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Homing limited Y]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from 0.000,-0.738,0.000]
[MSG:DBG: Planned move to 0.000,0.262,0.000 @ 100.000]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: CycleStop Pulloff0]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from -0.125,0.137,0.000]
[MSG:DBG: Planned move to -0.125,-3.362,0.000 @ 100.000]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Homing limited Y]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from -0.125,-0.762,0.000]
[MSG:DBG: Planned move to -0.125,0.238,0.000 @ 100.000]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was -0.125,0.238,0.000]
[MSG:Homed:Y]
[MSG:DBG: mpos becomes -0.125,1.000,0.000]
[MSG:DBG: mpos transformed -0.125,1.000,0.000]
[MSG:DBG: Y Axis Stallguard]
[MSG:DBG: Y2 Axis Stallguard]
[MSG:DBG: Homing done]

MitchBradley, 2024-05-27:
Dual Y motors with CoreXY.  Is that even possible?

FHS1959, 2024-05-27:
Is even in the wiki described. Search on squaring axes

MitchBradley, 2024-05-27:
Please tell me exactly where in the wiki you found a mention of dual-motor corexy kinematics with squaring.  I don't see it.

FHS1959, 2024-05-27:
Here you go
http://wiki.fluidnc.com/en/config/homing_and_limit_switches

MitchBradley, 2024-05-27:
That page does not contain the word "corexy".

Based on the fact that your config file starts with `name: "Fysetc Ant CoreXY"`, I suspect that you started with the example config file on the page http://wiki.fluidnc.com/en/hardware/3rd-party/fysetc_e4 and started modifying it without understanding everything.  CoreXY is a special kind of machine setup that uses a complicated cable arrangement as shown at https://corexy.com/theory.html .  With CoreXY, there are exactly two motors.  In order to move in the X direction only, both motors must turn at the same time, and to move in Y only, both motors also turn, but one turns in the opposite direction from the other. CoreXY has no provision for multiple motors on one axis.  There are only two motors that work in tandem to accomplish X and Y motion.

Please understand that I wrote the homing code, so I think I know what I am talking about.

You need to remove the kinematics: corexy: section.  Either leave it out completely, or change "corexy" to "cartesian".

FHS1959, 2024-05-27:
> That page does not contain the word "corexy".

Thank you very much MitchBradle, that is the solution. You where completely right about the basis I started from. 
One motor is now homing and stops at the end switch. As soon as the other motor hits the end switch they both pull back and are simultaneous homing again and pulling back. PERFECT

closed 2024-12-12 by bdring