bdring/FluidNC #1235: FluidDial Pendant no read sdcard and Macros don't load

sirias52, 2024-06-09:
### Wiki Search Terms

If you get a "Reading macros" message displayed, but the macro never load, you probably have a missing or incorrect file on the FluidNC ESP32. You need a file called macrocfg.json if you are using WebUI 2 or preferences.json if you are using WebUI 3. Use command $Localfs/list to see the contents of the ESP32 file system. Send $localfs/show=\<filename\> to see the contents of the file ( I have correctly the files  macrocfg.json)

### Controller Board

6-pack

### Machine Description

Gantry router

### Configuration file

```yaml
board: 6x CNC Controller
name: 6x Default
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 10
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 125.830
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.2:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 125.830
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.26:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  z:
    steps_per_mm: 99.559
    max_rate_mm_per_min: 700.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low    
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8
        
  a:
    steps_per_mm: 53.400
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  b:
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16

  c:      
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  
probe:
  pin: gpio.39:low
  toolsetter_pin: gpio.36:low
  
# Using MOSFETs (Check Spindle Pin Usage
coolant:
  flood_pin: gpio.12
  mist_pin: gpio.4
  delay_ms: 0

start:
  must_home: false
  
## Begin Huanyang  
# uart1:
  # txd_pin: gpio.15
  # rxd_pin: gpio.16
  # rts_pin: gpio.14
  # baud: 9600
  # mode: 8N1

uart1:
  txd_pin: gpio.25
  rxd_pin: gpio.27
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

uart_channel1:
  report_interval_ms: 75
  uart_num: 1

macros:

  macro0: G0X0Y0

# Huanyang:
  # uart_num: 1
  # modbus_id: 1
  # tool_num: 0
  # speed_map: 0=0% 0=25% 6000=25% 24000=100%
  # off_on_alarm: false

# #begin PWM
pwm:
  pwm_hz: 5000
  direction_pin: gpio.15
  output_pin: gpio.13
  enable_pin: gpio.14
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 10000=100.000%
  off_on_alarm: false
  
# #begin Laser
# Laser:
  # pwm_hz: 5000
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true
  
# #begin 10V
# 10V:
  # forward_pin: gpio.15
  # reverse_pin: gpio.14
  # pwm_hz: 5000
  # output_pin: gpio.13
  # enable_pin: NO_PIN
  # direction_pin: NO_PIN
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0  
  # tool_num: 0
  # speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  # off_on_alarm: false
```

### Startup Messages

```text
$G
$/axes/x/max_travel_mm
$/axes/x/homing/mpos_mm
$/axes/x/homing/positive_direction
$/axes/y/max_travel_mm
$/axes/y/homing/mpos_mm
$/axes/y/homing/positive_direction
$/axes/x/max_travel_mm=300.000
ok
$/axes/x/homing/mpos_mm=150.000
ok
$/axes/x/homing/positive_direction=false
ok
$/axes/y/max_travel_mm=300.000
ok
$/axes/y/homing/mpos_mm=150.000
ok
$/axes/y/homing/positive_direction=true
ok
```

### User Interface Software

candle, Webui.

### What happened?

No read sdcard and macros

### Other Information

I have a macro configured and it works correctly. I have a 4gb sdcard and the webui reads correctly. fluid pendant works perfectly  but without reading macros and sdcard

MitchBradley, 2024-06-09:
We need the startup messages that you can get by sending $SS.  You posted the output from $G, not the output from $SS

sirias52, 2024-06-09:
I already made the correction. excuse the mistake.

MitchBradley, 2024-06-09:
What is the output from $SD/List

sirias52, 2024-06-09:
$SD/List
[FILE: Perfilado 5.tap|SIZE:12262]
[FILE: 50.tap|SIZE:5969]
[DIR:sd]
[FILE:  50.tap|SIZE:5969]
[FILE:  arm 04 +2.gcode|SIZE:3662061]
[FILE:  arm 06.gcode|SIZE:374159]
[FILE:  Base arm^arm kuka (1).gcode|SIZE:6715166]
[FILE:  base mahole.gcode|SIZE:4119181]
[FILE:  Body1 +1.gcode|SIZE:1904231]
[FILE:  c1.tap|SIZE:13900]
[FILE:  casebottomeliot.gcode|SIZE:2975751]
[FILE:  caseupeliot.gcode|SIZE:6334614]
[FILE:  FluidDialBottom.gcode|SIZE:7339556]
[FILE:  FluidDialTop.gcode|SIZE:1914589]
[FILE:  macro1.g|SIZE:6]
[FILE:  Perfilado 1.tap|SIZE:12260]
[FILE:  Perfilado 4.tap|SIZE:12682]
[FILE:  Perfilado 50.tap|SIZE:12261]
[FILE:  tapa manhole.gcode|SIZE:2257737]
[FILE:  two.gcode|SIZE:1489159]
[FILE:  visel.gcode|SIZE:21615652]
[/sd/ Free:3.61 GB Used:58.01 MB Total:3.67 GB]
ok

MitchBradley, 2024-06-09:
What does the About screen on the FluidDial say?

sirias52, 2024-06-09:
[image]

MitchBradley, 2024-06-09:
Since you are compiling from source that was downloaded without using git, I cannot tell which version of the code you are using.  Please use git clone to get the source code.

sirias52, 2024-06-09:
I installed this version an hour ago.   https://github.com/bdring/FluidDial, I used vscode + platformio and upload the program. I'm going to try the git thing

sirias52, 2024-06-09:
[image]

MitchBradley, 2024-06-09:
[image]

sirias52, 2024-06-09:
The version is already in the previous message

sirias52, 2024-06-15:
the same problem. update the firmware.
[image]

sirias52, 2024-07-15:
Update the fluidcnc firmware , problem solved. Thank you.

closed 2024-07-15 by MitchBradley