bdring/FluidNC #1241: Problem: Homing only turns on 1 axis, and that starts to block and whine

Gwebber-Net, 2024-06-14:
### Wiki Search Terms

Homing does not turn axes

### Controller Board

MKS TINYBEE

### Machine Description

Custom built CNC for education.
Cartesian style model 
KM3 is the designation 

### Configuration file

```yaml
board: MKS TinyBee V1.0_001
name: K40 Laser
meta: (01.02.2022) by Skorpi
kinematics: {}
stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  x:
    steps_per_mm: 525
    max_rate_mm_per_min: 10000
    acceleration_mm_per_sec2: 1500
    max_travel_mm: 325
    soft_limits: false
    motor0:
      limit_neg_pin: gpio.33:pu:low
      hard_limits: true
      pulloff_mm: 2
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      stepstick:
        step_pin: i2so.1
        direction_pin: i2so.2:low
        disable_pin: i2so.0
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
    homing:
      cycle: -1
  y:
    steps_per_mm: 525
    max_rate_mm_per_min: 12000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 220
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 200
      seek_mm_per_min: 5000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.32:pu:low
      hard_limits: false
      pulloff_mm: 2
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      stepstick:
        step_pin: i2so.4
        direction_pin: i2so.5
        disable_pin: i2so.3
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
  z:
    steps_per_mm: 525
    max_rate_mm_per_min: 12000
    acceleration_mm_per_sec2: 500
    max_travel_mm: 80
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: true
      mpos_mm: 0
      feed_mm_per_min: 300
      seek_mm_per_min: 1000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.22:pu:low
      hard_limits: false
      pulloff_mm: 1
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      stepstick:
        step_pin: i2so.7
        direction_pin: i2so.8
        disable_pin: i2so.6
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.4:pu:low
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
macros:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
Laser:
  pwm_hz: 5000
  output_pin: gpio.13
  enable_pin: I2SO.16
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
start:
  must_home: false
  deactivate_parking: false
```

### Startup Messages

```text
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.7.17 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config_custom.yaml]
[MSG:INFO: Machine K40 Laser]
[MSG:INFO: Board MKS TinyBee V1.0_001]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-325.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.33:low:pu]
[MSG:INFO: Axis Y (0.000,220.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:  Y Neg Limit gpio.32:low:pu]
[MSG:INFO: Axis Z (-80.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:  Z Neg Limit gpio.22:low:pu]
[MSG:INFO: macro0_pin gpio.4:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:I2SO.16 Out:gpio.13 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:Fix Trade 24]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.2.118]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.17 (wifi) '$' for help]
ok
[VER:3.7 FluidNC v3.7.17:]
[OPT:PHS]
[MSG:Machine: K40 Laser]
[MSG:Mode=STA:SSID=Fix Trade 24:Status=Connected:IP=192.168.2.118:MAC=80-64-6F-31-F9-5C]
ok

Grbl 3.7 [FluidNC v3.7.17 (wifi) '$' for help]
```

### User Interface Software

Web

### What happened?

I am trying to home my axis

I have set up the directions correctly.

When I home Y the motor wants to spin, but it blocks for some reason, and starts to whine

X and Z don't move at ALL when pressing home button

In web UI terminal, I get te
 message : 
$HX
ok 

but the axis has not turned, and the limit switch has not been triggered.

bdring, 2024-06-14:
See this wiki page.

http://wiki.fluidnc.com/en/support/setup/motion

Gwebber-Net, 2024-06-14:
> See this wiki page.
> 
> http://wiki.fluidnc.com/en/support/setup/motion

None of my issues are described there

So for some magic reason, Homing X and Z now goes to the home switch.

Y motor still locks and whines when pressing home button
But I am able to JOG the Y motor forward and backwards without any issues or wrong noises etc.

So That issue is still present

bdring, 2024-06-14:
Funny noise is typically a wiring problem.

Gwebber-Net, 2024-06-14:
So why can I jog it normally, but not home it ?

MitchBradley, 2024-06-14:
You didn't tell us either the jog speed or the jog distance so we cannot determine how fast it was trying to go during the jog. 
Nor did you tell us anything useful about the machine - "custom built" and "KM3" are not descriptive of the motors, the stepper drivers, the mechanic structure or the coupling between motors and mechanical parts.  Lacking such information, it is impossible to know whether your configuration parameters are appropriate for the system.

I can say that the 525 steps/mm against 12000 mm/min is very aggressive.  That works out to 105000 steps/sec which is quite close to the theoretical limit for I2S stepping, especially in I2S_STATIC mode.  The acceleration is on the high side too.

Please provide more comprehensive, detailed, and unambiguous information.  For example, when you say that the User Interface is "Web", we don't know if that means WebUI version 2, WebUI version 3, or some other web-based user interface like CNCjs or gSender.  When you say "pressing home button", we don't know exactly which button on which user interface, since most UIs have several such buttons for homing different combinations of axes.  Similarly, when you say "I am able to jog", that does not tell us exactly which button you used (or which command in the case of a manually issued jog command via a console window) nor what the settings were.

Gwebber-Net, 2024-06-15:
Ok so I will change the 12k to something like 5k

Im using the ESP WEB UI that comes default from the installer

The acceleration you are talking about, is that on X or Y ?

The only downside t your statement, is that

closed 2024-06-15 by Gwebber-Net

MitchBradley, 2024-06-15:
> The acceleration you are talking about, is that on X or Y ?

The acceleration values could be high on either of those axes.  Since I know almost nothing about the machine it is hard to say.  I do not know the type of the mechanism (gantry or mill or whatever), moving mass per axis, drive type (belt, lead screw, ball screw, rack and pinion, ...),  gearing ratios,  motor size, driver type, driver current, microstepping setting, inherent friction (affected by alignment, lubrication, etc), ...  All of those can affect the achievable speeds and accelerations.

> The only downside to your statement, is that homing is done at a significant lower speed then jogging

Maybe, but since you have not told us the jog rate settings in WebUI, I have no way of verifying that.

[image]

WebUI v2 defaults to 1000 mm/min for the jog rate on X and Y.  You have the Y homing seek rate set to 5000, which is not "significantly slower" than 1000, but rather 5x faster.

Gwebber-Net, 2024-06-15:
Ok so I just downloaded the base config,   apparently that has some faults in it

What would you change in my current config in order for it to probably work correctly ?

MitchBradley, 2024-06-15:
I am guessing that, by "the base config", you mean [this Tinybee example](https://github.com/bdring/fluidnc-config-files/blob/main/official/MKS_TinyBee_1_XYZAB.yaml).  A config file is specific to not only the controller board but also to the machine.  The "faults" in that file reflect differences between your machine and whatever machine the config was written for.  Based on the commentary in that file, it seems to be for a K40 laser machine.  Even if you had a K40 laser, you would no doubt need to change some things, because there are countless variants of the generic "K40 laser" product,

Perhaps your "Custom built CNC for education.", "KM3 is the designation" is like a K40 laser, but it seems unlikely.

In order to tell you the specific mods you would need to make, someone would need to know pretty much every detail of your machine, including but not limited to all of those things I mentioned in a previous comment.  We do not offer free consulting services for creating config files for random machines.  The FluidNC Wiki has extensive documentation about how the config file items apply to different situations.

MitchBradley, 2024-06-15:
[image]

Gwebber-Net, 2024-06-16:
I have experience with klipper for 3d printers.

I already calibrated the Steps/mm for the stepper motors of my machine.

I will adjust the settings to be very safe, and report back if it made the Y axis function properly

Thanks