bdring/FluidNC #1242: Problem: Trying to Run JTAG debugging

MiguelASTavares, 2024-06-18:
### Wiki Search Terms

JTAG; Debug

### Controller Board

esp32

### Machine Description

Dual axis

### Configuration file

```yaml
board: TMC2209 Pen
name: TMC2209 Pen
meta: "update for 3.6.8"

stepping:
  engine: RMT
  idle_ms: 250
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0
  
uart1:
  txd_pin: gpio.22
  rxd_pin: gpio.21
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1

axes:
  shared_stepper_disable_pin: gpio.25:low
  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 30000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 6.000
      tmc_2300:
        uart_num: 1
        addr: 0
        r_sense_ohms: 0.110
        run_amps: 1.00
        hold_amps: 0.500
        microsteps: 64
        stallguard: 0
        stallguard_debug: true
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
        use_enable: false
        step_pin: gpio.26
        direction_pin: gpio.27
        disable_pin: NO_PIN

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

  
start:
  check_limits: true
```

### Startup Messages

```text
N
```

### What happened?

Can't upon the debug terminal in Plataformio inside VScode

### Other Information

I am working on adding support for the TMC2300 stepper driver into FluidNC, aiming to create a system that can be battery operated. I have made changes to several files to incorporate this support. However, I am encountering issues with performing certain operations, such as sensorless homing.

To debug these problems, I wanted to use JTAG. Unfortunately, I am unable to open a debug terminal. My suspicion is that some or all of the JTAG pins might be in use within the existing codebase.

Questions:

1. Are any JTAG pins being utilized in the current implementation that could prevent opening a debug terminal?
2. If so, is there a way to enable JTAG debugging without conflicting with the current pin assignments?
3. Are there alternative debugging options available for this situation?

bdring, 2024-06-18:
Early version of the 6 Pack controller had a JTAG connector. It worked fine, but the ESP32 JTAG is quite limited with few breakpoints and is slow. We have found that serial port debugging and backtraces are much more useful.

look at `log_info` , `log_error`, etc statements.

[image]

bdring, 2024-06-18:
You only need to add the `log_...` statement. Be sure your `$Message/Level` setting is set the the appropriate level. The messages are sent to uart0. This is typically the USB connection.

If you are seeing the startup messages, you should see yours. They use `log_info`

MitchBradley, 2024-06-18:
I gave JTAG a very good try.  I got it to basically work, but it was very difficult to use and I eventually gave up.  The ESP32 has only 2 (3?) hardware breakpoints which is not enough for any serious use.  Once they are assigned, the debugger starts using software breakpoints which, except in the rare cases where the code is in RAM, requires a complex operation to rewrite an entire sector of FLASH.  Then if the program crashes and reboots, the FLASH has already been changed and the debugger does a poor job of cleaning it up.  The result is that you spend all of your attention trying to figure out what is going wrong with the debugger instead of thinking about the problem.

MitchBradley, 2024-06-18:
log_info() is the easiest serial debug method.  In the few instances when it causes problems - like when you are trying to debug some part of the logging or channel infrastructure - you can go directly to Uart 0.  To do that, you can #include "src/UartChannel", then write a stream IO expression like `Uart0 << "The value is " << some_variable << "\n";``

MiguelASTavares, 2024-06-19:
Thank you for your help. I have the debugging via the log_... call working. It already proved useful.

closed 2024-06-19 by MiguelASTavares