bdring/FluidNC #1246: Problem: Homing and TMC2209 Driver Response

MiguelASTavares, 2024-07-01:
### Wiki Search Terms

UART; Homing

### Controller Board

Single esp32 with 2 tmc2209

### Machine Description

Simple machine with two axis controlled by a TMC2209

### Configuration file

```yaml
board: TMC2209 Pen
name: TMC2209 Pen
meta: "update for 3.6.8"

stepping:
  engine: RMT
  idle_ms: 250
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0
  
uart1:
  txd_pin: gpio.22
  rxd_pin: gpio.21
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1

axes:
  shared_stepper_disable_pin: gpio.13:low
  y:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 30000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 30000.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 1500.000
      feed_mm_per_min: 4000.000
      seek_mm_per_min: 6000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.19:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 150.000
      tmc_2209:
        uart_num: 1
        addr: 3 
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 128
        stallguard: 0
        stallguard_debug: true
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
        use_enable: false
        step_pin: gpio.26
        direction_pin: gpio.15
        disable_pin: NO_PIN  

  x:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 30000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 30000.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 1500.000
      feed_mm_per_min: 4000.000
      seek_mm_per_min: 6000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.23:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 150.000
      tmc_2209:
        uart_num: 1
        addr: 0
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 128
        stallguard: 0
        stallguard_debug: true
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
        use_enable: false
        step_pin: gpio.27
        direction_pin: gpio.2
        disable_pin: NO_PIN
```

### Startup Messages

```text
[MSG:INFO: Caution: Unlocked]
[MSG:DBG: Homing Cycle X]
[MSG:INFO: Setting registers with isHoming: 1]
[MSG:INFO: Selected mode: 2]
[MSG:INFO: Run current (mA): 500]
[MSG:INFO: RMS current and hold percentage set.]
[MSG:INFO: Microsteps set to: 128]
[MSG:INFO: X Axis StallGuard mode selected.]
[MSG:INFO: Homing feed rate: 4000.000]
[MSG:INFO: Tstep calculation: 892]
[MSG:INFO: StallGuard configuration applied.]
...
[MSG:INFO: Axis not moving, tstep indicates idle.]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Homing limited X]
[MSG:INFO: Setting disable to: 0]
[MSG:INFO: Setting disable to: 0]
[MSG:DBG:  X Neg Limit 0]
...
[MSG:DBG: Homing nextPhase SlowApproach]
...
[MSG:INFO: Generating debug message...]
[MSG:INFO: Axis not moving, tstep indicates idle.]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Homing limited X]
[MSG:INFO: Setting disable to: 0]
[MSG:INFO: Setting disable to: 0]
[MSG:DBG:  X Neg Limit 0]
...
[MSG:DBG: Starting from -230.469,0.000,0.000]
[MSG:DBG: Planned move to -80.469,0.000,0.000 @ 4000.000]
[MSG:DBG: protocol_initiate_homing_cycle Home]
...
[MSG:INFO: X Axis SG_Val: 10 Rate: 100.017 mm/min SG_Setting: 0]
[MSG:INFO: Generating debug message...]
[MSG:INFO: Axis not moving, tstep indicates idle.]
[MSG:INFO: Setting disable to: 0]
[MSG:INFO: Setting disable to: 0]
[MSG:DBG: CycleStop Pulloff1]
...
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was -80.469,0.000,0.000]
[MSG:Homed:X]
[MSG:DBG: mpos becomes 1500.000,0.000,0.000]
[MSG:DBG: mpos transformed 1500.000,0.000,0.000]
[MSG:INFO: Setting registers with isHoming: 0]
[MSG:INFO: Selected mode: 0]
[MSG:INFO: Run current (mA): 1000]
[MSG:INFO: RMS current and hold percentage set.]
[MSG:INFO: Microsteps set to: 128]
[MSG:INFO: X Axis StealthChop mode selected.]
[MSG:INFO: StealthChop configuration applied.]
[MSG:DBG: Homing done]
[MSG:INFO: Setting disable to: 1]
[MSG:INFO: StartDisable called with disable: 1]
[MSG:INFO: Setting disable to: 1]
[MSG:INFO: StartDisable called with disable: 1]
[MSG:DBG: Homing Cycle Y]
[MSG:INFO: Setting registers with isHoming: 1]
[MSG:INFO: Selected mode: 2]
[MSG:INFO: Run current (mA): 500]
[MSG:INFO: RMS current and hold percentage set.]
[MSG:INFO: Microsteps set to: 128]
[MSG:INFO: Y Axis StallGuard mode selected.]
...
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Homing limited Y]
[MSG:INFO: Setting disable to: 0]
[MSG:INFO: Setting disable to: 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: Homing nextPhase Pulloff0]
...
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was 1500.000,-416.456,0.000]
[MSG:Homed:Y]
[MSG:DBG: mpos becomes 1500.000,1500.000,0.000]
[MSG:DBG: mpos transformed 1500.000,1500.000,0.000]
[MSG:INFO: Setting registers with isHoming: 0]
[MSG:INFO: Selected mode: 0]
[MSG:INFO: Run current (mA): 1000]
```

### What happened?

## Homing Process:

- The system initiates the homing cycle for both X and Y axes.
- Occasionally, the homing cycle completes successfully for both axes.
- At other times, the homing cycle completes for only one axis, failing to home the other.

## Driver Response Post-Homing:

- After the homing cycle (regardless of success or failure), the TMC2209 drivers become unresponsive.
- The drivers appear to go down, ceasing to respond to subsequent commands.

## Web Interface Interaction:

- Despite the TMC2209 drivers becoming unresponsive, the web interface remains operational.
- Users can still interact with the system through the web interface.

## Command Visibility:

- Commands sent through the web interface terminal (e.g.,` $J=G91 G21 F1000 X100`) are visible in the terminal log.
- However, these commands are not visible in the UART monitoring tools such as PlatformIO.

##File Upload Behavior:

- When uploading files via the web interface, changes are detectable in the UART monitoring.
- Example log entries:

```
[MSG:DBG: pathname /spiffs/TMC2209_pen_SG.yaml]
[MSG:Files changed]
```

bdring, 2024-07-01:
That is not the startup messages. See the wiki on how to show the messages.

MiguelASTavares, 2024-07-01:
Sorry, this are the startup messages:

[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.0.x (main-f996d68d-dirty) https://github.com/MiguelASTavares/FluidNC.git]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:TMC2209_pen_SG.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine TMC2209 Pen]
[MSG:INFO: Board TMC2209 Pen]
[MSG:INFO: UART1 Tx:gpio.22 Rx:gpio.21 RTS:NO_PIN Baud:115200]
[MSG:DBG: SPI not defined]
[MSG:DBG: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13:low]
[MSG:INFO: Axis X (1500.000,31500.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Initializing TMC2209Driver...]
[MSG:INFO: TMC2209Stepper object created with UART: 0x3ffb3dc4, R_sense: 0.110, Addr: 0]
[MSG:DBG: TMCStepper Library Ver. 0x703]
[MSG:INFO:     tmc_2209 UART1 Addr:0 CS:NO_PIN Step:gpio.27 Dir:gpio.2 Disable:NO_PIN R:0.110]
[MSG:INFO: TMC2209Driver initialized and registered.]
[MSG:INFO:  X Neg Limit gpio.23]
[MSG:INFO: Axis Y (1500.000,31500.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Initializing TMC2209Driver...]
[MSG:INFO: TMC2209Stepper object created with UART: 0x3ffb3dc4, R_sense: 0.110, Addr: 3]
[MSG:INFO:     tmc_2209 UART1 Addr:3 CS:NO_PIN Step:gpio.26 Dir:gpio.15 Disable:NO_PIN R:0.110]
[MSG:INFO: TMC2209Driver initialized and registered.]
[MSG:INFO:  Y Neg Limit gpio.19]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Configuring motor...]
[MSG:INFO: TMC2209Stepper begin() called.]
[MSG:INFO: Testing TMC2209Driver version...]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Version check passed.]
[MSG:INFO: IFCNT before: 31]
[MSG:INFO: IFCNT after: 32]
[MSG:INFO: Communication test passed.]
[MSG:INFO: Setting registers with isHoming: 0]
[MSG:INFO: Selected mode: 0]
[MSG:INFO: Run current (mA): 1000]
[MSG:INFO: RMS current and hold percentage set.]
[MSG:INFO: Microsteps set to: 128]
[MSG:INFO: X Axis StealthChop mode selected.]
[MSG:INFO: StealthChop configuration applied.]
[MSG:INFO: Base motor configuration completed.]
[MSG:INFO: Configuring motor...]
[MSG:INFO: TMC2209Stepper begin() called.]
[MSG:INFO: Testing TMC2209Driver version...]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Version check passed.]
[MSG:INFO: IFCNT before: 31]
[MSG:INFO: IFCNT after: 32]
[MSG:INFO: Communication test passed.]
[MSG:INFO: Setting registers with isHoming: 0]
[MSG:INFO: Selected mode: 0]
[MSG:INFO: Run current (mA): 1000]
[MSG:INFO: RMS current and hold percentage set.]
[MSG:INFO: Microsteps set to: 128]
[MSG:INFO: Y Axis StealthChop mode selected.]
[MSG:INFO: StealthChop configuration applied.]
[MSG:INFO: Base motor configuration completed.]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]

closed 2025-01-03 by bdring