bdring/FluidNC #1257: PWM and RS485 issues

muellercandd, 2024-07-12:
### Wiki Search Terms

Set up config with laser and spindle

### Controller Board

XPro V5

### Machine Description

Workbee dual y with spindle rs458

### Configuration file

```yaml
board: XPro V5
name: xPro V5 
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 200
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 1250.00
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 00.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.12
        direction_pin: gpio.14:low
        use_enable: true
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  y:
    steps_per_mm: 200.00
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 1230.00
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 00.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.27
        direction_pin: gpio.26
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 2
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.5000
      tmc_5160:
        step_pin: gpio.33
        direction_pin: gpio.32
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  z:
    steps_per_mm: 200.00
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 100.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 00.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.15
        direction_pin: gpio.2:low
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  frequency_hz: 10000000

probe:
  pin: gpio.22:low
  check_mode_start: true

control:
  safety_door_pin: gpio.16:low
  macro0_pin: gpio.13:low
  macro1_pin: gpio.0:low

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

coolant:
  mist_pin: gpio.21
  delay_ms: 0
 

pwm:
 pwm_hz: 10000
 direction_pin: NO_PIN
 output_pin: gpio.25
 enable_pin: gpio.4
 disable_with_s0: false
 s0_with_disable: true
 spinup_ms: 0
 spindown_ms: 0
 tool_num: 0
 speed_map: 0=0.000% 100=100.000%

uart1:
  txd_pin: gpio.4
  rxd_pin: gpio.25
  rts_pin: NO_PIN
  baud: 9600
  mode: 8N1

Huanyang:
  Uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  off_on_alarm: false
```

### Startup Messages

```text
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4

[MSG:INFO: FluidNC v3.7.6]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:ERR: ERR: gpio.4 - Pin is already used.]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008d044  PS      : 0x00060230  A0      : 0x801d3aa8  A1      : 0x3ffb18f0
A2      : 0x00000006  A3      : 0x00000004  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x002e0455  A9      : 0x3ffb1bc0
A10     : 0x00000003  A11     : 0x00060223  A12     : 0x00060220  A13     : 0x00000014
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000017  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x4008d059  LEND    : 0x4008d069  LCOUNT  : 0xffffffff

Backtrace: 0x4008d041:0x3ffb18f0 0x401d3aa5:0x3ffb1900 0x401d07a1:0x3ffb1c10 0x400ed46a:0x3ffb1cd0 0x400d6e66:0x3ffb1e20 0x400e2065:0x3ffb1e50 0x400fb07b:0x3ffb1e80 0x400fb0e1:0x3ffb1ea0 0x400e32da:0x3ffb1ec0 0x400e21e1:0x3ffb1f10 0x400e2dc4:0x3ffb1f30 0x400e32da:0x3ffb1f60 0x400e3805:0x3ffb1fb0 0x400e3dcd:0x3ffb2030 0x400e3f31:0x3ffb2230 0x400e533c:0x3ffb2260 0x4011af82:0x3ffb2290

ELF file SHA256: 1d1d266e85df2483

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4

[MSG:INFO: FluidNC v3.7.6]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### What happened?

I am trying to set up a PWM laser to my already working system using RS458 and spindle. I though maybe I could just add the pwm info to my config and switch the switch from rs458 to pwm, but the config file does not seem play nice. An issue with GPIO4. I am wondering if it is possible to have both set up at the same time and switch back and forth. I really do not want to load a new config each time I switch between devices. Any help is appreciated.

Skorpi08, 2024-07-29:
It's not possible, you should load new config.

MitchBradley, 2024-07-30:
You can have two config files on the machine at once , with different file names like laser.yaml and vfd.yaml. Then you can choose the one you want with $config/filename=vfd.yaml

MitchBradley, 2024-07-30:
Extra credit: set up two macros, one with **$config/filename=laser.yaml**  followed  by   **$bye**  and the other, similarly, with vfd.yaml   Then execute the macro to change the config file then reboot.

closed 2025-01-03 by bdring