bdring/FluidNC #1261: upload any file via web and X move 1mm

sirias52, 2024-07-14:
### Wiki Search Terms

ninguno.

### Controller Board

6-pack

### Machine Description

Gantry router

### Configuration file

```yaml
board: 6x CNC Controller
name: 6x Default
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 10
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 125.830
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.2:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 125.830
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.26:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  z:
    steps_per_mm: 99.559
    max_rate_mm_per_min: 700.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low    
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8
        
  a:
    steps_per_mm: 53.400
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  b:
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16

  c:      
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  
probe:
  pin: gpio.39:low
  toolsetter_pin: gpio.36:low
  
# Using MOSFETs (Check Spindle Pin Usage
coolant:
  flood_pin: gpio.12
  mist_pin: gpio.4
  delay_ms: 0

start:
  must_home: false
  
## Begin Huanyang  
# uart1:
  # txd_pin: gpio.15
  # rxd_pin: gpio.16
  # rts_pin: gpio.14
  # baud: 9600
  # mode: 8N1

uart1:
  txd_pin: gpio.25
  rxd_pin: gpio.27
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

uart_channel1:
  report_interval_ms: 75
  uart_num: 1

macros:

  macro0: G0X0Y0

# Huanyang:
  # uart_num: 1
  # modbus_id: 1
  # tool_num: 0
  # speed_map: 0=0% 0=25% 6000=25% 24000=100%
  # off_on_alarm: false

# #begin PWM
pwm:
  pwm_hz: 5000
  direction_pin: gpio.15
  output_pin: gpio.13
  enable_pin: gpio.14
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 10000=100.000%
  off_on_alarm: false
  
# #begin Laser
# Laser:
  # pwm_hz: 5000
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true
  
# #begin 10V
# 10V:
  # forward_pin: gpio.15
  # reverse_pin: gpio.14
  # pwm_hz: 5000
  # output_pin: gpio.13
  # enable_pin: NO_PIN
  # direction_pin: NO_PIN
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0  
  # tool_num: 0
  # speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  # off_on_alarm: false
```

### Startup Messages

```text
$SS
[MSG:INFO: FluidNC v3.7.12 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6x Default]
[MSG:INFO: Board 6x CNC Controller]
[MSG:INFO: UART1 Tx:gpio.25 Rx:gpio.27 RTS:NO_PIN Baud:1000000]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:10us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 6]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.2:low:pu]
[MSG:INFO: Axis Y (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y Neg Limit gpio.26:low:pu]
[MSG:INFO: Axis Z (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:  Z Neg Limit gpio.33:low]
[MSG:INFO: Axis A (150.000,1110.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO:  A Neg Limit gpio.32:low]
[MSG:INFO: Axis B (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]
[MSG:INFO:  B Neg Limit gpio.35:low]
[MSG:INFO: Axis C (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.21 Dir:I2SO.20 Disable:I2SO.23]
[MSG:INFO:  C Neg Limit gpio.34:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.14 Out:gpio.13 Dir:gpio.15 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant gpio.12]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe Pin: gpio.39:low]
[MSG:INFO: Connecting to STA SSID:TIGO-B240]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.0.6]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

Fluidpendant, Webui

### What happened?

upload any file via web and X move 1mm

[MSG:INFO: Program End]  (file over and return to origin)
<Idle|MPos:-15.998,9.799,9.412,0.000,0.000,0.000|FS:0,0|WCO:-15.995,9.800,-0.592,0.000,0.000,0.000|SD:100.00,/sd/5.tap>
[GC:G1 G54 G17 G21 G90 G94 M5 M9 T1 F480 S16000]
[MSG: /sd/5.tap file job succeeded]  (Upload file with  machine stopped)
$J=G91 G21 F2000 X1                            (here add jog command and X1, WHY?)
ok
<Jog|MPos:-15.990,9.799,9.412,0.000,0.000,0.000|FS:258,0>
<Idle|MPos:-14.996,9.799,9.412,0.000,0.000,0.000|FS:0,0>
<Idle|MPos:-14.996,9.799,9.412,0.000,0.000,0.000|FS:0,0>
[MSG:INFO: Channel auto report interval set to 200 ms]

MitchBradley, 2024-07-14:
You are 6 release behind.  The current release is 3.7.18, and we are close to releasing 3.8.0.  There is a good chance that we have already fixed the problem.  Please do not submit problem reports against ancient releases.

sirias52, 2024-07-15:
Update the firmware, problem solved. Thank you.

sirias52, 2024-07-17:
I have been update the firmware to the last version (3.7.18) , but the problem still exist and I still found ramdon movement mean while I upload the file via webUI. I´m Use fluidpendant to control cnc.

$G
$J=G91 G21 F2000 Y-100 (in this case Y axis move -100)
[MSG:Files changed]
ok
<Jog|MPos:2.209,13.192,-1.708,0.000,0.000,0.000|FS:1560,0>

bdring, 2024-07-17:
- Is it possible the pendant is sending the jog commands?
- Is the pendant the FluidDial?
- Is the pendant on a jog screen when this happens. If so, does the jog amount match the increment on the pendant?

sirias52, 2024-08-23:
Is it possible the pendant is sending the jog commands?

is possible, but only upload any file via webUi

Is the pendant the FluidDial?

Yes

Is the pendant on a jog screen when this happens. If so, does the jog amount match the increment on the pendant?

yes, upload file, and fluid pendant usually stay on the screen jog. yes match whit increment value.

despite having updated to version 3.8.1 the problem persists

https://github.com/user-attachments/assets/10f3a21b-5a3c-45a6-90ba-6096fb626e35

MitchBradley, 2024-08-23:
> yes, upload file, and fluid pendant usually stay on the screen jog. yes match whit increment value.

That answer does not agree with your video.  In your video, the jog increment is 0.1, since the tenths digit is highlighted, but the jog is much longer.

[image]

The jog settings in WebUI are impossible to determine from your video which moves around so much that we cannot see the job settings.

The job command listed above, i.e. `$J=G91 G21 F2000 Y-100 (in this case Y axis move -100)` is not in the form that FluidDial sends, because FluidDial sends without spaces between G91, G21, and the F value.

Which version of WebUI are you using, and where did you get it?  This capture from the video makes me think that it is not built from our version of the WebUI source code, since our version does not have a TFT-SD button

[image]

The only version of WebUI v2 that we support is our fork at https://github.com/MitchBradley/ESP3D-WEBUI/tree/revamp

sirias52, 2024-08-23:
(That answer does not agree with your video. In your video, the jog increment is 0.1, since the tenths digit is highlighted, but the jog is much longer)

The video was an unusual issue, but usually it sends a random command ($J=G91 G21 F2000 Y-100, $J=G91 G21 F2000 Y-10, $J=G91 G21 F2000 X10, etc. integer values between axis X,Y.) That day the machine did several jobs properly, but at that moment it generated that problem, very strange. because it only generated those random commands.

Which version of WebUI are you using

[image]

our version does not have a TFT-SD button

[image]
 this button is optional.

MitchBradley, 2024-08-23:
Does it fail every time you upload a file, or only sometimes?

Does it fail if the pendant is not connected?

sirias52, 2024-08-23:
Does it fail every time you upload a file, or only sometimes?

sometimes. I've tried to observe the pattern, but it's very random to be able to explain to you what is happening

Does it fail if the pendant is not connected?

I haven't tried it, I only use the pendant because it's very useful to me.

MitchBradley, 2024-08-23:
Problems that do not happen consistently are very difficult to fix, especially if the developers never experience the problem.

Right now we do not know whether the problem is related to the use of the pendant or not, which makes the search space twice as large.

sirias52, 2024-08-23:
I think I can survive with that, mean while i hope this not fail machining parts on cnc. Thank you.

MitchBradley, 2024-08-23:
If you can find a way to make it fail consistently, so we can reproduce the problem, we can probably solve it.

sirias52, 2024-08-23:
I will be waiting to give you feedback. I use the CNC very often, I'm sure I'll find a way to make it fail consistently. Thank you.

sirias52, 2024-08-31:
Files are loaded onto the SD card. For example, I have 5 files loaded, I delete all the files to insert a new one, then the error occurs. The error pattern is in those file deletion cycles.

[MSG:Files changed] "file deleted"
[MSG:Files changed] "file deleted"
[MSG:Files changed] "file deleted"
[MSG:Files changed] "file deleted"
[MSG:Files changed] "upload file"
$J=G91 G21 F2000 X10 "add Gcode comand"

always with the pendant.

MitchBradley, 2024-09-01:
I reproduced the problem.

MitchBradley, 2024-09-01:
Okay, this one is almost too weird to believe.  Here's what's happening.  When you hit the Upload button on WebUI, a file selector window appears on top of the WebUI window, covering the Jog panel.  When you select a file by double clicking, there is a fair chance that the mouse happens to be above the Jog control.  If you double-click and immediately release the mouse button, all is well.  But if you double-click and hold the mouse button for a time, the file selector window disappears and the mouse is sitting on top of the jog control circle with the button held down, thus choosing a jog.  When you finally release the mouse button, that jog command is executed.

I will have to try some things to see if I can suppress this behavior.  It could be tricky because those jog commands are attached to parts of an SVG image, rather than being controlled by javascript code.

Meanwhile, there are several ways to avoid the problem, with care:
1. Instead of double-clicking to select the upload file, single click and then hit the Open button
2. Move the file selector window away from the jog controller so the file that you select is not right on top of the jogger
3. Double-click and immediately release so the mouse is not depressed when the dialog disappers.

MitchBradley, 2024-09-01:
Changing the SVG image event triggers to use "onclick" instead of "onmouseup" seems to fix it.  Who knows what side effect that will have, thought.

commit a1c9be22 (2024-09-25): New index.html.gz to fix #1261
commit a60db824 (2024-09-25): Merge pull request #1305 from bdring/Fix1261

New index.html.gz to fix #1261
closed 2024-09-25 by bdring