bdring/FluidNC #1262: Problem: ESP32, WIFI on causes consumption jump from 0.07A to 0.13A and overhearts

aguaviva, 2024-07-15:
### Wiki Search Terms

heat, power consumption, esp32

### Controller Board

none, just the ESP32

### Machine Description

nothing

### Configuration file

```yaml
default
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.7.18 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:ERR: Cannot open configuration file:config.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.18 (wifi) '$' for help]
[MSG:ERR: Configuration is invalid. Check boot messages for ERR's.]
```

### User Interface Software

serial port

### What happened?

Esp32 consumes 
- 140mA with WIFI on
- 70mA with WIFI off

Note that HTTP and Telnet are turned off all the time.

Is this expected?

MitchBradley, 2024-07-15:
Ask Espressif to make a radio that can transmit without using power.

aguaviva, 2024-07-15:
> Ask Espressif to make a radio that can transmit without using power.

It shouldn't be transmitting anything right after flashing it and after turning HTTP & Telnet off.

Moreover I flashed CircuitPython (WIFI on + a webserver running) consumes 50mA, so there might be something in the firmware that is making consumption skyrocket.

MitchBradley, 2024-07-15:
It could be scanning.  I have no idea what mode CircuitPython is using.  See https://deepbluembedded.com/esp32-sleep-modes-power-consumption/ .  If you want to figure out what exactly is happening in your situation, you can modify WifiConfig.cpp to test various modes and setup steps.

MitchBradley, 2024-07-15:
Since it is in AP mode it is probably sending beacons.

aguaviva, 2024-07-15:
Circuitpython is also in AP mode and sending beacons, plus it has a webserver with websockets server running in the background. 

BTW my situation is just a bare ESP32 ( I tried two units and both showing the same heating/consumption issue with HTTP/Telnet off) so potentially more people might be seeing this same issue.

V1EngineeringInc, 2024-07-15:
> Ask Espressif to make a radio that can transmit without using power.

I have actually had two users both using genuine ESP32 modules in the last week ask why their esp's are now showing 80C temps. Might be worth taking a second look at. One of them has two esp's and only one shows the issue so I was not sure what to think about it but after reading this post I am wondering if it is getting hit a little harder than before?

aguaviva, 2024-07-15:
I'll be happy to help debug the issue and try instrumented builds that help narrow down the issue (but cant instrument the code myself as I am not familiar with it and would need some time to learn it.)

MitchBradley, 2024-07-15:
Mine is running at 88F not C in AP mode with the Test Drive config file - or with a valid one.  Barely above ambient.  http/enable and telnet/enable off or on, doesn't matter.  Same with 3.7.18 and 3.8.0

V1EngineeringInc, 2024-07-15:
Thanks for double-checking. 

I will keep at it on my end then.

MitchBradley, 2024-07-15:
I just did a full install on a bare ESP32 with an attached external antenna.  84F according to an infrared thermometer.  Unplugged the antenna - still cold.

The wifi startup code is in WebUI/WifiConfig.cpp in the StartAP() method.  If someone wants to comment out parts of it to see what happens, go wild.  I don't have time to add a bunch of debugging stuff and document it and build a test release and ...  Setting up a compilation environment is pretty easy.  Install VsCode and the PlatformIO extension, start a project by git cloning the FluidNC repo, and hit the upload icon at the bottom of the vscode window.

V1EngineeringInc, 2024-07-15:
I think your tests, and the test I have seen from my user that have two different ESP's acting differently, are telling me it is most likely some sort of hardware issue. Mine do not seem to get that hot and the ones I flash and ship have all seemed good for the short time they are on for testing.

aguaviva, 2024-07-15:
> Setting up a compilation environment is pretty easy. Install VsCode and the PlatformIO extension, start a project by git cloning the FluidNC repo, and hit the upload icon at the bottom of the vscode window.

That is the easy part, the hard part is to know what line to comment out.

Are your Esp32 consuming 50mA or 130mA?

MitchBradley, 2024-07-15:
130mA would be over 400 mW at 3.3V which would at least be warm to the touch.

The lines of code starting with WIFI. are the interesting ones.

aguaviva, 2024-07-15:
> I think your tests, and the test I have seen from my user that have two different ESP's acting differently, are telling me it is most likely some sort of hardware issue.

I am not sure it is a HW issue as Circuitpython works well when flashed.

MitchBradley, 2024-07-15:
The ESP32 hides some radio-related stuff in a section of FLASH.  It controls what the radio does right after startup.  Maybe there is something in there that is causing a problem.  You could try doing an erase and full install.  Perhaps CircuitPython left something lying around that plays badly with FluidNC, but works well with CircuitPython.

MitchBradley, 2024-07-16:
It might not be a "hardware" issue exactly, but it could be in some of the semi-magic Espressif bootloader and FLASH layout stuff.  That would fall into a gray area where it is not strictly hardware, but neither is it a problem in the FluidNC code per-se.  There is a lot going on behind the scenes with the Espressif startup code.  When you do a fresh install on a virgin ESP32, it tends to work right, to the extent that it works for hundreds or thousands of users without complaint.  One thing that is unusual about your chips is that they have experienced a CircuitPython installation.  I am not sure that is actually what is causing the problem, but it is certainly something that is not the norm.

aguaviva, 2024-07-16:
I did a fresh install of your firmware in a new Esp32, later is when I installed Circuitpython to determine whether this was a HW or SW issue.

If you could list a few lines to comment out that would help.

MitchBradley, 2024-07-16:
The lines that start with WIFI. as I said before.

aguaviva, 2024-07-16:
I tried disabling almost everything and can't get rid of the 130mA.

I was about to start thinking this is a HW issue when I decided to run https://github.com/espressif/arduino-esp32/blob/master/tests/validation/wifi/wifi.ino and got 2~4mA again.

MitchBradley, 2024-07-16:
Try a noradio build

MitchBradley, 2024-07-16:
What kind of ESP32 module is it?

aguaviva, 2024-07-16:
Sorry, my bad, I meant that as soon as I call `WiFi.mode(WIFI_STA);` I get the 130mA

MitchBradley, 2024-07-16:
If you set CORE_DEBUG_LEVEL to 5 in platformio.ini, you will get a lot of messages from the underlying stack.  Sometimes that provides a clue.

aguaviva, 2024-07-18:
I found the issue, it is this line https://github.com/MitchBradley/FluidNC/blob/7deec3327ff4339dae7e989729a34632de228b1d/FluidNC/src/WebUI/WifiConfig.cpp#L785  what is causing the esp32 to draw so much current

MitchBradley, 2024-07-18:
You should probably be looking at the code in the bdring FluidNC repo.  My fork is not maintained; it is only present for some old experiments.  Regardless, here are the possible values for that function:
```
typedef enum {
    WIFI_PS_NONE,        /**< No power save */
    WIFI_PS_MIN_MODEM,   /**< Minimum modem power saving. In this mode, station wakes up to receive beacon every DTIM period */
    WIFI_PS_MAX_MODEM,   /**< Maximum modem power saving. In this mode, interval to receive beacons is determined by the listen_interval parameter in wifi_sta_config_t */
} wifi_ps_type_t;
```
I don't know the performance impact of the various values.

aguaviva, 2024-07-18:
thanks, I was using bdring, I just happend to send your repo's link.

It would help if that particular line had a comment documenting why it is there and that it can cause overheating in some models.

bdring, 2024-07-18:
The line was probably set that way because we generally don't need power savings as the ESP32 is not run off batteries.

I have never seen overheating. We could possibly a new NVS setting like `$Wifi/Powermode`

MitchBradley, 2024-07-19:
> It would help if that particular line had a comment documenting ...

Great idea.  We should probably similarly explain all the history and implications of every one of the ~30K lines of code.  Assuming that it takes 5 minutes per line to research, recall, and explain each one clearly, that works out to about 2500 hours of programmer time.  Multiply that by the market rate for programmers of the caliber necessary to work on FluidNC, and then also calculate the opportunity cost of development and support of FluidNC coming to a grinding halt because the only people capable of so documenting it are also the only people capable of working on it.  And then compare that number to the average donation rate of  less than 20 cents per hour for the time the developers have spent on this code base.

V1EngineeringInc, 2024-07-19:
Not all of us understand programming like you do. That seems like a very odd response from someone that understands ever single line in great detail and knows why he chose that exact option and didn't break it out. 

I privately asked someone I know about this very thing, "should we have control over this?" His response was "I am not sure, it might help it might not"

MitchBradley, 2024-07-19:
The point is that programming is very time consuming and documentation is even more so.  You have to draw the line somewhere.  Explaining things in excruciating detail quickly becomes counterproductive, especially in light of the fact that few people read the code, and commentary often becomes stale, incorrect, and thus of negative value.  Git makes it possible to trace the history of lines of code and often to correlate them to specific issues and pull requests, which serves the purpose better than trying to summarize the situation in few enough lines of commentary so as not to obscure the flow of the code with chattiness.

bdring, 2024-07-28:
I tried a new setting. `$Wifi/PsMode={NONE,MIN,MAX}`, to test the power consumption. I used an easy, but low quality meter.

 
[image]

Here are the values I found on a raw chip with a default config.

$Wifi/mode=AP
Max= 0.11 to 0.14 (toggles slowly)
Min = 0.11 to 0.14 (toggles quickly)
None=0.14

$Wifi/Mode=STA
Max=0.04
Min=0.04
None=0.11

closed 2024-12-17 by bdring

0x5e, 2025-07-23:
My test result:

`$WiFi/Mode=Off` (0.07A)

`$WiFi/Mode=STA>AP`, `$Wifi/PsMode=None` (0.15A)
`$WiFi/Mode=STA>AP`, `$Wifi/PsMode=Min` (0.09A)

`$Wifi/PsMode` is working for me, it takes effect after reboot.