bdring/FluidNC #1263: controllare un motore nidec a 6 fili mediante mks dlc32

distebia, 2024-07-16:
### Machine Context

test

### Feature Description

Salve, ho un motore Nidec brushless con 6 fili. 
La piedinatura è 24v, Gnd, 5V, Brk, Pwm e Dir.
Avevo pensato di collegare il 24v e Gnd alla stessa fonte che alimenta la Mks Dlc32, il 5v da un qualsiasi terminale 5v sopra la scheda e il pwm dal gpio.32 
Per Brk e Dir non so come fare...magari modificare qualche esempio di mandrino tipo 10V, ma non so se è corretto
Secondo voi è possibile controllare un motore del genere?

  

### Other Approaches

oppure ho bisogno di una scheda esterna per controllare il motore?

### How I Can Help

.

bdring, 2024-07-16:
you can use the PWM spindle, but we do not support BRK. (Brake?)

distebia, 2024-07-16:
Non traduco bene quel poco di documentazione che ho trovato...
il link è qualcosa di simile https://it.aliexpress.com/i/32650831394.html?gatewayAdapt=glo2ita e mi sembra di capire il brk se collegato a gnd permette il movimento, se scollegato da gnd è freno

bdring, 2024-07-16:
just wire it to ground then.

link does not work for me

distebia, 2024-07-16:
[image]
https://it.aliexpress.com/i/32650831394.html?gatewayAdapt=glo2ita
non capisco perche non funziona il link
Se 24v, gnd, 5v e pwm sono corretti, quali sarebbero i migliori pin per Dir e brk? Esiste qualche pin che posso usare per dare gnd a brk tipo pin enable o qualcosa del genere?

bdring, 2024-07-16:
Enable with the `:low` attribute would work.

distebia, 2024-07-16:
ma proprio il pin enable?

bdring, 2024-07-16:
Everything we know about that cheap chinese controller is here.

http://wiki.fluidnc.com/en/hardware/3rd-party/MKS_DLC32

distebia, 2024-07-16:
ok, grazie, provo a costruire un file yaml e fare alcuni test

distebia, 2024-07-17:
pwm:
  pwm_hz: 500
  direction_pin: gpio.25:low
  output_pin: gpio.32
  enable_pin: gpio.26:low
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 1000=50.000%
  off_on_alarm: false

Salve, ho collegato tutto e riesco ad eseguire i comandi della rotazione orario e antiorario con 
orario
M3 S10
G1 F1000
e antiorario
M4 S10
G1 F1000

quello che non riesco a regolare è la velocità...se dico m3 s1000 g1 f1000 la rotazione è uguale a m3 s1 g1 f1000
l'unico modo per aumentare la velocità è cambiando il valore pwm_hz. ho provato a 20000 e gira veloce e fluido, se metto 100-300 vedo una rotazione lenta e a scatti, come se non sono sintonizzato bene
il gpio.32 emette regolarmente 0-5v al variare del comando m3 s...

bdring, 2024-07-17:
PWM of 500 is a little low. Do you have a datasheet for the spindle?

Most would use 5000-10000. Try measuring the voltage of S1000 and S500 with PWM of 5000.

distebia, 2024-07-18:
Forse ho capito perchè non posso regolare la velocità...il motore, molto probabilmente, è regolato da impulsi di CLK timer. Esiste un modo per sostituire il pwm in uscita con CLK?

distebia, 2024-07-19:
Esiste qualche mandrino da configurare per comandarlo in frequenza variabile?

bdring, 2024-07-19:
Is there a datasheet for this spindle?

Have you contacted the supplier with questions?

distebia, 2024-07-19:
non riesco a trovare niente online. Il motore è Nidec G1271170 DC24V 1.5A 9.6W.

bdring, 2024-07-19:
Sorry, I can't help you without a datasheet.

9.6 watts? The weakest spindle I ever used was 300w

distebia, 2024-07-19:
lo so che è bassa potenza. Il motore a bassi giri ha un'alta coppia. Ne potrei ricavare un sistema di filettatura automatico. quello che mi serve è soltanto avere un pin su mks che genera frequenza variabile anzichè pwm, sempre se è possibile farlo

MitchBradley, 2024-07-19:
ESP32 has no obvious way to generate variable frequency drive signals directly.  You would need to use external hardware to generate the signals, like a BESC controller.  We have no way of supporting undocumented third-party hardware, so you are on your own.

closed 2024-07-19 by MitchBradley

distebia, 2024-07-20:
[pdfcoffee.com_nidec-24h-bl4234-4-pdf-free.pdf](https://github.com/user-attachments/files/16322256/pdfcoffee.com_nidec-24h-bl4234-4-pdf-free.pdf)
ecco la scheda tecnica, il modello è 27h

MitchBradley, 2024-07-20:
ESP32 cannot drive 3-phase brushless motors directly.  To drive such a motor, you need a "BLDC motor driver".  In order to use it with FluidNC, you need one that has a PWM input.  Here is an example https://www.amazon.com/UMLIFE-Three-Phase-Brushless-Controller-Control/dp/B09BHXLGW6

distebia, 2024-07-20:
[image]
il motore ha già la sua scheda integrata, se vario il pwm_hz cambio la velocità, se invio m3....vado in senso orario e con m4 vado in senso antiorario.e m5 arresta il motore...farò delle macro in LB dove vario il pwm_hz per avere diverse velocità

MitchBradley, 2024-07-20:
The PWM spindle code is not intended for dynamic control of the frequency.  The frequency is set once at startup, then the motor speed is controlled by changing the duty cycle.  It might be possible to write a new Spindle code module that dynamically controls the frequency base on the GCode S value, but FluidNC currently has no such code.

distebia, 2024-07-20:
ok, capisco....grazie...se un giorno verrà implementata questa nuova funzione la prego di avvisarmi

MitchBradley, 2024-07-20:
You are the only person who has ever asked for this capability so I don't expect that we will be implementing it any time soon.

distebia, 2024-07-20:
oggi sono l'unico (pazzo) che vuole controllare un motore brushless con alta coppia a bassi giri...domani magari se ne aggiungeranno altri, chissà...