bdring/FluidNC #1264: Problem: Can't use SYNRAD CO2 RF laser

sk8erpunkd, 2024-07-16:
### Wiki Search Terms

my laser need 1us tickle on pwm to start

### Controller Board

ESP32-WROOM

### Machine Description

Homemade CNC router (arduino uno with grbl 1.1) updated to esp32 with fluidNC, trying to upgrade from pwm laser diode (40W, 10W optical power) to synrad firestar v30

### Configuration file

```yaml
Laser:
  pwm_hz: 5000
  output_pin: gpio.23
  disable_with_s0: true
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
Connected
```

### User Interface Software

WebUI - LaserGRBL

### What happened?

I tried to upgrade the laser head from a "LaserTree 40W laser diode" to "Synrad Firestar v30 75W RF CO2" but it seems the pwm control pin needs a 1us "tickle" to start rf modulation.
i want to know if there's a way to add this the pwm output
i'm a little noob and i don't know if i need this 1us as value for 0% power or it needs to be part of pwm signal all the time

bdring, 2024-07-16:
See this page.

http://wiki.fluidnc.com/en/config/spindle_speed_maps

Calculate the percent from the pulse width and frequency

bdring, 2024-12-11:
Closing stale issue

closed 2024-12-11 by bdring