bdring/FluidNC #1266: Problem: Stepper motors not moving

DarkSlaayer, 2024-07-19:
### Wiki Search Terms

Motor, Axes, Motor issues, pretty much everything

### Controller Board

6-pack Universal

### Machine Description

Shapeoko 3XXL
[image]
[image]

### Input Circuits

```yaml
IO module in spot one for Limit Switches they are working and reporting back fine with $limits and the calibration screen when connected via usb. The RS485 & relay module outputs arent currently connected to anything.
```

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack TMC5160 XYZ
stepping:
  engine: I2S_STATIC
  idle_ms: 250
  pulse_us: 6
  dir_delay_us: 1
  disable_delay_us: 1
  segments: 6
axes:
  shared_stepper_disable_pin: NO_PIN
  homing_runs: 2
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 200
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.33:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      tmc_5160:
        cs_pin: i2so.3
        r_sense_ohms: 0.075
        run_amps: 2
        hold_amps: 0.5
        microsteps: 16
        stallguard: 0
        stallguard_debug: true
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: i2so.2
        direction_pin: i2so.1
        disable_pin: I2SO.0
  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 200
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      tmc_5160:
        cs_pin: i2so.6
        r_sense_ohms: 0.075
        run_amps: 2
        hold_amps: 0.5
        microsteps: 16
        stallguard: 0
        stallguard_debug: true
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: i2so.5
        direction_pin: i2so.4
        disable_pin: I2SO.7
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35
      hard_limits: false
      pulloff_mm: 1
      tmc_5160:
        cs_pin: i2so.11
        r_sense_ohms: 0.075
        run_amps: 2
        hold_amps: 0.5
        microsteps: 16
        stallguard: 0
        stallguard_debug: true
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: i2so.10
        direction_pin: i2so.9
        disable_pin: I2SO.8
  z:
    steps_per_mm: 80
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 200
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.32
      hard_limits: false
      pulloff_mm: 1
      tmc_5160:
        cs_pin: i2so.14
        r_sense_ohms: 0.075
        run_amps: 2
        hold_amps: 0.5
        microsteps: 16
        stallguard: 0
        stallguard_debug: true
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: i2so.13
        direction_pin: i2so.12
        disable_pin: I2SO.15
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
probe:
  pin: NO_PIN
  check_mode_start: true
  toolsetter_pin: NO_PIN
macros: {}
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
start:
  must_home: false
PWM:
  pwm_hz: 5000
  output_pin: gpio.26
  enable_pin: gpio.4
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.8.0 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6 Pack TMC5160 XYZ]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:6us Dsbl Delay:1us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:I2SO.2 Dir:I2SO.1 CS:I2SO.3 Disable:I2SO.0 Index:-1 R:0.075]
[MSG:INFO:  X Neg Limit gpio.33:pu]
[MSG:INFO: Axis Y (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:I2SO.5 Dir:I2SO.4 CS:I2SO.6 Disable:I2SO.7 Index:-1 R:0.075]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:I2SO.10 Dir:I2SO.9 CS:I2SO.11 Disable:I2SO.8 Index:-1 R:0.075]
[MSG:INFO:  Y2 All Limit gpio.35]
[MSG:INFO: Axis Z (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:I2SO.13 Dir:I2SO.12 CS:I2SO.14 Disable:I2SO.15 Index:-1 R:0.075]
[MSG:INFO:  Z All Limit gpio.32]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.4 Out:gpio.26 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Connecting to STA SSID:SlowWifi]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.50.145]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
```

### User Interface Software

WebUI

### What happened?

I tried to jog motor doesnt move while the jog is running motor is locked up until the jog command is finished

DarkSlaayer, 2024-07-19:
$100 Bounty info at bottom of post.

bdring, 2024-07-19:
What are the power supply specs?

MitchBradley, 2024-07-19:
Try with the other drivers, like the X and Y.  Also try using FluidTerm to send a motion command like "G1 Z20 F400".  If the motor is locked but not moving, that usually means that the driver is not getting STEP pulses or the STEP pulse rate is too fast.  I don't think that "too fast" is the problem here, because the acceleration and rates are reasonable, but it is always a good idea to start with slow movements when testing.

MitchBradley, 2024-07-19:
http://wiki.fluidnc.com/en/support/setup/testing-step-pins

DarkSlaayer, 2024-07-20:
> What are the power supply specs?

I’m using my bench power supply up to 30v 10amps.

i was possibly thinking maybe it didn’t like the variable power supply so took a 12v 50amp psu and hooked up a 24v buck booster that’s good for 6amps and nothing changed… I do have a 24v psu on the way though.

DarkSlaayer, 2024-07-20:
> Try with the other drivers, like the X and Y. Also try using FluidTerm to send a motion command like "G1 Z20 F400". If the motor is locked but not moving, that usually means that the driver is not [...]

I have tried every axis so far they all do the same thing. I’ll try the LED thing with the driver pins when I get back into town but I did try a smaller cheapo stepper motor and it worked fine…

MitchBradley, 2024-07-20:
Do you mean a cheapo stepper motor or a cheapo motor driver like an A4988?

MitchBradley, 2024-07-20:
That information (the fact that you tried different axes, and the fact that you got it to work under some circumstances) is very important.  Had we know that earlier, that would have eliminated many different possibilities.  Don't make us guess.

DarkSlaayer, 2024-07-20:
> Do you mean a cheapo stepper motor or a cheapo motor driver like an A4988?

Stepper motor

DarkSlaayer, 2024-07-20:
> That information (the fact that you tried different axes, and the fact that you got it to work under some circumstances) is very important. Had we know that earlier, that would have eliminated many different possibilities. Don't make us guess.

I apologize

DarkSlaayer, 2024-07-20:
Would it been fine to set the board to SD and try to control the steppers without using spi? What would cause it to work with a nema 17 motor I believe and not a nema 23?

DarkSlaayer, 2024-07-20:
I’m starting to think it’s my PSU my 24v one comes in today but I’ll be out of town still.. I was thinking maybe it’s fighting with the board in some way as I noticed even if I set it to 30v it would only go to 22ish volts.. I guess some bench PSU will only raise the voltage as the request for current increases.

bdring, 2024-07-20:
Try leaving all chips installed and disconnect all motors, but 1.

The 5160 is notorious for current spikes. I suggest at least 4 amps per motor regardless of current levels in the config file.

DarkSlaayer, 2024-07-20:
> Try leaving all chips installed and disconnect all motors, but 1.
> 
> The 5160 is notorious for current spikes. I suggest at least 4 amps per motor regardless of current levels in the config file.

I have tried one motor with all drives. With the config set to 2amp run will it still allow the spike or should
I bump it.

MitchBradley, 2024-07-20:
There is no way to predict the spike behavior - or whatever is causing the shutdown - without detailed knowledge of every aspect of the circuit.  Everything from the power supply through the wiring through the controller through the connector pins through the stepstick and its MSOFET through the motor wiring and the motors could be implicated in the dynamic behavior. This is the sort of situation that, in a commercial setting, could require days or weeks in the lab with expensive equipment, trained engineers, and perhaps involvement with vendors.  5160s were intended for commercial use by companies that have the resources to tune the circuits and settings for specific product configurations.  Putting them on stepsticks and throwing them out into the DIY world has caused no end of trouble.  Many people have had problems that were unsolvable with the resources available.  The stepstick format is particularly ill-suited for a driver with this voltage and current capability, even if you have it dialed down to a relatively low current.  The resistance and inductance of on-board traces and dupont pins can result in not only surges but also voltage dips that can trigger protection circuits.

DarkSlaayer, 2024-07-20:
> There is no way to predict the spike behavior - or whatever is causing the shutdown - without detailed knowledge of every aspect of the circuit. Everything from the power supply through the wiring [...]

Yes I understand and seen some comments about this specific step stick. I may just order the adapters to run external drivers and replace the step sticks with some drivers from steppers online

MitchBradley, 2024-07-20:
The bottom line is that we probably can't give you a definitive "do this and it should work" answer.  You might get lucky and try something that we or you think of, but don't expect anyone to just know.  For myself, I have quit using 5160s.  The 5160 setups I have tried are just too prone to shutting down in the middle of a job.  As an experienced electrical engineer, I probably would have been able to solve the problems given enough time and motivation, but for my present use cases, the motivation to spend the time was not present, so I switched to less-troublesome drivers.

MitchBradley, 2024-07-20:
Many people are very happy with DM5xx series of external drivers from reputable suppliers, but beware of cheap knock-offs.

3404gerber, 2024-07-20:
Hi, can I ask a picture of the TMC5160 jumpers? Can it be that you are in full SPI mode?

MitchBradley, 2024-07-20:
If it were in full SPI mode, it would not work  with NEMA 17 motors.

3404gerber, 2024-07-20:
You're right, i didn't see that information. Next guess, I found a couple of pictures showing that B1 and B2 were inverted on the new BTT 5160T V1.1. Did you try to invert the polarity of one of the coils?

MitchBradley, 2024-07-20:
Inverting one coil polarity makes the motor move in the opposite direction.  Inverting both has no effect - the first first inversion reverses the direction and the second reverses it back to normal.

3404gerber, 2024-07-20:
Ok, now it looks like I'm the one who need's help... But hey, quite new to stepper and still learning! ;)

Don't know how relevant this is, but your step per mm seems completely off: I'd go with 640, if all axes have a 5mm pitch screw, and 200 steps per revolution motor running at 16 microsteps. Can it be that you jog one millimeter and it's moving only 0 .125mm?

MitchBradley, 2024-07-20:
Again, doesn't explain the difference in behavior between NEMA 17 and NEMA 23 motors, which generally have the same steps per revolution.

3404gerber, 2024-07-20:
> Again, doesn't explain the difference in behavior between NEMA 17 and NEMA 23 motors, which generally have the same steps per revolution.

Well, what does? If I understand it right, the driver limits the current, no matter what motor is connected. For me, if it "works" with the NEMA 17, it should work with the NEMA 23.Maybe not good, but something should happen.

For the record, I just double checked: the X and Y axis have a belt with a pitch of 2mm and a 20 teeth pulley, so 80 steps per mm is correct...

BlueOrangeLive, 2024-07-21:
I once had a problem with motors from different manufacturers where the four cables had the same colors as the two motors, but the coils were connected differently. So unplug the plug from the board and measure it.

DarkSlaayer, 2024-07-26:
I think it may be a board issue.. I had some a4988 step sticks laying around and they do the same thing..

MitchBradley, 2024-07-26:
A4988 step sticks require different jumper settings than TMC5160s.  If "the same thing" means that NEMA17 works but NEMA23 does not, I suspect that you have the NEMA23s wired wrong.  Did you check the coils as BlueOrangeLive suggested?

DarkSlaayer, 2024-07-26:
> A4988 step sticks require different jumper settings than TMC5160s. If "the same thing" means that NEMA17 works but NEMA23 does not, I suspect that you have the NEMA23s wired wrong. Did you check the coils as BlueOrangeLive suggested?

I did not check them on the nema 17s but yes I did change the jumpers and configuration.

Yes coils have been verified both via ohm meter and jumping the wires and pushing the gantry and checking for resistance.

I’m really stumped I’m not new to this kind of stuff and I was hoping it was some kind of configuration issue.

I even said hell with it and switched pairs around etc and nothing new.

The steppers are free moving but if I jog them they lock up until the jog command is over and they go back to be free moving manually by hand

I ordered some external drivers. I wired up the OEM board back up for the time being.

MitchBradley, 2024-07-26:
I did a quick test where I swapped motor wires so coil 1 was connected to driver A- B- and coil 2 to driver A+ B+.  The motor locked, but when I tried a move, it just shook and did not turn.  This is as I expected.  It is a 6-pack universal board with A4988 step sticks driving a short-length NEMA23 motor with no load.  It spun as expected with the coils connected correctly.

MitchBradley, 2024-07-26:
I did some DC and AC voltage measurements across the coils.  I have $stepping/idle_ms=255 so the motors stay locked.  My power supply voltage is 12V, using A4988 drivers.

The DC voltage across a coil is in range from -1.5V to +1.5V depending on the microstep phase.  I tested that by going into G91 relative mode then repeatedly issuing G0 Y0.02 command to step by one microstep - since this system has $/axes/y/steps_per_mm=50.  For 80 steps/mm the increment would be 0.0125.

I then ran long fast moves while measuring the AC voltage across the coil.  During the motion, I saw an AC voltage of 8V.  That would depend on the meter; a true RMS meter would probably measure lower.

MitchBradley, 2024-07-26:
The exact voltages would depend on several factors including driver current settings and coil resistances, but the qualitative behavior should be similar.  The motor I tested is rated for 1.8 ohm coil resistance.  The A4988 drivers are set for Vref=0.32V with Rsense=0.068R (pre-2017 Pololu md09b), for a current limit of 0.59A.  That corresponding to a coil voltage of 1.8R * 0.59A = 1.06V, but that is the RMS AC voltage, so the peak DC voltage would be 1.414 time that, or 1.5VDC, as I measured.

DarkSlaayer, 2024-07-27:
I went through again and the coils are being paired accordingly nema 17s work fine but nema 23s arent i noticed when the board is powering up the stepper is locked but then goes limp does that mean they are holding then stopping?

MitchBradley, 2024-07-27:
It could mean that the driver is shutting down to protect itself.  Perhaps you could find an alarm pin to probe.

DarkSlaayer, 2024-07-27:
> It could mean that the driver is shutting down to protect itself. Perhaps you could find an alarm pin to probe.

One thing I do see is people are grounding the clk pin to use internal clock vs external is that needed with this board?

MitchBradley, 2024-07-27:
I know nothing about that.  I get not paid for writing the software.

DarkSlaayer, 2024-07-27:
> I know nothing about that. I get not paid for writing the software.

Update clk pin did nothing but soldering a jumper from the enable pin to gnd to force it to stay enabled worked.. now why isnt the enable pin working?

MitchBradley, 2024-07-27:
Maybe the enable needs to be inverted - but that does not explain the NEMA17's working.

DarkSlaayer, 2024-07-27:
> Maybe the enable needs to be inverted - but that does not explain the NEMA17's working.

Yeah strange.. Im just going to jumper all my drivers and call it good i guess!

3404gerber, 2024-07-27:
Hi again, seems like neither your board nor the drivers have a pull-down resistor on the Enable pin. Floating enable pin leads to strange behavior I guess; had something very similar in full SPI mode with end switch pin left floating. If you add a resistor to the ground rather than a jumper, you'll still be able to use the disable function.

I also have the last FluidNC board for external drivers, which have all the Step/Dir/Enable pulled down and had no issue with the TMC5160 from BTT. By the way, the clk pin should be connected to ground. Again, on the drivers I have, there is a zero ohm resistor connecting the clk to the ground.

MitchBradley, 2024-07-27:
[image] 

The enables on a 6 pack universal board are driven from 74AHCT595 chips which have "totem-pole" outputs, and the OE_ pin for those chips is grounded, so pull-down is not appropriate for such a circuit.  The '595 chip always drives either solid low or solid high.  A pulldown resistor would have no effect other than to waste current in the high state.

The same is true of the 6 pack external board.  There is a resistor+LED on the enable signal going to the external driver, but that does not serve as a pulldown, but rather as an indicator.  It does not affect the VI characteristics as seen by the external driver - and external driver current-mode signaling is quite different from stepstick voltage-mode signaling.

3404gerber, 2024-07-27:
Thanks for the information! The datasheet gives a VCC value between 4.5 and 5.5V for the 74HCT595 used. If I read the schematic correctly, they are supplied with 3.3V depending on the VCC jumper, right?

MitchBradley, 2024-07-27:
AHCT not HCT.  I did an analysis of the electrical characteristics somewhere.  Maybe you can find it.  The bottom line is that, even though it is not rated for use at 5V, since it is intended for replacing 5V TTL, it works just fine at 3.3V.

closed 2024-12-12 by bdring