bdring/FluidNC #1268: Problem: Unable to clear "X Axis tmc_2209 homing current not in config: Using run current" error

bzkfrl1, 2024-07-24:
### Wiki Search Terms

3.8.0, homing current, YAML.config

### Controller Board

FluidNC Pen/Laser TMC2209 V1.4 @buildlog 4/2022

### Machine Description

ZenXY V1 Engineering 730mmx730mm CoreXY Sand Table

### Configuration file

```yaml
name: "TMC2209 XY Zen Table"
board: "FluidNC Pen/Laser 2209"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

kinematics:
  corexy:
  
uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 749
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 28
      feed_mm_per_min: 200.000
      seek_mm_per_min: 500.000
      settle_ms: 500
    
    motor0:
      limit_neg_pin: gpio.36:high
      #Diag pin is gpio.34
      tmc_2209:
        uart_num: 1
        addr: 0
        r_sense_ohms: 0.110
        run_amps: 0.500
        hold_amps: 0.250
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        homing_amps: 0.250
        use_enable: false
        direction_pin: gpio.12
        step_pin: gpio.14
        disable_pin: NO_PIN
    
    
    motor1:
      null_motor:

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 749
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 18.5
      feed_mm_per_min: 200.000
      seek_mm_per_min: 500.000
      settle_ms: 500

    motor0:
      limit_neg_pin: gpio.39:high
      #diag pin is gpio.35
      tmc_2209:
        uart_num: 1
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 0.500
        hold_amps: 0.250
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        homing_amps: 0.250
        use_enable: false
        direction_pin: gpio.26
        step_pin: gpio.25
        disable_pin: NO_PIN
    motor1:
      null_motor:

  z:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 200.000
    soft_limits: false
        
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

start:
  must_home: false
```

### Startup Messages

```text
Config File Error
Fix it or FluidNC will not work
   
[MSG:INFO: FluidNC v3.8.0 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: X Axis tmc_2209 homing current not in config. Using run current]
[MSG:WARN: Y Axis tmc_2209 homing current not in config. Using run current]
[MSG:INFO: Machine TMC2209 XY Zen Table]
[MSG:INFO: Board FluidNC Pen/Laser 2209]
[MSG:INFO: UART1 Tx:gpio.17 Rx:gpio.16 RTS:NO_PIN Baud:115200]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (28.000,777.000)]
[MSG:INFO: Motor0]
[MSG:INFO: tmc_2209 UART1 Addr:0 CS:NO_PIN Step:gpio.14 Dir:gpio.12 Disable:NO_PIN R:0.110]
[MSG:INFO: X Neg Limit gpio.36]
[MSG:INFO: Motor1]
[MSG:INFO: Axis Y (18.500,767.500)]
[MSG:INFO: Motor0]
[MSG:INFO: tmc_2209 UART1 Addr:1 CS:NO_PIN Step:gpio.25 Dir:gpio.26 Disable:NO_PIN R:0.110]
[MSG:INFO: Y Neg Limit gpio.39]
[MSG:INFO: Motor1]
[MSG:INFO: Axis Z (-200.000,0.000)]
[MSG:INFO: Motor0]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:bzkfrl1]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.44]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI v2

### What happened?

After upgrading from 3.7.16 to 3.8 I noted the 'homing current' config file errors in WebUI upon my first startup.  I am still able to run and control the sand table with these warning messages. I added the 'homing_amps: 0.250' lines to their respective x and y tmc_2209 sections and replaced the config.yaml file on the SD card  in my controller.  However, I continue to receive these errors  and would appreciate your guidance in correcting my config.yaml file and eliminating these errors.

MitchBradley, 2024-07-24:
It worked for me.  Perhaps you did not successfully replace the config.yaml file?  Send $sd/show=config.yaml to see if the homing_amps lines are there.

bzkfrl1, 2024-07-25:
Thank you.  I'll give that a try in the morning.

bzkfrl1, 2024-07-25:
I confirmed that the "homing_amp: lines are there pre your suggestion (see below).  I'm still getting the warnings.  I would welcome any other diagnostic suggestions you might have.

bzkfrl1, 2024-07-25:
$sd/show=config.yaml
name: "TMC2209 XY Zen Table"
board: "FluidNC Pen/Laser 2209"
stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0
kinematics:
  corexy:
uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1
axes:
  shared_stepper_disable_pin: gpio.13:high
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 749
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 28
      feed_mm_per_min: 200.000
      seek_mm_per_min: 500.000
      settle_ms: 500
    motor0:
      limit_neg_pin: gpio.36:high
      #Diag pin is gpio.34
      tmc_2209:
        uart_num: 1
        addr: 0
        r_sense_ohms: 0.110
        run_amps: 0.500
        hold_amps: 0.250
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        homing_amps: 0.250
        use_enable: false
        direction_pin: gpio.12
        step_pin: gpio.14
        disable_pin: NO_PIN
    motor1:
      null_motor:
  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 749
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 18.5
      feed_mm_per_min: 200.000
      seek_mm_per_min: 500.000
      settle_ms: 500
    motor0:
      limit_neg_pin: gpio.39:high
      #diag pin is gpio.35
      tmc_2209:
        uart_num: 1
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 0.500
        hold_amps: 0.250
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        homing_amps: 0.250
        use_enable: false
        direction_pin: gpio.26
        step_pin: gpio.25
        disable_pin: NO_PIN
    motor1:
      null_motor:
  z:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 200.000
    soft_limits: false
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
start:
  must_home: false
ok

MitchBradley, 2024-07-25:
The config file is supposed to be on localfs not SD.  I mistakenly told you to issue $sd/show=config.yaml instead of $localfs/show=config.yaml .  The fact that it was found by $sd/show=config.yaml tells us what the problem is - you uploaded the file to the wrong filesystem.

bzkfrl1, 2024-07-31:
Yeah...that's what I did.  Thanks for your help.

closed 2024-07-31 by bzkfrl1