bdring/FluidNC #1272: Problem: yaml file.

Hooks321, 2024-07-26:
### Wiki Search Terms

Only new here.

### Controller Board

6-pack ESP-WROOM-32

### Machine Description

Gantry Laser setup.

### Configuration file

```yaml
Wont past
```

### Startup Messages

```text
Cannot open configuration file:<your_filename>
```

### User Interface Software

Fluidterm

### What happened?

Every time I upload the config yaml file it said the configuration is invalid.??? any help would be great thanks.

### GCode File

?

### Other Information

?

breiler, 2024-07-26:
Can you show the output from the command `$SS`

Hooks321, 2024-07-26:
Hope this helps.

breiler, 2024-07-26:
Did you try to attach something? That doesn't work when using email - you need to go to the issue to do that: https://github.com/bdring/FluidNC/issues/1272

Hooks321, 2024-07-26:
Did that come through?

Hooks321, 2024-07-26:
[image]
Sorry for the delay.

breiler, 2024-07-26:
The `<your_filename>` is probably the default setting, select your config file with:
`$Config/Filename=Laser Home XYZ (1).yaml`

Or you could use the https://installer.fluidnc.com/ go to the "File browser" and click the "Select Config" button:

[image]

Hooks321, 2024-07-26:
Still trying.

Hooks321, 2024-07-26:
OK, selected active config. what do I do next.

breiler, 2024-07-26:
Restart the controller and check the boot messages.

If you are doing this from the https://installer.fluidnc.com/ just go to the "Terminal" page and hit "Restart"

Hooks321, 2024-07-26:
Hi, looks like its loaded OK now. Thanks for all your help so far, much appreciated. Is there a way for this card to be seen in Lightburn software now.?

MitchBradley, 2024-07-26:
Did you search for "lightburn" in the fluidnc wiki?

Hooks321, 2024-07-27:
Yes, I found it. Thankyou.

closed 2024-12-12 by bdring