bdring/FluidNC #1273: Resetování ESP32 po instalaci FluidNC

jjvcela, 2024-07-26:
### Wiki Search Terms

resetování 

### Controller Board

CNC shield

### Machine Description

 vlastní 3d frézka

### Configuration file

```yaml
zatím ještě nemám
```

### Startup Messages

```text
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
ets Jul 29 2019 12:21:46
```

### User Interface Software

FluidTerm

### What happened?

Po nainstalování FluidNC se spustí FluidTerm a vypisuje opakovanou zprávu o resetování. Jiné moje kódy na všech mých deskách ESP32 fungují, problém dělá jen FluidNC

breiler, 2024-07-26:
Controller type "CNC shield" is not sufficient. If you are running a ESP32 devkit, please provide the the chip name. Preferably a link to the item you bought.

How did you install the FluidNC firmware?

jjvcela, 2024-07-26:
Mám několik desek ESP32_Devkitc_V4 (čip WROM-32D) a ESP-32S v1.1 (čip WROM-32) . Všechny se chovají stejně. Pro instalaci jsem použil Webovou instalační službu FluidNC, install-bt.bat a  flash_download_tool_3.9.7. S FluidNC začínám a proto ten CNC shield.

MitchBradley, 2024-07-26:
Did you try using erase.bat first?

What version of FluidNC are you using?

jjvcela, 2024-07-27:
Verzi jsem použil aktuální. Instaloval jsem do nových karet i do již použitých včetně erase.bat.

MitchBradley, 2024-07-27:
Version 3.8.0 has a problem with bluetooth that was fixed by https://github.com/bdring/FluidNC/pull/1265 .  The next release will include that fix.  You can try an earlier version while we prepare the new release.

jjvcela, 2024-07-27:
To je ono, bingo. Instaloval jsem verzi BT pomocí install-bt.bat. Kde, prosím, najdu předkompilovanou starší verzi? Na https://github.com/bdring/FluidNC/tree/v3.7.18 jsem našel jen zdroje.

MitchBradley, 2024-07-27:
[image]

jjvcela, 2024-07-28:
Moc děkuji, to jsem potřeboval. Verze v3.7.18 má tu samou chybu, ale verze v3.7.17 se mi podařilo nainstalovat. Ještě jednou děkuji za pomoc.
Honza

closed 2024-07-28 by MitchBradley