bdring/FluidNC #1275: Problem: config. yaml file

Hooks321, 2024-07-27:
### Wiki Search Terms

yaml file help.

### Controller Board

6 pack controller esp-wroom-32

### Machine Description

Gantry laser 1300x900 diode 10w output.

### Input Circuits

```yaml
?
```

### Configuration file

```yaml
Wont paste.
```

### Startup Messages

```text
machine won't move.
```

### What happened?

Machine won't move.

### Other Information

Just need a standard laser config.yaml file/ 3 axis suit 10w diode laser/ motor steps-533.366616 . used for cutting and engraving.

bdring, 2024-07-28:
We need to see your existing config file to be able to help.

Please cut and paste on a reply.

Hooks321, 2024-07-28:
[Laser Home XYZ.txt](https://github.com/user-attachments/files/16401896/Laser.Home.XYZ.txt)

Hooks321, 2024-07-28:
[image]
This is the boardThat I am trying to use.

bdring, 2024-07-28:
I have no knowledge of that controller. It is not on our [known hardware](http://wiki.fluidnc.com/en/hardware/existing_hardware) list.

Can you provide links to the schematic, datasheet and where you got it. Have you contacted the supplier for help?

I am pasting your file here. It has a lot of NO_PIN values, that we cannot help you with.

```yaml
board: 6 Pack
name:Laser Home XYZ
meta: 2024 Gary
stepping:
  engine: I2S_stream
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 12

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 533.366616
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 900.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 600.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 3.000
      stepstick:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: I2SO.3
        reset_pin: NO_PIN

  y:
    steps_per_mm: 533.366616
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 1300.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 600.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: I2SO.6
        reset_pin: NO_PIN

  z:
    steps_per_mm: 533.366616
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 20.000
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 600.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

start:
  must_home: false

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
NoSpindle:
```

bdring, 2024-07-28:
It might be this.

https://github.com/Macrobase-tech/CNC-Software/tree/main/6%20Axis%20Upgrade

It does not appear that they reply to issues. I did not see any source files. I did not open any of the .rar files. There is a config file here.

https://github.com/Macrobase-tech/CNC-Software/tree/main/6%20Axis%20Upgrade/Software%20instruction

I do not think it is worth our time to offer free support to this controller.

Hooks321, 2024-07-28:
[Board.pdf](https://github.com/user-attachments/files/16405298/Board.pdf)
Sorry it was night time down here when you guys replied. I have an old trimph C02 1300x900 laser which i am trying to convert to diode.
Had it running on a mach3 board but didnt engrave well, so just thought i would try Fluidnc download . Any help with a config 
yaml file would be great as it is all very confusing. Thanks all.

MitchBradley, 2024-07-28:
We spent a lot of time documenting and explaining the config file on the wiki, and provided numerous examples for different situations.  Now you are asking us to spend even more time spoon-feeding the information to you, on behalf of barely-documented third-party hardware that hardly anyone else uses.  I we created custom config files, for free, for every new user, development would come to a halt and it would not be long before the project died.

Hooks321, 2024-07-28:
I was only asking for help as it is so confusing on your website.

BlueOrangeLive, 2024-07-29:
Here is the diagram from AliExpress.

Have a nice day.
[Macrobase_6Axes_Schematic.pdf](https://github.com/user-attachments/files/16410584/Macrobase_6Axes_Schematic.pdf)

Hooks321, 2024-07-29:
Thanks for all your help, looks like I should back of this as it seems to complicated. Cheers.

bdring, 2024-07-29:
That is probably best. Converting a laser can be complicated and dangerous on a lot of levels.

I have added the board to our [hardware page](http://wiki.fluidnc.com/en/hardware/existing_hardware#macrobase-6-axis) for future reference.

closed 2024-07-29 by bdring

Skorpi08, 2024-07-29:
@Hooks321 
try this
[Macrobase 6 Axis.zip](https://github.com/user-attachments/files/16419259/Macrobase.6.Axis.zip)