bdring/FluidNC #1281: Problem: DM320T/ DM332T no movement

maxoutny, 2024-08-03:
### Wiki Search Terms

DM320T, DM332T

### Controller Board

6-pack 6x CNC Controller

### Machine Description

Root 4 Lite
[image]
[image]

### Configuration file

```yaml
board: 6x CNC Controller
name: 6x Default
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 10
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0:low

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 400.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.39
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7:low

    motor1:
      limit_neg_pin: NO_PIN   
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8:low
        
  z:
    steps_per_mm: 500.00
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 59.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 28.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.36
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15:low

  a:
    steps_per_mm: 53.4
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16:low

  #c:      
   # steps_per_mm: 808.000
   # max_rate_mm_per_min: 5000.000
   # acceleration_mm_per_sec2: 100.000
   # max_travel_mm: 200.000
   # soft_limits: false
   # homing:
   #   cycle: 1
   #   positive_direction: false
   #   mpos_mm: 150.000
   #   feed_mm_per_min: 100.000
   #   seek_mm_per_min: 800.000
   #   settle_ms: 500
   #   seek_scaler: 1.100
   #   feed_scaler: 1.100

   # motor0:
   #   limit_neg_pin: gpio.34:low
   #   limit_pos_pin: NO_PIN
   #   limit_all_pin: NO_PIN
   #   hard_limits: false
   #   pulloff_mm: 3.000
   #   standard_stepper:
   #     step_pin: I2SO.21
   #     direction_pin: I2SO.20
   #     disable_pin: I2SO.23

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  
probe:
  pin: gpio.32:low
  toolsetter_pin: gpio.33:low
  
# Using MOSFETs (Check Spindle Pin Usage
coolant:
  flood_pin: gpio.12
  mist_pin: gpio.4
  delay_ms: 0

start:
  must_home: false
  
## Begin Huanyang  
# uart1:
  # txd_pin: gpio.15
  # rxd_pin: gpio.16
  # rts_pin: gpio.14
  # baud: 9600
  # mode: 8N1

# Huanyang:
  # uart_num: 1
  # modbus_id: 1
  # tool_num: 0
  # speed_map: 0=0% 0=25% 6000=25% 24000=100%
  # off_on_alarm: false

# #begin PWM
pwm:
  pwm_hz: 5000
  direction_pin: gpio.15
  output_pin: gpio.13
  enable_pin: gpio.14
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 10000=100.000%
  off_on_alarm: false
  
# #begin Laser
# Laser:
  # pwm_hz: 5000
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true
  
# #begin 10V
# 10V:
  # forward_pin: gpio.15
  # reverse_pin: gpio.14
  # pwm_hz: 5000
  # output_pin: gpio.13
  # enable_pin: NO_PIN
  # direction_pin: NO_PIN
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0  
  # tool_num: 0
  # speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  # off_on_alarm: false
```

### Startup Messages

```text
$G
$SS
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
[MSG:INFO: FluidNC v3.7.18 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:root4l.yaml]
[MSG:INFO: Machine 6x Default]
[MSG:INFO: Board 6x CNC Controller]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:10us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0:low]
[MSG:INFO:  X All Limit gpio.35]
[MSG:INFO: Axis Y (-250.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y Pos Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Y2 Pos Limit gpio.34]
[MSG:INFO: Axis Z (28.000,87.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO:  Z All Limit gpio.36]
[MSG:INFO: Axis A (150.000,1110.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.14 Out:gpio.13 Dir:gpio.15 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant gpio.12]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe gpio.32:low]
[MSG:INFO: Toolsetter gpio.33:low]
[MSG:INFO: Connecting to STA SSID:MySwHome_5GHz]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.0.104]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://Root4.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
<Idle|MPos:0.000,0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000,0.000>
$J=G91 G21 F1000 X10
ok
<Jog|MPos:0.105,0.000,0.000,0.000|FS:900,0|Ov:100,100,100>
[MSG:Files changed]
$HX

  
Autoscroll
  
Verbose mode
```

### What happened?

No movements

MitchBradley, 2024-08-03:
According to Discord this problem was fixed by adding :low to step and enable pins

closed 2024-08-03 by MitchBradley

maxoutny, 2024-08-04:
Also, required pulse_us: 10