bdring/FluidNC #1285: Problem: Laser firing on G0 moves

tomgale, 2024-08-04:
### Wiki Search Terms

Laser Mode, Laser, Spindle, G0

### Controller Board

MKS Tinybee V1.0

### Machine Description

CoreXY CO2 laser with DM542 motor drivers, Cloudray M100 Laser PSU.

### Configuration file

```yaml
board: MKS TinyBee V1.0 CoreXY
name: GaleLaserDesign01

kinematics:
  corexy:

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  # uses TH2 IO34 active low - MAKE SURE jumper J2 is set to SDDET!!!
  card_detect_pin: gpio.34:low

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 2

axes:
  x:
    steps_per_mm: 10.66666
    max_rate_mm_per_min: 50000.000
    acceleration_mm_per_sec2: 750.000
    max_travel_mm: 1035.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low:pu
      hard_limits: true
      pulloff_mm: 4.000
      standard_stepper:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0:high

   

  y:
    steps_per_mm: 10.666
    max_rate_mm_per_min: 50000.000
    acceleration_mm_per_sec2: 750.000
    max_travel_mm: 700.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low:pu
      hard_limits: false
      pulloff_mm: 4.000
      standard_stepper:
        step_pin: I2SO.4
        direction_pin: I2SO.5
        disable_pin: I2SO.3:high

   

  z:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 60.000
    max_travel_mm: 80.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.22:low
      hard_limits: true
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.7
        direction_pin: I2SO.8
        disable_pin: I2SO.6
             
    motor1:
      limit_pos_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.11
        disable_pin: I2SO.9
   

  

control:
  safety_door_pin: NO_PIN
  # on MT_DET connector
  reset_pin: NO_PIN
  # on TH1 connector
  feed_hold_pin: gpio.36:low
  # on TB connector
  cycle_start_pin: gpio.39:low
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  # Heated Bed Terminal Block
  flood_pin: i2so.16
  # HE0 Terminal Block
  mist_pin: i2so.17
  delay_ms: 0

Laser:
  pwm_hz: 5000
  output_pin: gpio.15
  enable_pin: i2so.18
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: true

uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

uart_channel1:
  report_interval_ms: 75
  uart_num: 1
  
start:
  must_home: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.8.0 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine GaleLaserDesign01]
[MSG:INFO: Board MKS TinyBee V1.0 CoreXY]
[MSG:INFO: UART1 Tx:gpio.17 Rx:gpio.16 RTS:NO_PIN Baud:1000000]
[MSG:INFO: uart_channel1 created at report interval: 75]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.34:low freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:2us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1035.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.1 Dir:I2SO.2 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.33:low:pu]
[MSG:INFO: Axis Y (0.000,700.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:  Y Neg Limit gpio.32:low:pu]
[MSG:INFO: Axis Z (-80.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:  Z Pos Limit gpio.22:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.11 Disable:I2SO.9]
[MSG:INFO:  Z2 Pos Limit gpio.35:low]
[MSG:INFO: feed_hold_pin gpio.36:low]
[MSG:INFO: cycle_start_pin gpio.39:low]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Laser Ena:I2SO.18 Out:gpio.15 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Flood coolant I2SO.16]
[MSG:INFO: Mist coolant I2SO.17]
[MSG:INFO: Connecting to STA SSID:VM8669094]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.59]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

LightBurn

### What happened?

Hi, I tried a test cut of two rectangles in Lightburn, the machine moved as expected, through both rectangles. The issue is the laser Enable pin switched on with the M4 command, and stayed on during the G0 moves, both to the first rectangle and between the two rectangles. Fluid NC is recognising the Laser spindle, and test firing the laser through both WebUI and Lightburn works as expected. I think Fluid NC is in laser mode, because as I understand it a 'GcodeUnsupportedCommand' error should be thrown if the M4 command is sent to an unsupported spindle. Lightburn's terminal shows no errors while or after the GCode is running. Could the issue be with the GCode being sent from Lightburn? Any help would be massively appreciated, thanks!

### GCode File

; LightBurn 1.6.03
; GRBL device profile, absolute coords
; Bounds: X14 Y21 to X127 Y103
G00 G17 G40 G21 G54
G90
M4
; Cut @ 100 mm/sec, 20% power
M8
G0 X61Y22
; Layer C00
G1 X14S200F6000
G1 Y57
G1 X61
G1 Y22
G0 X127Y21
G1 X82
G1 Y103
G1 X127
G1 Y21
M9
G1 S0
M5
G90
; return to user-defined finish pos
G0 X0Y0
M2

bdring, 2024-08-04:
I think enable stays on until M5. The output should be off during G0

tomgale, 2024-08-04:
ah ok, might be my misunderstanding then, so the enable pin stays on, but the PWM output should drop to zero? I'll double check tomorrow, but I think the tube is staying powered, (or at least I can see the plasma in it). Is this the use case for DisableWithS0? I can try that too. Thanks!

MitchBradley, 2024-08-04:
Have you tried disable_with_s0: true  ?

regin92, 2024-08-05:
here are my settings on the same board and everything works fine for me

`board: MKS TinyBee V1
name: TinyBee corexy
meta: (17.01.2024) by Skorpi

#$config/filename=TinyBee.yaml $ESP115 =ON

kinematics:
  corexy:

stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  

axes:
  x:
    # X
    steps_per_mm: 80.000
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 6000.000
    max_travel_mm: 350.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: -3.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 4000.000
      settle_ms: 250
      seek_scaler: 1000.100
      feed_scaler: 3.100
      
    motor0:
      limit_pos_pin: gpio.33:low
      hard_limits: true
      pulloff_mm: 3.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0
        

  y:
    # Y
    steps_per_mm: 80.000
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 6000.000
    max_travel_mm: 350.000
    soft_limits: true
    homing:
      cycle: 3
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 4000.000
      settle_ms: 500
      seek_scaler: 1000.100
      feed_scaler: 3.100

    motor0:
      limit_pos_pin: gpio.32:low
      hard_limits: true
      pulloff_mm: 3.000
      stepstick:
        step_pin: I2SO.4
        direction_pin: I2SO.5
        disable_pin: I2SO.3

  z:
    # Z
    steps_per_mm: 930.000
    max_rate_mm_per_min: 1200.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 260.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: true

    motor0:
      hard_limits: false
      stepstick:
        step_pin: I2SO.7
        direction_pin: I2SO.8
        disable_pin: I2SO.6
        
        
        
i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  frequency_hz: 1000000

coolant:
  flood_pin: gpio.17 
  mist_pin: I2SO.20  
  delay_ms: 0
  

 
 

Laser:
  pwm_hz: 5000
  output_pin: gpio.13
  #enable_pin: I2SO.20
  #direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: true
  
  
start:
  must_home: false
`

tomgale, 2024-08-06:
**disable_with_s0: true** is the fix! thanks all. On the off chance that it helps someone else, the tube staying energised was because the little control panel which comes with the laser PSU overrides the PWM signal, which is a bit annoying.

closed 2024-08-06 by tomgale