bdring/FluidNC #1286: RS485 error:  Skipping key uart1 indent 1 this indent 0

maxoutny, 2024-08-04:
### Wiki Search Terms

RS485, H100, VFD

### Controller Board

6x CNC Controller

### Machine Description

Root 4 Lite

### Input Circuits

```yaml
VFD H100-1.5C2-1B

485+ connected to RS485 B on controlled
485- connected to RS485 A
Gnd to Gnd

Tried change combinations as recommended on Wiki page with no luck
```

### Configuration file

```yaml
board: 6x CNC Controller
name: 6x Default
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 10
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 320.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2:low
        direction_pin: I2SO.1
        disable_pin: I2SO.0:low

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 370.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.39
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5:low
        direction_pin: I2SO.4:low
        disable_pin: I2SO.7:low

    motor1:
      limit_neg_pin: NO_PIN   
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10:low
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8:low
        
  z:
    steps_per_mm: 500.00
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 59.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.5
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.36
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.13:low
        direction_pin: I2SO.12
        disable_pin: I2SO.15:low

  a:
    steps_per_mm: 200
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.18:low
        direction_pin: I2SO.17
        disable_pin: I2SO.16:low

  #c:      
   # steps_per_mm: 400.000
   # max_rate_mm_per_min: 5000.000
   # acceleration_mm_per_sec2: 100.000
   # max_travel_mm: 200.000
   # soft_limits: false
   # homing:
   #   cycle: 1
   #   positive_direction: false
   #   mpos_mm: 150.000
   #   feed_mm_per_min: 100.000
   #   seek_mm_per_min: 800.000
   #   settle_ms: 500
   #   seek_scaler: 1.100
   #   feed_scaler: 1.100

   # motor0:
   #   limit_neg_pin: NO_PIN
   #   limit_pos_pin: NO_PIN
   #   limit_all_pin: NO_PIN
   #   hard_limits: false
   #   pulloff_mm: 3.000
   #   standard_stepper:
   #     step_pin: I2SO.21
   #     direction_pin: I2SO.20
   #     disable_pin: I2SO.23

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  
probe:
  pin: gpio.32:low
  toolsetter_pin: gpio.33:low
  
# Using MOSFETs (Check Spindle Pin Usage
coolant:
  flood_pin: gpio.12
  mist_pin: gpio.4
  delay_ms: 0

start:
  must_home: false
  
## Begin Huanyang  
  uart1:
    txd_pin: gpio.15
    rxd_pin: gpio.16
    rts_pin: gpio.14
    baud: 9600
    mode: 8N1

# Huanyang:
    uart_num: 1
    modbus_id: 1
    tool_num: 0
    speed_map: 0=0% 0=25% 6000=25% 24000=100%
    off_on_alarm: false

# #begin PWM
#pwm:
#  pwm_hz: 5000
#  direction_pin: gpio.15
#  output_pin: gpio.13
#  enable_pin: gpio.14
#  disable_with_s0: false
#  s0_with_disable: true
#  spinup_ms: 0
#  spindown_ms: 0
#  tool_num: 0
#  speed_map: 0=0.000% 10000=100.000%
#  off_on_alarm: false
  
# #begin Laser
# Laser:
  # pwm_hz: 5000
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true
  
# #begin 10V
# 10V:
  # forward_pin: gpio.15
  # reverse_pin: gpio.14
  # pwm_hz: 5000
  # output_pin: gpio.13
  # enable_pin: NO_PIN
  # direction_pin: NO_PIN
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0  
  # tool_num: 0
  # speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  # off_on_alarm: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.8.0 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:root4l.yaml]
[MSG:ERR: Skipping key uart1 indent 1 this indent 0]
[MSG:ERR: Skipping key txd_pin indent 3 this indent 0]
[MSG:ERR: Skipping key rxd_pin indent 3 this indent 0]
[MSG:ERR: Skipping key rts_pin indent 3 this indent 0]
[MSG:ERR: Skipping key baud indent 3 this indent 0]
[MSG:ERR: Skipping key mode indent 3 this indent 0]
[MSG:ERR: Skipping key uart_num indent 3 this indent 0]
[MSG:ERR: Skipping key modbus_id indent 3 this indent 0]
[MSG:ERR: Skipping key tool_num indent 3 this indent 0]
[MSG:ERR: Skipping key speed_map indent 3 this indent 0]
[MSG:ERR: Skipping key off_on_alarm indent 3 this indent 0]
[MSG:INFO: Machine 6x Default]
[MSG:INFO: Board 6x CNC Controller]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:10us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (150.000,470.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2:low Dir:I2SO.1 Disable:I2SO.0:low]
[MSG:INFO:  X All Limit gpio.35]
[MSG:INFO: Axis Y (-220.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5:low Dir:I2SO.4:low Disable:I2SO.7:low]
[MSG:INFO:  Y Pos Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10:low Dir:I2SO.9:low Disable:I2SO.8:low]
[MSG:INFO:  Y2 Pos Limit gpio.34]
[MSG:INFO: Axis Z (-58.500,0.500)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13:low Dir:I2SO.12 Disable:I2SO.15:low]
[MSG:INFO:  Z All Limit gpio.36]
[MSG:INFO: Axis A (150.000,1110.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.18:low Dir:I2SO.17 Disable:I2SO.16:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Flood coolant gpio.12]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe gpio.32:low]
[MSG:INFO: Toolsetter gpio.33:low]
[MSG:INFO: Connecting to STA SSID:MySwHome_5GHz]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.0.104]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://Root4.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
<Idle|MPos:0.000,0.000,0.000,0.000|FS:0,0|WCO:250.000,-220.000,30.500,0.000>
[MSG:Files changed]
[MSG:INFO: Restarting]

  
Autoscroll
  
Ve
```

### User Interface Software

Webui, UGS

### What happened?

Updated VFD setting:

F011 100
F005 400

F001 default 0 (keyboard) changed to 2 (communication port)
F002 default 3 (operate potentiometer) changed to 2 (communication port)

### Other Information

[MSG:ERR: Ignored key uart1]
[MSG:ERR: Skipping key txd_pin indent 4 this indent 2]
[MSG:ERR: Skipping key rxd_pin indent 4 this indent 2]
[MSG:ERR: Skipping key rts_pin indent 4 this indent 2]
[MSG:ERR: Skipping key baud indent 4 this indent 2]
[MSG:ERR: Skipping key mode indent 4 this indent 2]
[MSG:ERR: Skipping key uart_num indent 4 this indent 2]
[MSG:ERR: Skipping key modbus_id indent 4 this indent 2]
[MSG:ERR: Skipping key tool_num indent 4 this indent 2]
[MSG:ERR: Skipping key speed_map indent 4 this indent 2]
[MSG:ERR: Skipping key off_on_alarm indent 4 this indent 2]

MitchBradley, 2024-08-04:
The messages indicate that the indentation of the uart1: section is wrong.  I don't know how to say it more clearly.

It also appears that you tried to comment out the huanyang: section but did not comment out its contents.  If the config file parser does not "see" the huanyang line, it cannot interpret the following lines which only make sense in context.

maxoutny, 2024-08-04:
Was reviewing the working config for the Root4 Lite and uncomment huanyang and uploaded it to the controller:

# Begin Huanyang  
  uart1:
    txd_pin: gpio.15
    rxd_pin: gpio.16
    rts_pin: gpio.14
    baud: 9600
    mode: 8N1

Huanyang:
    uart_num: 1
    modbus_id: 1
    tool_num: 0
    speed_map: 0=0% 0=25% 6000=25% 24000=100%
    off_on_alarm: false
Got error:

[MSG:ERR: Ignored key uart1]
[MSG:ERR: Skipping key txd_pin indent 4 this indent 2]
[MSG:ERR: Skipping key rxd_pin indent 4 this indent 2]
[MSG:ERR: Skipping key rts_pin indent 4 this indent 2]
[MSG:ERR: Skipping key baud indent 4 this indent 2]
[MSG:ERR: Skipping key mode indent 4 this indent 2]

maxoutny, 2024-08-04:
rts_pin is not connected to VFD, not sure where to connect it

maxoutny, 2024-08-05:
Changed config to:

uart2:
    txd_pin: gpio.15
    rxd_pin: gpio.16
    rts_pin: gpio.14
    baud: 9600
    mode: 8N1

Huanyang:
    uart_num: 2
    modbus_id: 1
    tool_num: 0
    speed_map: 0=0% 0=25% 6000=25% 24000=100%
    off_on_alarm: false
After restarting the firmware no error but, io14 LED blinking, the spindle not starting

bdring, 2024-08-05:
The 6x wiki page has an example rs485 config.

http://wiki.fluidnc.com/en/hardware/official/6x_CNC_Controller#rs485

maxoutny, 2024-08-05:
My config is the same now, if I use uart1 I have an error, if I use uart2 in config no error but the spindle not starting.
Changing the combination of wires doesn't help. What else should I try?

bdring, 2024-08-05:
" if I use uart1 I have an error"

What is the error? If the error is in the startup messages, please paste those in a reply.

MitchBradley, 2024-08-05:
It will take forever to solve your problem unless you ask questions more clearly.  For example

> Was reviewing the working config for the Root4 Lite

What exactly is "the working config for the Root4 Lite".  I searched the internet for "root4 lite fluidnc config" and did not find a clear answer, so I'm not sure exactly what config you are referring to.

> uart1:
> txd_pin: gpio.15
> rxd_pin: gpio.16

Indentation, i.e. the number of spaces at the beginning of a line, matters a lot for yaml, so if you paste in a way that loses the indentation, we cannot be sure what you have.  The way to preserve indentation is to mark the lines as a "code block" usint this control in the github form

[image]

Then use Preview to see if it worked right.

The error messages:
```
[MSG:ERR: Ignored key uart1]
[MSG:ERR: Skipping key txd_pin indent 4 this indent 2]
```
tell me that your indentation is bad.  `uart1:` must be at the beginning of a line with no spaces before it.  The lines from `txd_pin: `to `mode:` must be preceded by the same number (more than 0) of spaces, like this.

```
uart1:
  txd_pin: gpio.15
  rxd_pin: gpio.16
  rts_pin: gpio.14
  baud: 9600
  mode: 8N1
```

> My config is the same now

The same as what?

> rts_pin is not connected to VFD, not sure where to connect it

None of those pins - neither txd_pin, rxd_pin, nor rts_pin - are directly connected to the VFD.  Instead, they are connected to an on-board RS485 interface chip whose two outputs "RS485 A" and "RS485 B" should be connected to the VFD

[image]

>  if I use uart2 in config no error 

Either uart1 or uart2 should work.  I suspect that you fixed the indentation when you changed to uart2.

> but the spindle not starting.

It can be very difficult to get a VFD setup correctly, especially with RS485, because there are a lot of VFD settings that must be just right.  http://wiki.fluidnc.com/en/config/config_spindles#variable-frequency-drives-vfd-controlled-with-rs485  .  It is often much easier to use 0-10V control.

maxoutny, 2024-08-05:
Thank you for the clear explanation. Updated my config:

# Begin Huanyang
uart1:
  txd_pin: gpio.15
  rxd_pin: gpio.16
  rts_pin: gpio.14
  baud: 9600
  mode: 8N1

Huanyang:
  uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  off_on_alarm: false

No more errors on startup:

[MSG:INFO: FluidNC v3.8.0 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:root4l.yaml]
[MSG:INFO: Machine 6x Default]
[MSG:INFO: Board 6x CNC Controller]
[MSG:INFO: UART1 Tx:gpio.15 Rx:gpio.16 RTS:gpio.14 Baud:9600]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:10us Dsbl Delay:0us Dir Delay:1us
Idle Delay:255ms]
[MSG:INFO: Axis count 4]

But, the spindle not start after: S6000 M3 command

in console:
S6000 M3
ok

bdring, 2024-08-05:
The 6x wiki page mentions using LEDs to help debug the rs485.

Tell what happened?

maxoutny, 2024-08-05:
io14 - blinking
LED14 - is led up and off at the same time as io14 led.

Found additional settings on VFD for Communication function parameters:
F163  set to 1 (communication address)
F164  set to 1 (communication transmission speed 9600 default is 2- 19200)
F165 set to  0 (0- 8N1 for ASCII was 3- 8N1 for RTU)
F169 default 0 (0- Communication 0201H register adopts 1 bit decimal, 1-
Communication 0201H register adopt 2 bit decimal)

bdring, 2024-08-05:
As the wiki says...If LED 14 and **LED15** are blinking the controller is talking to the VFD, but the VFD is not responding.

There are a lot of registers that must be set on the VFD. See this wiki page for that.

http://wiki.fluidnc.com/en/config/config_spindles#huanyang-vfd-with-rs485

Make sure that you read all relevant wiki sections and pages before trying any config sections or wiring. It is easy to break the controller if you make mistakes like sending outputs in shorts, etc.

bdring, 2024-08-05:
8n1

maxoutny, 2024-08-05:
> Have you seen this? It is on the wiki.
>
> image.png (view on web)

Yes, TY. Trying to much it with my settings on my VFD, my not PD... on my
it is F..

MitchBradley, 2024-08-05:
Show a picture of your VFD

closed 2024-12-12 by bdring