bdring/FluidNC #1291: Problem: Low memory error while running G-code

zizheng1, 2024-08-08:
### Wiki Search Terms

"Low Memory"
"SD Card"
"LocalFS"

### Controller Board

MSK TinyBee

### Machine Description

Using MSK TinyBee to control four motors. UART is used to connect with an external serial display. I created a serial monitor software for that external display and I can type commands to control the board.
[image]

### Configuration file

```yaml
board: MKS TinyBee V1.0_001
name: K40 Laser
meta: (01.02.2022) by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  rts_pin: NO_PIN
  baud: 115200
  mode: 8N1

uart_channel1:
  uart_num: 1
  report_interval_ms: 400

  
axes:
  x:
    # X
    steps_per_mm: 640
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 50
    max_travel_mm: 411.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0
[... 439 lines cut ...]
z-36
y0
x0
y-198
z-60
m63 p4
g04 p1
z-53
y0
x-96
y-107
m62 p4
z0
x0
y-151
z-70
m63 p4 
y0
G28
M30
%

### Other Information

If I use a shorter G-code, I do not have such issue.

bdring, 2024-08-08:
Update the firmware. v3.7.17 addressed a memory issue.

zizheng1, 2024-08-08:
Thank you! I will update the firmware.

MitchBradley, 2024-08-08:
In addition to the version problem, there are other problems that can happen when running long gcode files from localfs.  Accessing FLASH as data can interfere with interrupts, which can cause problems with realtime step generation.  We recommend using SD cards for gcode files.

bdring, 2024-12-12:
Closing stale issue

closed 2024-12-12 by bdring