bdring/FluidNC #1295: Problem: if i conect on webserver while executing a job from memory , job halts

Ams-Laser, 2024-08-15:
### Wiki Search Terms

job halts

### Controller Board

MKS tinybee

### Machine Description

Diode laser gantry

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: AMS-Laser
meta: AMS 20/06/2024
planner_blocks: 120
kinematics: {}
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  x:
    steps_per_mm: 20
    max_rate_mm_per_min: 9000
    acceleration_mm_per_sec2: 800
    max_travel_mm: 550
    soft_limits: false
    motor0:
      limit_neg_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 10
      stepstick:
        step_pin: i2so.1
        direction_pin: i2so.2:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  y:
    steps_per_mm: 20
    max_rate_mm_per_min: 9000
    acceleration_mm_per_sec2: 800
    max_travel_mm: 780
    soft_limits: false
    motor0:
      limit_neg_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 10
      stepstick:
        step_pin: i2so.4
        direction_pin: i2so.5
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
    motor1:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 10
      stepstick:
        step_pin: i2so.7
        direction_pin: i2so.8:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
  z:
    steps_per_mm: 40
    max_rate_mm_per_min: 500
    acceleration_mm_per_sec2: 100
    max_travel_mm: 10000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 200
      seek_mm_per_min: 500
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      stepstick:
        step_pin: i2so.10
        direction_pin: i2so.11:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  cs_pin: gpio.5
  card_detect_pin: gpio.34:low
macros: {}
coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
Laser:
  pwm_hz: 1000
  output_pin: gpio.2:high:pd
  disable_with_s0: true
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
  enable_pin: NO_PIN
  off_on_alarm: true
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
start:
  must_home: false
```

### Startup Messages

```text
$ss
[MSG:ERR: Showing startup log from previous panic]
[MSG:INFO: FluidNC v3.8.0 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:cintos.yaml]
[MSG:INFO: Machine AMS-Laser]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.34:low freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-550.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO: Axis Y (-780.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8:low Disable:NO_PIN]
[MSG:INFO: Axis Z (0.000,10000.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.11:low Disable:NO_PIN]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.2:pd Freq:1000Hz Period:65535]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:LaserMaquinaria]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.0.101]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://Cintos1.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

webui

### What happened?

Machine was running a job from memory card and y connected to webui from laptop

MitchBradley, 2024-08-15:
FluidNC is supposed to reject attempts to load WebUI when a job is running.  Did that not happen?

The reason is because the ESP32 runs most of its instructions from FLASH, and loading WebUI requires FLASH access to serve image.html.gz from the same FLASH, interfering with timely instruction fetch.

MitchBradley, 2024-08-15:
The following information from your $SS output shows that FluidNC crashed (paniced):
```
[MSG:ERR: Showing startup log from previous panic]
```
In order to debug this, we will need the crash information that is displayed on FluidTerm when the panic happens.

Ams-Laser, 2024-08-27:
Hello
here is the otput from terminal at panic

Grbl 3.8 [FluidNC v3.8.0 (wifi) '$' for help]
[MSG:INFO: websocket auto report interval set to 300 ms]
[MSG:INFO: websocket auto report interval set to 50 ms]
[MSG:INFO: WebUI: Request from 192.168.0.100]
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40101859  PS      : 0x00060c30  A0      : 0x800f776e  A1      : 0x3ffb2150  
A2      : 0x3ffc5820  A3      : 0x3ffc5724  A4      : 0x00000000  A5      : 0x3ffc572c  
A6      : 0x00000001  A7      : 0x3ffc56d8  A8      : 0x3ffc5728  A9      : 0x00000003  
A10     : 0xffffffff  A11     : 0x40866667  A12     : 0x00000002  A13     : 0x00000003  
A14     : 0x00000000  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000030  LBEG    : 0x4008bf40  LEND    : 0x4008bf4b  LCOUNT  : 0x00000000  

Backtrace: 0x40101856:0x3ffb2150 0x400f776b:0x3ffb21b0 0x400f7d7a:0x3ffb21d0 0x400f804b:0x3ffb2210 0x400e86bc:0x3ffb2250 0x40125a9d:0x3ffb2290

ELF file SHA256: e6173bfb5d10076b

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.8.0 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:LaserMaquinaria]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.0.101]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://Cintos1.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.8 [FluidNC v3.8.0 (wifi) '$' for help]
[MSG:ERR: Configuration is invalid. Check boot messages for ERR's.]

Ams-Laser, 2024-08-27:
if you need to reproduce error just start a job from a file in webui and then refresh the webpage

MitchBradley, 2024-08-27:
This is turning out to be very tricky to solve.  We have long known that reloading WebUI while running a job is inherently problematic, since it requires a lot of FLASH access to serve up index.html.gz, and ESP32 has deep problems with FLASH data access during real-time operations involving interrupts.  At one point we solved that by making such reloads serve up a "no can do that" web page.  That sort of worked, but then I added the ability to respond to file access with a "the file that you requested has not changed, so use a cached copy if you have it".  That almost works, but then after WebUI restarts, it spews a lot of commands to configure itself, and those interfere with pre-existing motion.  Not only that, but the websocket that the original instance of webui was using to report status and whatnot gets closed on the reload, so the existing job loses its output/status channel.

Ideally, one would just disable the ability to issue the reload request entirely, but the reload button is at the browser level, outside of the control of the web page that is being displayed.

jeyeager65, 2024-08-28:
I wonder if you could use the beforeunload event to warn the user.  It certainly wouldn't be a perfect solution but I think this is how web pages warn about unsaved changes before navigating away.  There could be some kind of flag so it only does it if a job is running.

MitchBradley, 2024-08-28:
As it turns out, I am working on exactly that.  It helps, but I still have to do some stuff on the FluidNC end to unwind jobs in case the user bypasses the beforeunload dialog and refreshes anyway, and then wants to stop the job.  It's a multi-pronged problem.

MitchBradley, 2024-08-28:
[image]

MitchBradley, 2024-08-28:
It also needs to be tested with WebUI 3.  The proliferation of UIs creates a combinatorial nightmare.

MitchBradley, 2024-08-28:
The complete fix included only the reload-blocking dialog, but also changes to FluidNC to handle the case where the user clicks "Reload" in the dialog or attempts to load WebUI from another device.  In that case, you get a set of simple screens that let you pause, resume, or stop the program before reloading:

[image]

[image]

[image]

It would be nice if someone would volunteer to make a wiki page explaining all this.  The "Reload Site? Changes you made might not be saved" message is not ideal (does not explain the real situation), but browsers don't let you change it for security reasons.

MitchBradley, 2024-08-28:
The fix to WebUI might also apply to WebUI3 when using the tablet extension (but probably not with the main WebUI3 dashboard).  It would be great if someone could test that.  I have other things I want to work on.

masterxq, 2024-09-17:
I wrote a similar software and it seems the problem is slightly different. I had the same issue when trying to load something from NVS or SPIFFS. That's why I decided to stop using them and instead wrote a CMake script that embeds all web files from a given Path directly into the binary. Since then, I haven't had these types of issues anymore.

It might be worth considering, as it would also reduce the process of flashing FluidNC to just one step. Of course, the downside is that the web files would no longer be separate from the firmware. I don’t know FluidNC well enough to assess whether that separation is necessary or not.

But this would fix the problem completely :)

MitchBradley, 2024-09-18:
There are some tradeoffs:
1. The partition map would have to change to make room for the now-larger program image.  There isn't enough FLASH space for two of the larger images, so we would have to give up the OTA firmware update feature.  That could be mitigated by having asymmetric images where you reboot into the smaller, less-capable image, and use it to OTA into the larger partition, but that would require retooling of the OTA process and the maintenance of two different images.
2. There are now two different versions of WebUI - v2 and v3.  Each has its proponents, so we would need more binaries.  In some respects the effort of doing that would be mitigated by reduction of complexity in other areas, but nevertheless it would require a fair amount of retooling of build, release, and install scripting.

I am less than enthusiastic about doing either version of the retooling, as I find release engineering to be the ultimate tedium.  It it the sort of thing that one often has to do, and there is no joy or glory in it whatsoever.  But if you wanted to take on the full task, as opposed to just throwing your scripts over the wall and leaving me to do the rest of the dirty work, that would be great.

4.

masterxq, 2024-09-18:
By the way, what’s stored in littleFS besides the config?
Your points could potentially be resolved quite easily, and in the process, a lot of future work could be saved. However, this would require a more in-depth exchange. I'd need to understand more about the user requirements, workflows, and the team's skillset.
If you wish that, I could imagine cleaning things up for you and briefing you on what that would actually entail. I find your project super impressive, but there’s so much untapped potential that I’m sure leads to a lot of painstaking work and micromanagement. What makes it even more impressive that it's as good as it is! There must be an enormous amount of effort and love behind it :)

MitchBradley, 2024-09-18:
We could talk more on Discord if you wish.

commit 74972e13 (2024-09-25): Fix #1295 - Crash when reloading WebUI while running (#1334)

* Fix #1295 - Crash when reloading WebUI while running

* Update index.html.gz
closed 2024-09-25 by MitchBradley