bdring/FluidNC #1298: Problem: ALARM 9 homing fail

landisjohn, 2024-08-22:
### Wiki Search Terms

alarm, error, homing, limit switch 

### Controller Board

6x v1.2

### Machine Description

CL57T drivers, normally open limit switches at both ends of each axis wired in parallel, X homes in the negative direction, YZ  in positive direction

### Configuration file

```yaml
board: 6x
name: 6x Default
stepping:
  engine: I2S_STREAM
  idle_ms: 254
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 236.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.2:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 207.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 207.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.26:low:pu
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  z:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 100.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 100.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN    
      limit_pos_pin: gpio.33:low
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
        
  a:
    steps_per_mm: 53.400
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  b:
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16
        
  c:      
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23
        
       
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  
probe:
  pin: gpio.39:low
  toolsetter_pin: gpio.36:low
  
# Using MOSFETs (Check Spindle Pin Usage
coolant:
  flood_pin: gpio.12
  mist_pin: gpio.4
  delay_ms: 0

start:
  must_home: false
  
## Begin Huanyang  
# uart1:
  # txd_pin: gpio.15
  # rxd_pin: gpio.16
  # rts_pin: gpio.14
  # baud: 9600
  # mode: 8N1

# Huanyang:
  # uart_num: 1
  # modbus_id: 1
  # tool_num: 0
  # speed_map: 0=0% 0=25% 6000=25% 24000=100%
  # off_on_alarm: false

# #begin PWM
pwm:
  pwm_hz: 5000
  direction_pin: NO_PIN
  output_pin: gpio.13
  enable_pin: gpio.14
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 10000=100.000%
  off_on_alarm: false
  
# #begin Laser
# Laser:
  # pwm_hz: 5000
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true
  
# #begin 10V
# 10V:
  # forward_pin: gpio.15
  # reverse_pin: gpio.14
  # pwm_hz: 5000
  # output_pin: gpio.13
  # enable_pin: NO_PIN
  # direction_pin: NO_PIN
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0  
  # tool_num: 0
  # speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  # off_on_alarm: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.8.1 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:6x_default (1).yaml]
[MSG:INFO: Machine 6x Default]
[MSG:INFO: Board 6x]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:254ms]
[MSG:INFO: Axis count 6]
[MSG:INFO: Axis X (0.000,236.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.2:low:pu]
[MSG:INFO: Axis Y (0.000,207.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y Pos Limit gpio.26:low:pu]
[MSG:INFO: Axis Z (0.000,100.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Z Pos Limit gpio.33:low]
[MSG:INFO: Axis A (150.000,1110.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO:  A Neg Limit gpio.32:low]
[MSG:INFO: Axis B (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]
[MSG:INFO:  B Neg Limit gpio.35:low]
[MSG:INFO: Axis C (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.21 Dir:I2SO.20 Disable:I2SO.23]
[MSG:INFO:  C Neg Limit gpio.34:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: PWM Spindle Ena:gpio.14 Out:gpio.13 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant gpio.12]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe gpio.39:low]
[MSG:INFO: Toolsetter gpio.36:low]

Grbl 3.8 [FluidNC v3.8.1 (noradio) '$' for help]
```

### User Interface Software

ugs, candle, fluidnc tablet

### What happened?

When I attempt to home in UGS or candle or fluidnc tablet, the z axis motor stops at the limit switch but doesnt pull off, theres a couple second delay , then I get ALARM 9 homing fail approach.

homing works fine for individual axes from the fluidnc dashboard and fluidterm, only when trying to home in the sender softwares or the fluidnc tablet do get the alarm.

MitchBradley, 2024-08-22:
Try a larger pulloff.  1mm can be close to the hysteresis of switches.  If that doesn't fix it, send $message/level=debug and post the debug messages that show up in the console.

landisjohn, 2024-08-22:
`[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.8.1 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:6x_default (1).yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine 6x Default]
[MSG:INFO: Board 6x]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:254ms]
[MSG:INFO: Axis count 6]
[MSG:INFO: Axis X (0.000,236.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.2:low:pu]
[MSG:INFO: Axis Y (0.000,207.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y Pos Limit gpio.26:low:pu]
[MSG:INFO: Axis Z (0.000,100.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Z Pos Limit gpio.33:low]
[MSG:INFO: Axis A (150.000,1110.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO:  A Neg Limit gpio.32:low]
[MSG:INFO: Axis B (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]
[MSG:INFO:  B Neg Limit gpio.35:low]
[MSG:INFO: Axis C (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.21 Dir:I2SO.20 Disable:I2SO.23]
[MSG:INFO:  C Neg Limit gpio.34:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: PWM Spindle Ena:gpio.14 Out:gpio.13 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant gpio.12]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe gpio.39:low]
[MSG:INFO: Toolsetter gpio.36:low]
[MSG:DBG:  B Neg Limit 0]

[MSG:DBG:  C Neg Limit 0]
Grbl 3.8 [FluidNC v3.8.1 (noradio) '$' for help]
[MSG:DBG:  Z Pos Limit 0]
[MSG:DBG: Run after_reset]
[MSG:DBG:  A Neg Limit 0]
[MSG:DBG: Run startup_line0]
[MSG:DBG:  Y Pos Limit 0]
[MSG:DBG: Run startup_line1]
[MSG:DBG:  X Neg Limit 0]
`

bdring, 2024-08-22:
Send `?` and then `$H` and show us the debug response.

landisjohn, 2024-08-22:
$h
[MSG:DBG: Homing Cycle ZBC]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,0.000]
[MSG:DBG: Planned move to 0.000,0.000,220.000 @ 1385.641]
[MSG:DBG:  Z Pos Limit 1]
[MSG:DBG: Homing limited Z]
[MSG:DBG: Homing replan with BC]
[MSG:DBG: Starting from 0.000,0.000,3.013]
[MSG:DBG: Planned move to 0.000,0.000,3.013 @ 1131.371]
[MSG:DBG: CycleStop FastApproach]
<Home|MPos:0.000,0.000,3.013,0.000,-223.021,-223.021|FS:0,0|Pn:Z|Ov:100,100,100>
[MSG:INFO: ALARM: Homing Fail Approach]
ALARM:9
[MSG:DBG: Run after_homing]

bdring, 2024-08-22:
- Test the limit switches manually to make sure each switch activates and is indicating the correct axis.
- Home axes individually with `$HX` etc. Report how each behaved and give the reponses to any that fail.

Be sure to thoroughly re-read this wiki page.

http://wiki.fluidnc.com/en/support/setup/limit_switches

landisjohn, 2024-08-22:
pressing the switches manually all report the correct axis.

$HZ, $HX, $HY home correctly 

I've tried everything that makes any sense to me on the wiki, change hard and soft limits, high and low on limit pin, added pu, pd 

does the disable pin play any role in this? I had to leave them off to get the closed loop drivers to work.

MitchBradley, 2024-08-22:
I'm not sure that homing B and C axes has ever been tested.  You can probably count the number of people who have more than 4 axes on one hand.

bdring, 2024-08-22:
Did you try homing AB and C?

MitchBradley, 2024-08-22:
Disable should not be an issue.  The homing process does not interact with it.

landisjohn, 2024-08-22:
did not attempt homing abc as I have nothing hooked up. is it best to delete them from the config file?

tried it, nothing happens

MitchBradley, 2024-08-22:
If you do not have ABC axes you should not include them in the config file, and they certainly should not have homing cycles defined.  If they are defined, $H is going to try and home them.  If nothing is hooked up, homing will try to move those axes toward the limit switches, but nothing will happen and the corresponding limit switches will not activate, so the homing process will fail.

landisjohn, 2024-08-22:
deleted A, B, C from the config file. homing is now working as it should. 

Thanks for all the help.

closed 2024-08-22 by landisjohn

bdring, 2024-08-22:
Thanks for the donation.

[Join us on Discord](https://discord.gg/xcgTy3AB).