bdring/FluidNC #1299: Podłączenie step_pin: gpio.12 do driverTB6600 (wspólna anoda)- ESP32 nie startuje LED blink

qulet, 2024-08-23:
### Wiki Search Terms

Konfigurowanie pinów

### Controller Board

ESP32 38 PIN module

### Machine Description

Router btamowy sterowniki TB6600 hard limit XYZ wrzeciono z falownikiem.

### Input Circuits

```yaml
Nie znanm natury problemu
```

### Configuration file

```yaml
'''yami
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.17:low:pu
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.25
    motor1:
      null_motor:

  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.4:low:pu
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      null_motor:

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: gpio.16:low:pu
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
    mist_pin:  gpio.21

        
probe:
  pin: gpio.32:low:pu

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 10000=100%
```

### Startup Messages

```text
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$SS
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
[MSG:ERR: Showing startup log from previous panic]
[MSG:INFO: FluidNC v3.8.2-pre https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:3axis_v4.yaml]
[MSG:ERR: Configuration parse error on line 15: Expected a float value like 123.456]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:>
[MSG:Files changed]
[MSG:Files changed]
```

### User Interface Software

webui

### What happened?

Po odlaczeniu sygnału step+ od płytki ESP  Fluid startuje.  Próba przypisania innego pinu gipo kończy się niepowodzeniem.

### Other Information

Fluid ma sterować Maszyną XYZ z limitami XYZ i wrzecionem z falownikiem, lub laserem LED.
Mój router aktualnie jest sterowany Mach3

qulet, 2024-08-23:
[image]

MitchBradley, 2024-08-23:
Which step+ pin?  There are three of them.

MitchBradley, 2024-08-23:
It also appears that you have something - perhaps A direction - connected to GPIO36.  GPIO36 cannot be used as an output.  There are many references on the internet that describe the restrictions of some ESP32 pins.

qulet, 2024-08-24:
defaltowy plik  3axis_V4.yami  wypiete kable od sterowników XYZA  Boot ESP32 Fluid ON

[image]

$SS :
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
$SS
[MSG:INFO: FluidNC v3.8.2-pre https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:3axis_v4.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13:low]
[MSG:INFO: Axis X (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.17:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-990.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.16:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:Orange_Swiatlowod_D290]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connected - IP is 192.168.1.22]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: PWM Spindle Ena:gpio.22 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant gpio.25]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe gpio.32:low:pu]
ok
Send Command...
 
 
 Podłaczony tylko sterownik osi X  
  direction_pin: gpio.14
 step_pin: gpio.12
 EESP NO BOOT Fluid OFF 
 
[image]

qulet, 2024-08-24:
Mam dwa pytania do ekspertów.
1. Czy na gołym module ESP32 można sterować 4  asiami XYZ A z krańcówkami na każdej osi?
2. Jaka kompilacja Fluid CNC jest potrzebna by pominąć Gipo12 = NCN

MitchBradley, 2024-08-24:
Regarding GPIO 12:
http://wiki.fluidnc.com/hardware/esp32_pin_reference#usable-io-pins

Unless you perform the one-time step of changing the strapping pin as documented above, you can only use GPIO 12 in a way that does not pull it up during boot.  In other words, if there is an external connection to GPIO 12, the hardware connected to it must not override the internal pulldown that controls the boot mode.

This has nothing to do with the way that FluidNC is compiled.  It is an inherent characteristic of the ESP32 that is independent of any software that is loaded.

FluidNC's use of GPIO 12 for some function, or lack of use of GPIO 12, is controlled by the "config.yaml" configuration file that you  install.  Regardless of whether you choose to use GPIO 12 or not, the way that GPIO 12 is connected to something outside controls the boot process.  That happens long before any FluidNC code runs. It is up to you to use properly-designed hardware.  There are many suitable controller boards that other people have already designed as per http://wiki.fluidnc.com/en/hardware/existing_hardware , as well as some guidelines at http://wiki.fluidnc.com/en/support/controller_design_guidelines .

We spent a lot of time designing and debugging controllers, including open source ones that can be purchased directly from contract manufacturers like Elecrow.  We documented the design considerations in the guidelines page. We do not have time to repeat the information bit by bit.

Re limit switches:  If you use the search term "switch" in the FluidNC Wiki search box, you will find several pages with information about limit switches.  To connect a limit switch to a bare ESP32, you will either need an external pullup resistor, or you will need to use one of the ESP32 GPIOs that is capable of internal pullup and configure it with the :pu attribute.

If you need additional design help, please consider becoming a project sponsor.

closed 2024-08-24 by MitchBradley

qulet, 2024-08-26:
Dla Wszystkich którzy mierzą się z tym problemem .
Sterowniki TB6600 myszą być podłączone w trybie WSPÓLNA KATODA . To rozwiązuje problem. Takiej odpowiedzi oczekiwałem od ekspertów a nie namawiania do zakupu profesjonalnych dedetykowanych PCB.

MitchBradley, 2024-08-26:
1. It is not correct that TB6600 drivers should be connected in common cathode mode.  That happens to solve your problem for your controller and your particular TB6600 driver boxes, but it is not generally true.  With a properly-designed controller, common anode mode works fine.  When designing a controller, it is important to pay attention to loading characteristics of the special ESP32 strapping pins, in consideration of the electrical characteristics of whatever drivers you might happen to use.  Not all TB6600 boxes are the same.  TB6600 is a chip with specified characteristics, but the boxes containing them have differing circuitry between the connector pins and the chip, and that circuitry is often poorly-documented.  Therefore, making blanket statements about "TB6600 drivers" is incorrect.
2. The "expert" who answered your question provided an accurate answer to the question that you posed, indicating that the failure to boot resulted from loading on GPIO12.  The solution that you found was specific to particular details of your system.
3. You seem to expect a lot from people who provide, at no cost, software who development cost ran into hundreds of thousands of dollars at market rates for programmers.  You expect them not only to provide no-cost software, but also system design/debugging help for free.