bdring/FluidNC #1307: Problem:

jasonwebb, 2024-09-02:
### Wiki Search Terms

motor lock tmc2209

### Controller Board

4 Axis TMC2209 CNC Controller

### Machine Description

No machine yet, just smoke testing the electronics components and learning FluidNC. One stepper motor installed on the X axis, with a 12V PSU wired to the motor power terminal block.

[image]

### Configuration file

```yaml
name: TMC2209 XYZA Huany
board: 4 Axis TMC2209 CNC Controller

# SPI necessary for SD card
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
  
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

# I2SO necessary for TMC2209 UART control
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
  
# UART1 for the TMC2209s
uart1:
  txd_pin: gpio.16
  rxd_pin: gpio.4
  baud: 115200
  mode: 8N1

stepping:
  engine: I2S_STREAM
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750

    motor0:
      tmc_2209:
        uart_num: 1
        addr: 0
        direction_pin: i2so.1
        step_pin: i2so.2
        disable_pin: i2so.0
        use_enable: false
        r_sense_ohms: 0.110
        run_amps: 1.500
        hold_amps: 1.000
        microsteps: 32
        
        # TMC2209 features
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep

  y:
    steps_per_mm: 53.33
    max_rate_mm_per_min: 18000
    acceleration_mm_per_sec2: 950
    max_travel_mm: 410

    motor0:
      tmc_2209:
        uart_num: 1
        addr: 1
        direction_pin: i2so.4
        step_pin: i2so.5
        disable_pin: i2so.7
        use_enable: false
        r_sense_ohms: 0.110
        run_amps: 1.500
        hold_amps: 1.000
        microsteps: 32
        
        # TMC2209 features
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        
start:
  must_home: false
```
```

### Startup Messages

```text
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.8.2 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: X Axis tmc_2209 homing current not in config. Using run current]
[MSG:WARN: Y Axis tmc_2209 homing current not in config. Using run current]
[MSG:INFO: Machine TMC2209 XYZA Huany]
[MSG:INFO: Board 4 Axis TMC2209 CNC Controller]
[MSG:INFO: UART1 Tx:gpio.16 Rx:gpio.4 RTS:NO_PIN Baud:115200]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-750.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 UART1 Addr:0 CS:NO_PIN Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0 R:0.110]
[MSG:INFO: Axis Y (-410.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 UART1 Addr:1 CS:NO_PIN Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7 R:0.110]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.8 [FluidNC v3.8.2 (noradio) '$' for help]
```
```

### What happened?

Upon powering up (motors first, then ESP32 board), the connected motor (X) does not lock/engage. I followed the [Debug No Motion](http://wiki.fluidnc.com/en/support/setup/motion#debug-no-motion) section of the Motor Setup Flowchart in the wiki, and have not been able to get the motor working.

I'm pretty sure [the stepper motor](https://www.amazon.com/dp/B00PNEQKC0?ref=ppx_yo2ov_dt_b_fed_asin_title) is wired up correctly, and I've tried a few different wiring combos just in case. I've identified the 2 pairs of coil wires using a continuity checker and by the "twist test" (holding 2 wires together and manually spinning the shaft while unpowered). The wire colors correspond to the coils like so:
* Red = B+
* Blue = B-
* Green = A-
* Black = A+

I've verified that the TMC2209 drivers are receiving 12V for their VMOT pins by measuring the voltage across the large cap C30.

I've verified that [my ESP32 board (Espressif ESP32-DevKitC-32E)](https://www.amazon.com/dp/B09MQJWQN2?ref=ppx_yo2ov_dt_b_fed_asin_title) uses the WROOM chip and not WROVER - because I ordered a WROVER one first and would always get startup errors about the X and Y drivers not responding.

I've tried manually sending `G0` commands via FluidTerm to the board with a wide range of values in case the steps per mm are just way off, but the shaft doesn't spin.

### Other Information

Grbl settings from `$$`
```
$22=0
$21=0
$20=0
$102=80.000
$101=53.330
$100=400.000
$112=1000.000
$111=18000.000
$110=5000.000
$122=25.000
$121=950.000
$120=150.000
$132=1000.000
$131=410.000
$130=750.000
$10=1
$30=0
```

MitchBradley, 2024-09-02:
Good bug report, thanks for providing clear information.  For future reference, we typically don't need the $$ info since it is all available in the config file and the startup messages.

Does the motor lock if you set idle_ms to 255?

bdring, 2024-09-02:
After setting idle_ms to 255 and restarting, try sending $ME and $MD to see if either locks.

Note: The power wiring is probably not an issue now, but will need to be increase to at least 16awg

jasonwebb, 2024-09-02:
> Does the motor lock if you set idle_ms to 255?

Nope. Restarted the ESP32 board a couple times to confirm.

> try sending $ME and $MD to see if either locks.

Both get an "ok" response message, but the motor does not lock. I'm not 100% confident I have the right config setup for the enable/disable feature, but it does match the [sample config on the board's wiki page](http://wiki.fluidnc.com/en/hardware/official/TMC2209_4_Axis#config-information). FWIW I've tried setting `use_enable` to both `true` and `false`, and with and without the `shared_stepper_disable_pin: NO_PIN` setting.

> Note: The power wiring is probably not an issue now, but will need to be increase to at least 16awg

Noted!

bdring, 2024-09-02:
I use that controller for my full time FluidNC test rig. This is my x axis config. You can ignore the low on the dir pin.

```yaml
x:
  steps_per_mm: 400.000000
  max_rate_mm_per_min: 1500.000000
  acceleration_mm_per_sec2: 100.000000
  max_travel_mm: 1000.000000
  soft_limits: false
  homing:
    cycle: 2
    allow_single_axis: true
    positive_direction: false
    mpos_mm: 0.000000
    feed_mm_per_min: 700.000000
    seek_mm_per_min: 1500.000000
    settle_ms: 250
    seek_scaler: 1.100000
    feed_scaler: 1.100000

  motor0:
    limit_neg_pin: 'gpio.35:low'
    limit_pos_pin: NO_PIN
    limit_all_pin: NO_PIN
    hard_limits: true
    pulloff_mm: 2.000000
    tmc_2209:
      addr: 0
      cs_pin: NO_PIN
      uart_num: 1
      step_pin: I2SO.2
      direction_pin: 'I2SO.1:low'
      disable_pin: I2SO.0
      r_sense_ohms: 0.110000
      run_amps: 1.800000
      hold_amps: 0.500000
      microsteps: 16
      toff_disable: 0
      toff_stealthchop: 5
      use_enable: false
      run_mode: StealthChop
      homing_mode: StealthChop
      homing_amps: 1.500000
      stallguard: 0
      stallguard_debug: false
      toff_coolstep: 3
   ```

jasonwebb, 2024-09-02:
I just copied that into my config over my X section, but got the same result. I also copied it over for my Y axis, replacing all the pin assignments to match what I had before. In each case I restarted the ESP32 a few times, tried $ME and $MD, and tried G0 cmds.

I also tried out a few different stepper motors, including a much smaller one just in case it was a current draw or internal wiring problem.

I'm attaching a closer pic of my ESP32 module, in case it sparks any questions.

Is the disable/enable handled via UART or through a direct pin connection? Is it possible that the pins on the ESP32 or some circuitry in between the chips is bad? So strange that the ESP32 can connect and send cmds to the TMC2209s over UART, and neither chip seems to think anything is wrong.

[image]
[image]

MitchBradley, 2024-09-02:
1. Measure the voltage on the + side of the electrolytic capacitor, being careful not to touch the metal case of the capacitor with the probe while doing so.  The + side is the side near the TMC2209 chip, opposite from the black semicircle on top.
2. Measure the voltages on the stepper connector pins.
3. Measure the voltage on pin 15 of U1, being careful to touch only that pin - do not let the probe touch two adjacent pins at once.

MitchBradley, 2024-09-02:
Also try with the second motor connector to see if the first TMC2209 could be faulty or poorly-soldered.

jasonwebb, 2024-09-03:
1. 11.9V for the + pin of C30 and C31
2. 12.3V for all four pins on both X and Y terminal blocks
3. 3.6V

The 12.3V on the terminal blocks seems interesting, right? Does that indicate that the motors are actually in a lock state, but maybe not getting enough current? I'm still able to spin the shaft by hand as well as when it's unpowered.

MitchBradley, 2024-09-03:
If the terminal blocks have 12.3 V on all the pins and the motors spin easily, the only things I can think of are either
1. The stepper wires are not inserted correctly in the terminal blocks, or
2. The TMC2209 drivers are somehow being configured for very low current, or
3. The current sense resistors are faulty so the driver thinks it is supplying more current than it is.

MitchBradley, 2024-09-03:
I suppose you could try 1 for hold_amps instead of 0.5 - but 0.5 should be enough to lock.  I guess you could put an ammeter in series with a motor winding.

MitchBradley, 2024-09-03:
Actually, I would expect that at least one of the terminals would be at some voltage other than 12.3, enough differential to force 0.5A of current through the motor.

jasonwebb, 2024-09-03:
I've increased the `hold_amps` for the X axis to 1.0. I've also verified that `r_sense` for both motors is `0.11`. Just for funsies I measured the resistance of the big 0.11 Ohm resistors near the X and Y terminal blocks (R24, R20, R25, R21) and got `0.9` Ohms -- but my multimeter only goes down to the 100s range with 1 decimal place, so maybe that's just a red herring.

I just tried a couple different ways of wiring up the motor wires. From left to right in the top-down pic you posted above, I've tried these, with the same results each time:
1. `A-  A+  B-   B+`
2. `A-  A+  B+  B-`
3. `A+  A-  B+  B-`
4. `A+  A-  B-  B+`

jasonwebb, 2024-09-03:
Given that (1) the config and status messages seem to be alright, (2) both the X and Y motor outputs seem to be acting identically and (3) the ESP32 seems to be able to talk to the TMC2209s OK, I'm starting to think that maybe my ESP32 board is not quite right. Seems more likely that the 1 ESP32 board is wonky vs both/all of the TMC2209s 🤔 They're cheap and easy to swap out though, so I'm down for getting another to try out.

Can either of you link to a known-good board/vendor that I can pick up for testing?

FWIW I bought the 32E variant of this: https://www.amazon.com/dp/B09MQJWQN2

bdring, 2024-09-03:
How about sending the board to me to figure out the problem?

bdring, 2024-09-03:
I measured the stepper driver terminals. They are all at VMot in disabled mode. 

It is possible there are problems with the I2SO pin expander circuit. It is used to control the stepper signals. That is used on most FluidNC controllers. It is a relatively simple circuit and I don't recall any previous issues other than some DIY breadboards.

bdring, 2024-09-03:
Forcing the i2so chips could damage them and the esp32.

Hot connecting should damage them, but it have done it several times without total failure.

MitchBradley, 2024-09-03:
It is more likely that the I2SO shift register chips are not working.

bdring, 2024-09-06:
I got your controller. I could not get it to work. I replaced the ESP32 and it appears to work. I'll do some more testing with your config and send it back.

jasonwebb, 2024-09-06:
Ahhh that's so annoying haha. I wonder if the chipset is a little different, or maybe the part was defective in some way 🤔 Or maybe I fried something without realizing it from plugging/unplugging it at the wrong times. Thanks for figuring that out!!