bdring/FluidNC #1310: Problem: 6x CNC Control does not start.

MichelBattistella, 2024-09-06:
### Wiki Search Terms

- Requesting help for FluidNC 

### Controller Board

6x CNC Controler V1.2

### Machine Description

Controlling a CNC with the “6x controller V1.2” card.
The configuration of the machine is:
X: Motor driver DM556 motor nema 23 (2.9 amp)
Y1, Y2 in gantry Motor driver DM556 motor nema 23 (2.9 amp) (2x)
Z1, Z2 in gantry Motor driver DM556 motor nema 23 (2.9 amp) (2x)
+ Fluidial
External Spindle 500W 12000rp

### Configuration file

```yaml
board: 6x CNC Controller
name: CNC_C_BEAM
## version 02/08/2024
stepping:
engine: I2S_STREAM
idle_ms: 255
pulse_us: 4
dir_delay_us: 1
disable_delay_us: 0
segments: 12
axes:
shared_stepper_disable_pin: NO_PIN
shared_stepper_reset_pin: NO_PIN
homing_runs: 2
x:
steps_per_mm: 800.000
max_rate_mm_per_min: 5000.000
acceleration_mm_per_sec2: 200.000
max_travel_mm: 850
soft_limits: true
homing:
cycle: 2
allow_single_axis: true
positive_direction: false
mpos_mm: -8.450
feed_mm_per_min: 100.000
seek_mm_per_min: 500.000
settle_ms: 500
seek_scaler: 1.100
feed_scaler: 1.100
motor0:
limit_neg_pin: NO_PIN
limit_pos_pin: gpio.32:low
limit_all_pin: NO_PIN
hard_limits: false
pulloff_mm: 1.000
standard_stepper:
step_pin: I2SO.2
direction_pin: I2SO.1
disable_pin: I2SO.0
y:
steps_per_mm: 800.000
max_rate_mm_per_min: 5000.000
acceleration_mm_per_sec2: 200.000
max_travel_mm: 430
soft_limits: true
homing:
cycle: 2
allow_single_axis: true
positive_direction: true
mpos_mm: 430.000
feed_mm_per_min: 100.000
seek_mm_per_min: 500.000
settle_ms: 500
seek_scaler: 1.100
feed_scaler: 1.100
# Y1
motor0:
limit_neg_pin: NO_PIN
limit_pos_pin: gpio.35:low
limit_all_pin: NO_PIN
hard_limits: false
pulloff_mm: 1.000
standard_stepper:
step_pin: I2SO.5
direction_pin: I2SO.4:low
disable_pin: I2SO.7
# Y2
motor1:
limit_neg_pin: NO_PIN
limit_pos_pin: gpio.34:low
limit_all_pin: NO_PIN
hard_limits: false
pulloff_mm: 1.000
standard_stepper:
step_pin: I2SO.10
direction_pin: I2SO.9:low
disable_pin: I2SO.8
z:
steps_per_mm: 800.000
max_rate_mm_per_min: 5000.000
acceleration_mm_per_sec2: 200.000
max_travel_mm: 260.000
soft_limits: true
homing:
cycle: 1
allow_single_axis: true
positive_direction: true
mpos_mm: 260
feed_mm_per_min: 100.000
seek_mm_per_min: 500.000
settle_ms: 500
seek_scaler: 1.100
feed_scaler: 1.100
# Z1
motor0:
limit_neg_pin: NO_PIN
limit_pos_pin: gpio.39:low
limit_all_pin: NO_PIN
hard_limits: false
pulloff_mm: 1.000
standard_stepper:
step_pin: I2SO.13
direction_pin: I2SO.12:low
disable_pin: I2SO.15
# Z2
motor1:
limit_neg_pin: NO_PIN
limit_pos_pin: gpio.36:low
limit_all_pin: NO_PIN
hard_limits: false
pulloff_mm: 2.100
standard_stepper:
step_pin: I2SO.18
direction_pin: I2SO.17:low
disable_pin: I2SO.16
i2so:
bck_pin: gpio.22
data_pin: gpio.21
ws_pin: gpio.17
spi:
miso_pin: gpio.19
mosi_pin: gpio.23
sck_pin: gpio.18
sdcard:
card_detect_pin: NO_PIN
cs_pin: gpio.5
frequency_hz: 8000000
control:
safety_door_pin: gpio.2:low:pu
reset_pin: NO_PIN
feed_hold_pin: NO_PIN
cycle_start_pin:
macro0_pin: gpio.26:low:pu
macro1_pin: NO_PIN
macro2_pin: NO_PIN
macro3_pin: NO_PIN
fault_pin: NO_PIN
estop_pin: NO_PIN
# Using MOSFETs (Check Spindle Pin Usage
coolant:
flood_pin: NO_PIN
mist_pin: NO_PIN
delay_ms: 0
probe:
pin: gpio.33:low
toolsetter_pin: NO_PIN
macros:
startup_line0:
startup_line1:
macro0:
# G90&G53G0Z-1&G0X0Y0
macro1:
# $SD/Run=drill.nc
macro2:
macro3:
after_homing:
# g0 x1 y1
after_reset:
# g20
after_unlock:
# g91
user_outputs:
analog0_pin: NO_PIN
analog1_pin: NO_PIN
analog2_pin: NO_PIN
analog3_pin: NO_PIN
analog0_hz: 5000
analog1_hz: 5000
analog2_hz: 5000
analog3_hz: 5000
digital0_pin: NO_PIN
digital1_pin: NO_PIN
digital2_pin: NO_PIN
digital3_pin: NO_PIN
digital4_pin: NO_PIN
digital5_pin: NO_PIN
digital6_pin: NO_PIN
digital7_pin: NO_PIN
# FluiDial
uart1:
txd_pin: gpio.25
rxd_pin: gpio.27
rts_pin: NO_PIN
cts_pin: NO_PIN
baud: 1000000
mode: 8N1
uart_channel1:
report_interval_ms: 75
uart_num: 1
start:
must_home: true
deactivate_parking: false
check_limits: false
# #begin PWM
pwm:
pwm_hz: 5000
direction_pin: gpio.15
output_pin: gpio.13
enable_pin: gpio.14
disable_with_s0: false
s0_with_disable: true
spinup_ms: 0
spindown_ms: 0
tool_num: 0
speed_map: 0=0.000% 10000=100.000%
off_on_alarm: false
## begin Laser
# Laser:
# pwm_hz: 5000
# output_pin: gpio.4
# enable_pin: gpio.12
# disable_with_s0: false
# s0_with_disable: true
# tool_num: 1
# speed_map: 0=0.000% 255=100.000%
# off_on_alarm: true
## begin 10V
# 10V:
# forward_pin: gpio.15
# reverse_pin: gpio.14
# pwm_hz: 5000
# output_pin: gpio.13
# enable_pin: NO_PIN
# direction_pin: NO_PIN
# disable_with_s0: false
# s0_with_disable: true
# spinup_ms: 0
# spindown_ms: 0
# tool_num: 0
# speed_map: 0=0.000% 1000=0.000% 24000=100.000%
# off_on_alarm: false
## Begin Huanyang
# uart1:
# txd_pin: gpio.15
# rxd_pin: gpio.16
# rts_pin: gpio.14
# baud: 9600
# mode: 8N1
# Huanyang:
# uart_num: 1
# modbus_id: 1
# tool_num: 0
# speed_map: 0=0% 0=25% 6000=25% 24000=100%
# off_on_alarm: false
```

### Startup Messages

```text
Analysis following the procedure given by fluidnc wiki: “Requesting help for FluidNC”
 
Start Fluidterm:

Select a COM port
0: COM1 (\Device\Serial0)
1: COM7 (\Device\Silabser0)
Choice: 1
FluidTerm v1.2.1 using COM7
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O

No action if the “return” key is pressed.
Unable to send an order via Fluidterm.
No access via WebUI console, because no access to the card address.
Card not recognized or seen on the network.

Message after power OFF/ON:
Select a COM port
0: COM1 (\Device\Serial0)
1: COM7 (\Device\Silabser0)
Choice: 1
FluidTerm v1.2.1 using COM7
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O
Serial port disconnected - waiting for reconnect
Type any key to quit
Serial port reconnected
Resetting MCU
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1 (DOWNLOAD_BOOT(UART0/UART1/SDIO_FEI_REO_V2))
waiting for download

and then nothing more.

Other tests:
1. fluidterm activated,
2. Power On
3. Display 

Serial port disconnected - waiting for reconnect
Type any key to quit
Serial port reconnected
Resetting MCU
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1 (DOWNLOAD_BOOT(UART0/UART1/SDIO_FEI_REO_V2))
waiting for download

4. CTRL U and launch erase.bat
waiting for download
FluidNC filename [erase.bat]:
XMODEM Upload D: \ Mes_dossiers \ My DIY DIY \ 09_FLUIDNC \ FLUIDNC-V3.8.0-win64 \ FLUIDNC-V3.8.0-win64 \ ERASE.BAT ERASE.BAT

And then nothing more.

5. CTRL U and launch install-wifi.bat
Select a COM port
0: COM1 (\Device\Serial0)
1: COM7 (\Device\Silabser0)
Choice: 1
FluidTerm v1.2.1 using COM7
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O
FluidNC filename [install-wifi.bat]:
XMODEM Upload D: \ Mes_dossiers \ Mes DIY DIY \ 09_FLUIDNC \ FLUIDNC-V3.8.0-win64 \ FLUIDNC-V3.8.0-win64 \ Install-wifi.bat install-wifi.bat

And then nothing more.
```

### User Interface Software

Fluidial, BCNC, WEBUI, Fluidtherm, Web installer

### What happened?

Description of the problem:
When powered on, several I2SO output LEDs remain lit.
5V LED ON, RS485 LED ON
No communication with the FluiDial pendant.
No communication possible with “Bcnc” via USB.
No access possible via WIFI.
“Reset” button with no effect.
After 10 minutes under power, no change.
New power OFF then ON.
LED 5V and RS485 ON.
No communication with Fluidial, nor with “BCNC”.
The system has not started, is not running.
The ESP32 module is very, very hot, abnormally hot. (comparison made with another ESP32).

### Other Information

Web Installer seems to work, but after loading the card still does not start.
I sincerely think it's a hardware problem (defective ESP32?).

The machine worked fine for about a month with no problems

This is all the information that I can provide you, following the Mail from Bart Dring of 05/09/2024

[image]
[image]
[image]

Please arrange to replace this card under warranty, as I see no other solution to resolve the problem.
I of course remain at your disposal for further information if necessary.

Michel

bdring, 2024-09-06:
I agree, it does sound like the ESP32 has failed. I can direct Elecrow to send you a new one. The latest version uses an external antenna which is included. Is that acceptable? See the [Elecrow product page](https://www.elecrow.com/6x-cnc-controller-for-fluidnc.html) for details.

ESP32 failure is pretty rare. I think there are 3 things to watch for in the future.

- Overheating. Make sure it is in a well ventilated place with airflow above and below (if possible) the controller. You can set the [$WiFi/PsMode](http://wiki.fluidnc.com/en/features/commands_and_settings#wifipsmode) to `MAX` or `MIN` to run cooler with virtually no impact on normal WiFi performance.
- Expansion port. The only unprotected ESP32 pins are on the expansion port module socket. If you are using this for displays, etc be very careful about the wiring. Use good quality shielded wires and filters to prevent noise, spikes and shorts from getting to the ESP32. It does not look like you are using this, but a few people have broken the ESP32 this way.
- Power Supply.  Be sure to use a good quality power supply. Note: Your use of red and green wire markers is inverted from standard practices (Red+, Green-) and could lead to misconnection.

MichelBattistella, 2024-09-06:
Thanks for the quick handling of this issue.
The version with the external antenna suits me very well.
Please ask Elecrow to send me this card.
I take note of the three things to watch out for.

As for cooling, I don't think I have any problems.
Cooling with 2 80mm diameter fans.
I will set $WiFi/PsMode to MIN.

Expansion port.
Thanks for making me aware that the expansion module pins are not protected.
I use FluiDial plugged into these pins.
I checked the wiring several times as recommended. (as specified in the FluiDial WIKI)
The whole thing (6x CNC and FluiDial) worked correctly for a month
The cable used is shielded.
As the expansion module pins are not protected, wouldn't it be wise to prohibit plugging or unplugging the pendant while the board is powered on.
Looking at the FluidDial Pendant Wiki, I couldn't find anything about it.

Power Supply
I use a 24V regulated and adjustable switching power supply. No-load output voltage: 24.3V
You are a very good observer, yes the colors of the wires are not right.
In the photo you saw, I am using, for testing purposes, a 12V power supply whose + and – outputs are reversed. 

But don't worry, I know the rules.
 I'm 68 years old, I'm retired and I do it all for fun.
It is only a continuation of my professional life of 39 years in the fields of machine tools (Sinumerik, Fanuc), robotics and Motion control (simotion) within the company Siemens in France.
Drives  and the G code have no secrets for me.
Thank you again for the flawless handling of this issue.

Michel

MitchBradley, 2024-09-06:
Given your experience, perhaps you could monitor the FluidNC Discord server and offer advice to newcomers who often have trouble with basic CNC issues.  The FluidNC developers are swamped with questions of all sorts, and would prefer to spend time on actual issues with FluidNC, instead of fielding questions that are more generic in nature.

bdring, 2024-09-06:
I have authorize Elecrow to send you a new controller.

Expansion module: There is some more information [here](http://wiki.fluidnc.com/en/hardware/official/6x_CNC_Controller#expansion-module-socket).

MichelBattistella, 2024-09-07:
I am flattered that you thought of me to do what we call in jargon the “hotline”.
But unfortunately I will disappoint you. 
I don't want to commit to a task that will take a lot of time.
Also, English is not my favorite language.
I devote my time primarily to my hobbies, one of which is the CNC, and to my family and grandchildren.
I would probably watch the discord server occasionally. 
But I don't promise anything.
Good luck and keep the wind in your sails. 
Thanks again to Bart for handling my problem.

closed 2024-09-08 by bdring