bdring/FluidNC #1311: Problem: I am using Fysetc E4, there is a voltage between ground laser enable and PWM pin

FHS1959, 2024-09-06:
### Wiki Search Terms

Fysetc E4 PWM
Fysetc E4 
PWM

### Controller Board

Fysetc E4

### Machine Description

DIY laser engraver with Fysetc E4 board and AENBUSLM 80W laser

### Configuration file

```yaml
name: "Fysetc Ant Cartesian"
board: "Fysetc E4"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

uart1:
  txd_pin: gpio.22
  rxd_pin: gpio.21
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1

kinematics:
  Cartesian:

axes:
  shared_stepper_disable_pin: gpio.25

  X:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 270.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 1.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 3.500
      feed_scaler: 3.500

    motor0:
      limit_neg_pin: gpio.34:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209:
        uart_num: 1
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.500
        microsteps: 16
        run_mode: stallguard
        homing_mode: stallguard
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        use_enable: false
        step_pin: gpio.27
        direction_pin: gpio.26:low
        disable_pin: NO_PIN

  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 505.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 1.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 3.500
      feed_scaler: 3.500

    motor0:
      limit_neg_pin: gpio.35:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209:
        uart_num: 1
        addr: 3
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.500
        microsteps: 16
        run_mode: stallguard
        homing_mode: stallguard
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        use_enable: false
        step_pin: gpio.33
        direction_pin: gpio.32:high
        disable_pin: NO_PIN

    motor1:
      limit_neg_pin: gpio.15:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209:
        uart_num: 1
        addr: 0
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.500
        microsteps: 16
        run_mode: stallguard
        homing_mode: stallguard
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        use_enable: false
        step_pin: gpio.14
        direction_pin: gpio.12:low
        disable_pin: NO_PIN

coolant:
  #air blower
  flood_pin: gpio.4:high

laser:
 pwm_hz: 20000
 output_pin: gpio.13
 enable_pin: gpio.2
 disable_with_s0: false
 s0_with_disable: true
 tool_num: 0
 speed_map: 0=0.000% 1000=100.000%
 off_on_alarm: true
   
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

start:
  must_home: false
  deactivate_parking: false
  check_limits: false
```

### Startup Messages

```text
Resetting MCU
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.7.17 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Fysetc Ant Cartesian]
[MSG:INFO: Board Fysetc E4]
[MSG:INFO: UART1 Tx:gpio.22 Rx:gpio.21 RTS:NO_PIN Baud:115200]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.25]
[MSG:INFO: Axis X (1.000,271.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 UART1 Addr:1 CS:NO_PIN Step:gpio.27 Dir:gpio.26:low Disable:NO_PIN R:0.110]
[MSG:INFO:  X Neg Limit gpio.34]
[MSG:INFO: Axis Y (1.000,506.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 UART1 Addr:3 CS:NO_PIN Step:gpio.33 Dir:gpio.32 Disable:NO_PIN R:0.110]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2209 UART1 Addr:0 CS:NO_PIN Step:gpio.14 Dir:gpio.12:low Disable:NO_PIN R:0.110]
[MSG:INFO:  Y2 Neg Limit gpio.15]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:gpio.2 Out:gpio.13 Freq:20000Hz Period:2047]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Flood coolant gpio.4]
[MSG:INFO: Connecting to STA SSID:Telfy_11461]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connected - IP is 192.168.1.72]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.17 (wifi) '$' for help]
```

### User Interface Software

Fluidnc

### What happened?

The laser AENBUSLM 80W is using 3 pins, 12V GND and PMW and is connected via 12V (power) and GND on the heater pins and PMW and GND of the fan pins. The ground of power and PWM pins are shared. If starting a print, The laser turns on and the PWM is varying between 0 and 12V. The problem is that the voltage of the ground pins of the heater and fan is not 0V. This varies between about 1 and 6 volt. So when I connect the Fan pins will this influence the voltage of the heater. If only connecting the PWM pin of the fan and not the ground then the voltage between ground of the power pin and the PWM pin is always 12v.

How can i solve this issue
     

### GCode File

G90 (use absolute coordinates)
G0 X3 Y3 S0
G1 M4 F1000
F1000
Y3 S0
X23 S100
X43 S200
X63 S300
X83 S400
X103 S500
X123 S600
X143 S700
X163 S800
X183 S900
X203 S1000
Y3.125 S0
X183 S1000
X163 S900
X143 S800
X123 S700
X103 S600
X83 S500
X63 S400
X43 S300
X23 S200
X3 S100
Y3.25 S0
X23 S100
X43 S200
X63 S300
X83 S400
X103 S500
X123 S600
X143 S700
X163 S800
X183 S900
X203 S1000
Y3.375 S0
X183 S1000
X163 S900
X143 S800
X123 S700
X103 S600
X83 S500
X63 S400
X43 S300
.... and so on

bdring, 2024-09-06:
The heater "gnd" is probably a MOSFET.

How are you controlling that. Are you just looking for a permanent ground?

FHS1959, 2024-09-06:
Yes it is a Mosfet and is controlled by the anable pin of the laser.  The voltage between the ground puns of the heater and poer in is 0V  as it should be. 
But why is is there a voltage between 2 ground pins of the heater and the fan

MitchBradley, 2024-09-06:
it is not difficult to find the Fysetc E4 schematic with a web search.

Zinyx, 2024-09-06:
All mosfet work N channel, when you have signal to fan IO13 mosfet is "on" and put to GND and you have R7 and Diode  D2 from VCC, and GND from Heat E0 then you have voltage on this two pin.

Find R10 and Q1 on your board and take signal PWM from gate Q1, Then, you have 5V PWM. 
Enable pin from IO2 can be connect to GND. 
and +12V from Input.
[image]

FHS1959, 2024-09-06:
I need a 12V pwm.
Did a test, also with mosfet turned of there is a voltage between the 2 ground pins

MitchBradley, 2024-09-07:
The heater and fan connectors do not have "GND" pins.  They have "open drain" pins that are connected to system GND only when the corresponding MOSFET is turned on.  When the Fan MOSFET is off, P2 pin 1 is pulled up to FAN_VCC (minus an LED Vf diode drop) through 2.2K ohms.  When the HEAT_BED MOSFET is off, P4 pin 2 is pulled up to 24V2 (which might be 12V depending on your power supply), minus an LED Vf drop, through 10K ohms.  HEAT_E0 is similar to HEAD_BED with P11 instead of P4.

FHS1959, 2024-09-07:
Is there another PWM pin i could use

bdring, 2024-09-07:
You probably want a full time ground and 12V and a 5V PWM.

12V PWM might break the laser control circuit.

FHS1959, 2024-09-07:
I will change it to 5v and test. 
How to get a full time ground

lols, 2024-09-07:
I did a search on google for your laser module, amazon page says "Power adjustable: supports 5V PWM modulation; Laser wavelength: 450nm; Input: DC 12V 3.2A; Input: XH2.54-3P socket; "
SO it is a 12v input with a ttl 5v pwm NOT 12v you might burn the circuit
The E4 board does not have header pin for the pwm so you will need to either solder 1 header pin to the mosfet gate pin(so you can plug and unplug) or solder the pwm wire directly to the mosfet gate pin or solder a header pin to the esp32 pad or directly on the esp32 pad

MitchBradley, 2024-09-07:
If you look at the Fysetc E4 schematic, which is easily found with a web search, you will see that there are many places to get a solid ground, such as P1, P3, P5, P6, P12, P13, P17, P18, P19, EXT-RST.  But you cannot use P2, P4, or P11 as GND because their low sides are switched with MOSFETs instead of being tied directly to ground.  If you jumper P16 so that FAN_VCC is connected to +5V, then P2 pin 1 will switch between 0V and 4V, which is sufficient to drive a 5V PWM input.

If you need any more hardware handholding, please consider donating to support the FluidNC project.  We provide software at no cost, but we do not offer free hardware consultation.

closed 2024-09-07 by MitchBradley