bdring/FluidNC #1312: Problem: GCode Logic has problems and throes errors

projecjeff, 2024-09-08:
### Wiki Search Terms

I am using the egsamles you supplied on the wiki page of the manual

### Controller Board

My own board using latese Ver 3.8.2  installed via Web installer

### Machine Description

3 Axis Router

### Configuration file

```yaml
board: Projec
name: PRJ XYZABC DEV ATC V4PCB

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false

start:
  must_home: false
  deactivate_parking: false
  check_limits: false

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 10
  dir_delay_us: 5
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 180.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1:low		
        disable_pin: I2SO.0

  y:
    steps_per_mm: 180.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 100.000
[... 351 lines cut ...]
Result..
> G91
> F500
> #<_x>=0
> o101while[#<_x>LT10]
> G1X5
> #<_x>=[#<-x>+1]
> o101endwhile
ok
ok
*** The communicator has been paused
error:180
*** The communicator has been paused
[MSG:ERR: Parameter Assignment Failed]
error:177
[MSG:ERR: Flow Control Not Executing Macro]
ok
*** The communicator has been paused
error:2
[MSG:ERR: Bad GCode number format]
error:177

**** Finished sending file in 00:00:00 ****

ERR: Flow Control Not Executing Macro

bdring, 2024-09-08:
It works for me. 

Are you running a file on the `SD` or `LocalFS` or streaming from a gcode sender?

```
$H
[MSG:Homed:Z]
[MSG:Homed:XY]
ok
G0 X0
ok
<Idle|WPos:-48.000,89.000,44.577,0.000|Bf:15,128|FS:0,0>
<Idle|WPos:-48.000,89.000,44.577,0.000|Bf:15,128|FS:0,0>
G90
ok
g0x0
ok
<Idle|WPos:0.000,89.000,44.577,0.000|Bf:15,128|FS:0,0>
$localfs/run=repeat.nc
ok
<Idle|WPos:20.000,89.000,44.577,0.000|Bf:15,128|FS:0,0>
$localfs/show=repeat.nc
#<_times>=4
o101 repeat[#<_times>]
G91 G1 X5 F500
o101 endrepeat
ok
```

projecjeff, 2024-09-09:
Thanks for the prompt reply.
I am loading the file through UGS
I havent tryed running on SD
and I dont use the local FS
Thans again Jeff

projecjeff, 2024-09-09:
I found the problem.
It runs from SD Card but not UGS Ver 2.6.1
[image]

projecjeff, 2024-09-09:
CSender also has a problem. looks like the Logic and Variable statements are causing problems when being sent from UGS ans GSender.

I there any way around this

Cheers
Jeff

bdring, 2024-09-09:
Gcode sent manually from the console or streamed from a gcode sender is not saved by FluidNC forever. An ESP32 has limited memory. FluidNC completes the action of the gcode, then forgets its it.  Any modal states or parameters set do remain in effect. 

Flow control like loops are meant to be done in the context of a file (or job). It needs to be able to loop back to previous statements.

The solution is to save you code to the SD card and run it from there.

I will work with @MitchBradley to improve the wiki in regards to how this affects flow control. You can search the wiki for "job" for some helpful info.

bdring, 2024-09-09:
I have added a [section to the wiki](http://wiki.fluidnc.com/en/features/gcode_parameters_expressions#scope-context-and-usage).

closed 2024-09-09 by bdring