bdring/FluidNC #1319: Problem: '$$' does not return '$32' for the laser configuration

UR3IQO, 2024-09-12:
### Wiki Search Terms

$$ $32 Laser

### Controller Board

MKS DLC32 V2.1

### Machine Description

DIY laser engraver with dual Y motors

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: Oleg
meta: 10 Sep 2024

kinematics:
  Cartesian:

#There is one UART defined here which is chared by all drivers. Drivers have individual
#adresses defined in the sections above
uart1:
  txd_pin: gpio.0
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  baud: 115200
  mode: 8N1

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 30000
    acceleration_mm_per_sec2: 5000
    max_travel_mm: 400
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 100
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:high
      hard_limits: false
      pulloff_mm: 2.000
      tmc_2209:
        uart_num: 1
        addr: 0 
        cs_pin: NO_PIN
        r_sense_ohms: 0.110
        run_amps: 0.800
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
        homing_amps: 0.400
        use_enable: false
        step_pin: I2SO.1
        direction_pin: I2SO.2:high
        disable_pin: NO_PIN

  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 30000
    acceleration_mm_per_sec2: 1500
    max_travel_mm: 400
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 2000.000
      settle_ms: 100
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:high
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209: 
        uart_num: 1
        addr: 1
        cs_pin: NO_PIN
        r_sense_ohms: 0.110
        run_amps: 0.800
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
        homing_amps: 0.400
        use_enable: false
        step_pin: I2SO.5
        direction_pin: I2SO.6:high
        disable_pin: NO_PIN

    motor1:
      limit_neg_pin: gpio.34:high
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209: 
        uart_num: 1
        addr: 2
        cs_pin: NO_PIN
        r_sense_ohms: 0.110
        run_amps: 0.800
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
        homing_amps: 0.400
        use_enable: false
        step_pin: I2SO.3
        direction_pin: I2SO.4:low
        disable_pin: NO_PIN
        
i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
#  card_detect_pin: gpio.39
  
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0: $H
  startup_line1:
  after_reset:
  macro0:
  macro1:
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

#probe:
#  pin: gpio.22
#  check_mode_start: true

Laser:
  pwm_hz: 5000
  output_pin: gpio.32
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
#PWM:
#  direction_pin: NO_PIN
#  spinup_ms: 1000
#  spindown_ms: 1000
  
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  #Beeper connector. Command: "M62 P0" turn beeper On. 
  #"G4 P0.5" wait for 0.5s. "M63 P0" turn beeper OFF
  digital0_pin: I2SO.7
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

#MKS DLC32 v2.1 pinMap

# X- gpio.36
# SD_DET gpio.39
# Z- gpio.34
# Y- gpio.35
# SPINDLE PWM gpio.32
# LCD_RS gpio.33
# LCD_CS_INV gpio.25
# LCD_TOUCH_CS_INV gpio.26
# LCD_RST_INV gpio.27
# SD_SCK gpio.14
# SD_DO gpio.12
# SD_DI gpio.13
# SD_CS gpio.15
# I2C_SDA gpio.0
# I2C_SCL gpio.4
# I2S_BCK gpio.16
# I2S_WS gpio.17
# LCD_EN_INV gpio.5
# LCD_SCK gpio.18
# LCD_MISO gpio.19
# I2S_DATA gpio.21
# RXD0 gpio.3
# TXD0 gpio.1
# Probe gpio.22
# LCD_MOSI gpio.23

# XYZ_EN I2SO.0
# X_STEP I2SO.1
# X_DIR I2SO.2
# Z_STEP I2SO.3
# Z_DIR I2SO.4
# Y_STEP I2SO.5
# Y_DIR I2SO.6
# BEEPER I2SO.7
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.8.2 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:My_Laser_XYY_2209.yaml]
[MSG:INFO: Machine Oleg]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: UART1 Tx:gpio.0 Rx:gpio.4 RTS:NO_PIN Baud:115200]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: User Digital Output:0 on Pin:I2SO.7]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 UART1 Addr:0 CS:NO_PIN Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN R:0.110]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO: Axis Y (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 UART1 Addr:1 CS:NO_PIN Step:I2SO.5 Dir:I2SO.6 Disable:NO_PIN R:0.110]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2209 UART1 Addr:2 CS:NO_PIN Step:I2SO.3 Dir:I2SO.4:low Disable:NO_PIN R:0.110]
[MSG:INFO:  Y2 Neg Limit gpio.34]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:FluidNC]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: No SSID]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.32 Freq:5000Hz Period:8191]
ok
```

### User Interface Software

fluidterm, LaserGRBL

### What happened?

When sending `$$` request the FluidNC does not return `$32=1`, so the LaserGRBL selects `M3` command for engraving (by default) and shows warning if `M4` is selected manually.

Here is the output of the `$$` command:
```
$22=1
$21=0
$20=1
$102=80.000
$101=80.000
$100=80.000
$112=1000.000
$111=30000.000
$110=30000.000
$122=25.000
$121=1500.000
$120=5000.000
$132=1000.000
$131=400.000
$130=400.000
$10=1
$30=1000
ok
```

### Other Information

I have resolved this problem, by adding a small fix to the firmware, and will be glad to share it. But I am not sure if it has to be fixed :)

bdring, 2024-09-12:
If you send $32, FluidNC responds correctly.

I think that used to be good enough.

UR3IQO, 2024-09-13:
I know that `$32` request results in correct `$32=1` answer. Unfortunately LaserGRBL expects `$32=1` to be in `$$` results. I am happy with my fix  https://github.com/bdring/FluidNC/commit/98f82176d6f5e0cd3f7530597264bb0e7e80fa4f . This might not be a best way cause I am not familiar with the FluidNC codebase, but if you think it is useful I can make pull request.

Reywas62, 2024-09-17:
I agree that this should be fixed.  I'm new to FluidNC and LaserGBRL  and I have spent many hours trying to figure out this issue.  The only way I could get LaserGRBL  to stop warning me that my GBRL version was "<=1.1" was to input "$32=1" into the command terminal.  I also suspect that this is causing LightBurn to incorrectly ID my laser as having old firmware that is not M4 compatible.  I will give UR31QO's fix a try if I can figure out how to do it.  I am not in any way a coder.

MitchBradley, 2024-12-11:
Is it better to base the $32 answer on "has laser" or is it better to base it on "the currently-selected tool number refers to a laser"?

bdring, 2024-12-13:
There is a PR for this now.

https://github.com/bdring/FluidNC/pull/1400

bdring, 2024-12-27:
Completed

closed 2024-12-27 by bdring