bdring/FluidNC #1320: Problem: Config Dump ($CD=config.yaml) Adds Invalid Lines to YAML, increases time to connect

jeyeager65, 2024-09-12:
### Wiki Search Terms

$CD
web_commands
wifi
ota
mdns
telnet_server
notifications:

### Controller Board

Jackpot

### Machine Description

V1E LowRider 3

### Configuration file

```yaml
*** ORIGINAL ***
board: Jackpot TMC2209
name: LowRider
meta: 03-15-2024 RyanZ
stepping: 
  engine: I2S_static
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  segments: 12

uart1: 
  txd_pin: gpio.0
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1

uart2: 
  txd_pin: gpio.14
  rxd_pin: gpio.13
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

uart_channel2: 
  report_interval_ms: 75
  uart_num: 2
  message_level: Verbose

i2so: 
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi: 
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard: 
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 20000000

kinematics: 
[... 734 lines cut ...]
ota: 
mdns: 
wifi: 
telnet_server: 
notifications: 

None of these are documented on the FluidNC wiki:
http://wiki.fluidnc.com/en/config/overview#documentation-of-each-section
http://wiki.fluidnc.com/en/config/top_level_config_items

What is happening is that every time I run $CD=config.yaml to update the file, those lines are added to the file.  So, if you run it a second time, there are two sets of those lines, etc.

Looking at the $SS output, you can see that those extra lines are causing it to connect and disconnect from the wifi multiple times.

### Other Information

I knew this used to work fine, so I rolled back to v3.8.1 and the same problem occurred.  However, when I rolled back to v3.8.0, it worked as I expected again.  So, this issue was introduced in v3.8.1.

My understanding of the FluidNC code is extremely limited, but I looked at the diff between 3.8.0 and 3.8.1:
https://github.com/bdring/FluidNC/compare/v3.8.0...v3.8.1

I see changes to FluidNC/src/Configuration/Generator.h and modules related to the extra lines in the yaml file like:
- FluidNC/src/WebUI/Mdns.cpp
- FluidNC/src/WebUI/OTA.cpp
- etc.

jeyeager65, 2024-09-13:
Installed Fix1320 branch via PlatformIO and confirmed it now behaves as expected.  If there's anything I can do to assist with further testing, let me know.  Thanks!

MitchBradley, 2024-09-13:
The things that are potentially affected by the change are configurable modules like oled and atc_manual, and the ability to exclude non-configurable network modules like telnet and mdns by blacklisting their files via platformio.ini build_src_filter.

jeyeager65, 2024-09-14:
I'm a bit out of my depth here.  I tried excluding modules without success but had the same compile issues on the Fix1320 branch and main.

PR #1321 (2024-09-25): Configurable and non-configurable modules to keep the config tidy
Fixes #1320

Needs a heck of a lot of testing

Non-configurable modules now do not appear in the configuration listing.  Configurable modules must now derive from ConfigurableModule instead of Module.
closed 2024-09-25 by bdring