bdring/FluidNC #1323: Problem: Steppers Don't Rotate

bakerr13, 2024-09-13:
### Wiki Search Terms

DM542T, Wiring Diagram, Nema 23, 5V Jumper, Clicking, rotating, locked, 6 pack, 6X

### Controller Board

6x CNC Controller for FluidNC (integrated ESP32) V1.1
Firmware V3.8.2

### Machine Description

Gantry router, dual -axis with dual endstops. Drivers are DM542T's paired to Nema 23 steppers. Board power supply is a benchtop supply set to 20V. Stepper power supply is a Meanwell LRS-350-48.

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYYZ No Square
meta: 2022-09-19 B. Dring for Dean Embrey

stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 750.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0:high

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

start:
  must_home: false
  deactivate_parking: true
  check_limits: false
  
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.8.2 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:exampleConfig.yml]
[MSG:INFO: Machine 6 Pack External XYYZ No Square]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-750.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:Get off My Lawn 2.4]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.124]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

FluidTerm, WebUI

### What happened?

Initially I had both Y steppers configured and wired to the drivers/controller. Upon testing the motors would make a faint whirring noise while the terminal showed the machine jogging, and then make a single click as the steppers were disabled. During the jog the steppers are both locked up and I can't rotate them by hand. After the jog, they move freely. Things I've tried:

1.  Confirmed wiring
2. disconnected enable pin
3. eliminated config for y axis motors, disconnected those drives and reduced the problem to a single axis (X, problem persists)
4. Adjusted current settings on driver
5. Adjusted supply voltage to controller
6. performed continuity tests between all connections
7. Adjusted pulse timing, accel rate,  max feed rates
8. Changed power supplies
9. Confirmed stepper coil pairs
10. Adjusted idel_ms to enable steppers 100% of the time and to disable after jogging

### Other Information

I've done as much trouble shooting that I can, trying many suggestions I've seen here and on Discord. One weird thing I noticed was when I took out the y axis config and disconnected those drives to isolate the problem, I still get noise and clicking form the X axis stepper when I jog in the y direction. I don't think that's normal. This may be a pin voltage issue, but I'm not 100% certain where the jumper on the board is to change that to 5V pins vs 3.3V. I have an idea, but can't find any information online regarding the jumper locations and I don't want to cook my board. Another thing I've noticed is that a lot of the time, but not always, after jogging the UI will continue display "Jog" instead of of "Idle" even after the move is complete. In the video below, the click is occurring at the end of the commanded jog (100 mm in this instance). 

[image]
[image]
[image]
[

https://github.com/user-attachments/assets/75f00681-3d2c-4797-b312-326ba0e2cd27

](url)

bdring, 2024-09-13:
That has a really long enable delay. Try changing 

- idle_ms: 255
- disable_delay_us: 200000

[image]

Try changing

bakerr13, 2024-09-13:
That didn't do it. I heard continuous noise from the stepper and only the dir pin for that motor section was illuminated. I changed it back, got rid of the :high attribute on disable_pin and I see only the dir pin illuminated when jogging, and the dir+ena pins illuminated when at idle.

bdring, 2024-09-13:
You should have full time motor lock now. You can disable the motors with $MD and enable with $ME.  The enable LED should toggle with the enable state. Please confirm.

The dir LED should only change when you change direction on the jogs. Try alternating jogs forward and back. Please confirm.

The Step LED should fade in brightness during the accel and decel phase of the jog. Try long jogs of 100mm. Please confirm.

bakerr13, 2024-09-13:
> You should have full time motor lock now. You can disable the motors with $MD and enable with $ME. The enable LED should toggle with the enable state. Please confirm.
> 
> The dir LED should only change when you change direction on the jogs. Try alternating jogs forward and back. Please confirm.
**> This behaves as expected. On one way, off the other.**
> The Step LED should fade in brightness during the accel and decel phase of the jog. Try long jogs of 100mm. Please confirm.
**Again, behaves as you stated.** 

The ena light turns on when I disable motor, off when I enable them.

bdring, 2024-09-13:
Do the motors follow $ME locks them and $MD unlocks them?

Also, there were 2 other things for you to confirm in the last post.

bakerr13, 2024-09-13:
$ME does lock the motor
$MD unlocks the motor
The LED does toggle with each state.

bdring, 2024-09-13:
....and the step LED ????????

bakerr13, 2024-09-13:
As stated in a previous reply, it does turn on/off when I change direction.

bdring, 2024-09-13:
The STEP LED?

There are 3 LEDs. I want to know about each of them. Please explain the behavior of EACH in a reply.

bakerr13, 2024-09-13:
The STEP LED fades in/out as expected. 
DIR LED changes from on/off depending on direction of jog
ENA LED turns on/off with $MD/$ME

bdring, 2024-09-13:
Send a picture of the position of the %V 24V switch.

[image]

bakerr13, 2024-09-13:
I'll be damned. I never noticed that switch there. It was switched to 24V. I switched to 5V and it works! Thank you so much!

bdring, 2024-09-13:
Where did you get the 6x controller?

bakerr13, 2024-09-13:
Tindie

closed 2024-09-13 by bdring