bdring/FluidNC #1325: Problem: TMC drive not detected.

Timacaro, 2024-09-19:
### Wiki Search Terms

TNC driver not detected

### Controller Board

6-pack

### Machine Description

Dual Y, X, Z . Nema 17, 24v.  TMC2130. x 6

### Configuration file

```yaml
board: 6 Pack
name: TMC2130 XXYYZ Relay
meta: 01/14/2022 B. Dring for Rob McMonigal
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      tmc_2130:
        cs_pin: i2so.3
        run_amps: 0.50
        hold_amps: 0.20
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0
        
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      tmc_2130:
        cs_pin: i2so.6
        run_amps: 0.50
        hold_amps: 0.20
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      tmc_2130:
        cs_pin: i2so.11
        run_amps: 0.50
        hold_amps: 0.20
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
        
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.14
        run_amps: 0.50
        hold_amps: 0.20
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.19
        run_amps: 0.50
        hold_amps: 0.20
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16
        
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

probe:
  pin: gpio.25
  check_mode_start: true

start:
  must_home: false
  
coolant:
  flood_pin: NO_PIN
  mist_pin: gpio.27
  delay_ms: 0
  
relay:
  output_pin: gpio.26
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 0=100.000% 1=100%
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.8.2 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine TMC2130 XXYYZ Relay]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.2 Dir:I2SO.1 CS:I2SO.3 Disable:I2SO.0 Index:-1 R:0.110]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2130 Step:I2SO.5 Dir:I2SO.4 CS:I2SO.6 Disable:I2SO.7 Index:-1 R:0.110]
[MSG:INFO: Axis Y (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.10 Dir:I2SO.9 CS:I2SO.11 Disable:I2SO.8 Index:-1 R:0.110]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2130 Step:I2SO.13 Dir:I2SO.12 CS:I2SO.14 Disable:I2SO.15 Index:-1 R:0.110]
[MSG:INFO: Axis Z (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.18 Dir:I2SO.17 CS:I2SO.19 Disable:I2SO.16 Index:-1 R:0.110]
[MSG:ERR: X Axis TMC driver not detected - expected 0x11 got 0x0]
[MSG:INFO: X2 Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x11 got 0x0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.26 Dir:NO_PIN]
[MSG:INFO: Mist coolant gpio.27]
[MSG:INFO: Probe gpio.25]
```

### What happened?

A few months ago I could not get the board to function and after a lot of support I decided to stop and stored the card. This week I reinstalled FLUIDNC because I wanted to have a go at it again to let drive my custom machine. I installed the yaml file which I downloaded. I changed the drivers to see if it would help, but no.

MitchBradley, 2024-09-19:
The X driver is not detected but the X2 one is detected.  Have you tried swapping them to see if the fault moves?  If so, it could be a faulty driver.  If not, it could be a faulty 6 pack board.

bdring, 2024-09-19:
A lot of circuitry is shared with x and x2, so it is most likely a driver problem if one works.

Try swapping some drivers.

bdring, 2024-12-17:
I think this is fixed in v3.9.3

closed 2024-12-17 by bdring