bdring/FluidNC #1327: Problem: Fluiddial does not show icons on menu screen

billmckillip, 2024-09-22:
### Wiki Search Terms

Fluiddial does not show icons menu screen fluidnc fluid nc dial pendant 

### Controller Board

MKS Tinybee v1.0

### Machine Description

Prototyping xyz frame for compact plasma cutter development
[image]
[image]

### Input Circuits

```yaml
Pendant exactly as shown in wiki. Black and Red on pendant cable to tinybee 24v out and data to gpio pins 17 and 16.
```

### Configuration file

```yaml
board: MKS TinyBee V1.0 XXYYZ
name:

kinematics:
  Cartesian:

uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

uart_channel1:
  report_interval_ms: 75
  uart_num: 1

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  # uses TH2 IO34 active low - MAKE SURE jumper J2 is set to SDDET!!!
  card_detect_pin: gpio.34:low

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 2

axes:
  x:
    steps_per_mm: 160
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 2500.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low:pu
      hard_limits: true
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0

    # use E0 driver for 2nd X axis motor
    motor1:
      hard_limits: true
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.10
        direction_pin: I2SO.11
        disable_pin: I2SO.9

  y:
    steps_per_mm: 160
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 70.000
    max_travel_mm: 1250.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low:pu
      hard_limits: true
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.4
        direction_pin: I2SO.5
        disable_pin: I2SO.3

    # use E1 driver for 2nd Y axis motor
    motor1:
      limit_neg_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.13
        direction_pin: I2SO.14
        disable_pin: I2SO.12

  z:
    steps_per_mm: 160
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 60.000
    max_travel_mm: 80.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.22:low:pu
      hard_limits: true
      pulloff_mm: 3.000
      stepstick:
        step_pin: I2SO.7
        direction_pin: I2SO.8
        disable_pin: I2SO.6

control:
  safety_door_pin: NO_PIN
  # on MT_DET connector
  reset_pin: gpio.35:low
  # on TH1 connector
  feed_hold_pin: gpio.36:low
  # on TB connector
  cycle_start_pin: gpio.39:low
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  # Heated Bed Terminal Block
  flood_pin: i2so.16
  # HE0 Terminal Block
  mist_pin: i2so.17
  delay_ms: 0

# spindle PWM signal
PWM:
  pwm_hz: 2500
  # on EXP1 IO15 connector.
  # note IO15 will give some short pulses on boot, which may activate spindle
  # use IO17 on EXP1 to avoid
  output_pin: gpio.15:high
  s0_with_disable: true
  tool_num: 0
  spinup_ms: 4000
  spindown_ms: 4000
  speed_map: 0=0.000% 12000=100.000%

Laser:
  pwm_hz: 5000
  # on 3D Touch connector (has pdwn + PWM)
  output_pin: gpio.2:high:pd
  s0_with_disable: true
  tool_num: 1
  speed_map: 0=0.000% 1000=100.000%

start:
  must_home: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.8.2 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:MKS_TinyBee_Pendant_XYZ.yaml]
[MSG:INFO: Machine ]
[MSG:INFO: Board MKS TinyBee V1.0 XXYYZ]
[MSG:INFO: UART1 Tx:gpio.17 Rx:gpio.16 RTS:NO_PIN Baud:1000000]
[MSG:INFO: uart_channel1 created at report interval: 75]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.34:low freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:2us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,2500.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.33:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.11 Disable:I2SO.9]
[MSG:INFO: Axis Y (0.000,1250.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:  Y Neg Limit gpio.32:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.14 Disable:I2SO.12]
[MSG:INFO: Axis Z (-80.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:  Z Pos Limit gpio.22:low:pu]
[MSG:INFO: reset_pin gpio.35:low]
[MSG:INFO: feed_hold_pin gpio.36:low]
[MSG:INFO: cycle_start_pin gpio.39:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:BELL622]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.2.16]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.15 Dir:NO_PIN Freq:2500Hz Period:16383]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.2:pd Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant I2SO.16]
[MSG:INFO: Mist coolant I2SO.17]
```

### User Interface Software

webui, software installer, fluidterm, platformio

### What happened?

Pendant is fully functional but icons are not displayed on menu screen, just plain white disks.  Disks expand as expected when selected. The M5 example programs display icons correctly.

### Other Information

The only thing to note is that I am new to platformio and may have left out a file or firmware update step to get the icons on the m5 dial.

MitchBradley, 2024-09-22:
http://wiki.fluidnc.com/en/hardware/official/M5Dial_Pendant#upload-the-image-files

billmckillip, 2024-09-22:
Thanks, all sorted.  And, thanks for not saying 'you dummy'.
This is an awesome project - a great step forward for/from grbl. I was amazed how easy it was to get my machine up and running on fluidnc and, apart from my difficulty with simple instructions, how easy it was to put together and add the pendant.
thanks,
Bill

http://wiki.fluidnc.com/en/hardware/official/M5Dial_Pendant#upload-the-image-files

MitchBradley, 2024-09-22:
I'm glad it's working for you.  Donations to support all the work we have put into this, and to buy hardware for new development and testing, are always appreciated.  There are links on the GitHub main page.

closed 2024-09-22 by MitchBradley

billmckillip, 2024-10-03:
Mitch,
  I have made a small donation.  Thanks again for the support.
Robert (Bill) McKillip

MitchBradley, 2024-10-03:
Thanks!  Tell us your Discord name and you will get a sponsor badge there.

billmckillip, 2024-10-03:
It's billmckillip.