bdring/FluidNC #1328: Problem: Tablet View in WebUI - file upload broken

lytke, 2024-09-22:
### Wiki Search Terms

webui tablet

### Controller Board

6-pack

### Machine Description

QueenAnt Pro CNC - machine working fine.

### Configuration file

```yaml
board: MKS TinyBee V1.0 XYYZ
name: TinyBee

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  # uses TH2 IO34 active low - MAKE SURE jumper J2 is set to SDDET!!!
  card_detect_pin: gpio.34:low

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 2

axes:
  x:
    steps_per_mm: 160.080
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 15
    max_travel_mm: 700
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 1.000
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_all_pin: gpio.33:pu:low
      #limit_neg_pin: gpio.33:low:pu
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.7
        direction_pin: I2SO.8
        disable_pin: I2SO.6

  y:
    steps_per_mm: 160.080
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 15.000
    max_travel_mm: 800.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low:pu
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.4
        direction_pin: I2SO.5
        disable_pin: I2SO.3

    # use E0 driver for 2nd Y axis motor
    motor1:
      hard_limits: false
      #hard_limits: true
      #limit_neg_pin: NO_PIN
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.11
        disable_pin: I2SO.9

    # use E1 driver for 2nd Y axis motor
    #motor1:
    #  limit_neg_pin: NO_PIN
    #  hard_limits: false
    #  pulloff_mm: 4.000
    #  stepstick:
    #    step_pin: I2SO.13
    #    direction_pin: I2SO.14
    #    disable_pin: I2SO.12

  z:
    steps_per_mm: 313.725
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 10.000
    max_travel_mm: 150.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_all_pin: gpio.22:low:pu
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0

control:
  safety_door_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN

  # on MT_DET connector
  reset_pin: gpio.35:low
  
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

probe:
  # Thermal Heated Bed TB 
  pin: gpio.39:low
  toolsetter_pin: NO_PIN
  check_mode_start: true
  
coolant:
  # Heated Bed Terminal Block
  flood_pin: i2so.16
  
  # HE0 Terminal Block
  mist_pin: i2so.17

  # HE1 Terminal Block
  #mist_pin: i2so.18
  delay_ms: 1000

10V:
  # EXP1 connector
  forward_pin: gpio.0
  reverse_pin: gpio.4
  output_pin: gpio.21:high
  pwm_hz: 5000
  tool_num: 0
  spinup_ms: 4000
  spindown_ms: 4000
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  off_on_alarm: false
  #speed_map: 0=0.000% 12000=100.000%
  speed_map: 0=0.000% 1000=0.000% 24000=100.000%

Laser:
  pwm_hz: 5000
  # on 3D Touch connector (has pdwn + PWM)
  output_pin: gpio.2:high:pd
  enable_pin: NO_PIN
  disable_with_s0: false  
  s0_with_disable: true
  off_on_alarm: true
  tool_num: 1
  speed_map: 0=0.000% 100=100.000%

start:
  must_home: false
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.8.2 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config-TinyBee.yaml]
[MSG:INFO: Machine TinyBee]
[MSG:INFO: Board MKS TinyBee V1.0 XYYZ]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.34:low freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:2us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (1.000,701.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.11 Disable:I2SO.9]
[MSG:INFO:  X All Limit gpio.33:low:pu]
[MSG:INFO: Axis Y (0.000,800.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:  Y Neg Limit gpio.32:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.1 Dir:I2SO.2 Disable:I2SO.0]
[MSG:INFO:  Y2 Neg Limit gpio.36:low]
[MSG:INFO: Axis Z (0.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:  Z All Limit gpio.22:low:pu]
[MSG:INFO: reset_pin gpio.35:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:Woodrat]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.2.90]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.21 Dir:NO_PIN Fwd:gpio.0 Rev:gpio.4 Freq:5000Hz Period:8191]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.2 Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant I2SO.16]
[MSG:INFO: Mist coolant I2SO.17]
[MSG:INFO: Probe gpio.39:low]
```

### User Interface Software

WebUI

### What happened?

I got tired of always having problems with g-code senders and how they try to detecting grbl version. When it fails I'm unable to use homing. Setting a fixed start message ($Start/Message=Grbl 1.1g ['$' for help]) -helps but not completely. 

What have I done, I have spend all afternoon getting familiar with the WebUI and especially the Tablet view. I find it very useful, it looks different that then many gcode senders but it has the essentials and what matters. I discovered an issue/problem with the "Upld" button described below and have question about colors.

 
**Issue:**
Upload of gcode files using the "tablet" view in WebUI is broken. Press "Upld" button -> It open's a window listing files locally, then selecting one of them to be uploaded and press "open". Window closes (no beep) and file does not get uploaded.

If I switch to "Dashboard" and upload file to SD card, file gets uploaded just fine. Then switch back to tablet view to select newly uploaded file and press start works. I consider this a workaround on short term basis, ideally the "Upld" would be able to do the job.

**Question:**
Is there a way to change the colors of the tablet view, perhaps a css file or config parameters ?

MitchBradley, 2024-09-22:
That upload bug was fixed a few weeks ago but maybe it hasn't propagated to a FluidNC release.  Try https://github.com/MitchBradley/ESP3D-WEBUI/blob/revamp/index.html.gz to see if it fixes it.

Re the colors, you can change them by rebuilding the WebUI image from the repo https://github.com/MitchBradley/ESP3D-WEBUI/tree/revamp (be sure to use the "revamp" branch), 

Another avenue might be to use developer tools to override the styles - see https://blog.kieranroberts.dev/how-to-persist-style-changes-through-reloads-using-overrides-in-dev-tools .  I don't have a lot of experience with doing that on a wholesale basis, although I often use Dev Tools to make temporary style changes during development.  There are some stylists on the FluidNC Discord server who might be able to help.

lytke, 2024-09-24:
Thanks Mitch, the link you provided did the trick. Upld button works great 👍 

I have some reading to do before making any changes.

closed 2024-09-24 by lytke