bdring/FluidNC #1331: Problem: Pull-down on GPIO.0 does not work

michue, 2024-09-25:
### Wiki Search Terms

[Configuring I/O pins](http://wiki.fluidnc.com/en/config/config_IO), [ESP32](http://wiki.fluidnc.com/en/hardware/esp32_pin_reference), [Strapping](http://wiki.fluidnc.com/en/hardware/ESP32-S3_Pin_Reference#strapping-pins)

### Controller Board

[Makerbase XY DLC32 V1.0](http://wiki.fluidnc.com/en/hardware/3rd-party/XY_DLC32_V10)

### Machine Description

[Sculpfun S9 Laser Engraver](https://www.sculpfun.com/products/sculpfun-s9-laser-engraver)

### Input Circuits

```yaml
No circuit connected yet - just measuring the voltage of GPIO.0 against ground.
```

### Configuration file

```yaml
board: Makerbase XY DLC32 V1.0
name: Sculpfun S9
meta: M. Hüpkes 2024-05-27 Rev 1
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.01
verbose_errors: true
report_inches: false
start:
  must_home: false
  deactivate_parking: true
  check_limits: false
stepping:
  engine: I2S_STATIC
  idle_ms: 25
  pulse_us: 10
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12
axes:
  shared_stepper_disable_pin: I2SO.0
  shared_stepper_reset_pin: NO_PIN
  homing_runs: 2
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 6000
    acceleration_mm_per_sec2: 1000
    max_travel_mm: 375
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 1
      seek_mm_per_min: 5000
      feed_mm_per_min: 250
      settle_ms: 250
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.36:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 6000
    acceleration_mm_per_sec2: 1000
    max_travel_mm: 395
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 1
      seek_mm_per_min: 5000
      feed_mm_per_min: 250
      settle_ms: 250
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6
  z:
    steps_per_mm: 250
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 1000
    max_travel_mm: 200
    soft_limits: true
    homing:
      cycle: -1
      allow_single_axis: false
      positive_direction: false
      mpos_mm: 1
      seek_mm_per_min: 5000
      feed_mm_per_min: 250
      settle_ms: 250
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4
coolant:
  mist_pin: gpio.13
  flood_pin: NO_PIN
  delay_ms: 0
macros:
  startup_line0: null
  startup_line1: null
  macro0: M64 P6
  macro1: M65 P6
  macro2: null
  macro3: null
  after_homing: null
  after_reset: null
  after_unlock: null
control:
  safety_door_pin: gpio.5:pd
  reset_pin: gpio.2:pd
  feed_hold_pin: gpio.0:pd
  cycle_start_pin: gpio.4:pd
  macro0_pin: gpio.19:pd
  macro1_pin: gpio.18:pd
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
user_outputs:
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: gpio.15
  digital7_pin: NO_PIN
probe:
  pin: gpio.22:low
  toolsetter_pin: NO_PIN
  check_mode_start: false
  hard_stop: false
Laser:
  pwm_hz: 5000
  output_pin: gpio.23
  enable_pin: gpio.32
  disable_with_s0: true
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: true
i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17
```

### Startup Messages

```text
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.7.17 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:XY_DLC32_V10_laser.yaml]
[MSG:INFO: Machine Sculpfun S9]
[MSG:INFO: Board Makerbase XY DLC32 V1.0]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: Stepping:I2S_static Pulse:10us Dsbl Delay:0us Dir Delay:0us Idle Delay:25ms]
[MSG:INFO: User Digital Output:6 on Pin:gpio.15]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (1.000,376.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36:low]
[MSG:INFO: Axis Y (1.000,396.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35:low]
[MSG:INFO: Axis Z (1.000,201.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34:low]
[MSG:INFO: safety_door_pin gpio.5:pd]
[MSG:INFO: reset_pin gpio.2:pd]
[MSG:INFO: feed_hold_pin gpio.0:pd]
[MSG:INFO: cycle_start_pin gpio.4:pd]
[MSG:INFO: macro0_pin gpio.19:pd]
[MSG:INFO: macro1_pin gpio.18:pd]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:gpio.32 Out:gpio.23 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Mist coolant gpio.13]
[MSG:INFO: Probe gpio.22:low]
[MSG:INFO: WiFi is disabled]

Grbl 3.7 [FluidNC v3.7.17 (wifi) '$' for help]
[MSG:ERR: feed_hold_pin is active at startup]
[MSG:INFO: Running macro startup_line0: null]
[MSG:INFO: Running macro startup_line1: null]
[MSG:INFO: ALARM: Control Pin Initially On]
ALARM:11
```

### User Interface Software

FluidNC Web Installer

### What happened?

I configured some pins as [Control Inputs](http://wiki.fluidnc.com/en/config/control) be used with external push buttons. The button push would connect its GPIO to 3.0 V (high) to activate the corresponding function.

In preparation I loaded the configuration without any buttons attached. On restart, the configuration threw the following error:
> ...
> [MSG:INFO: feed_hold_pin gpio.0:pd]
> ...
> [MSG:ERR: feed_hold_pin is active at startup]
> ...

This behavior is unexpected, as the corresponding GPIO.0 is configured with pull-down and high as the active state - therefore expecting the pin to be inactive without external connection.  
Measuring the voltage showed, that the pin is at 2.53 V yielding the high reading.

While researching the issue, I found [ESP32-S3: GPIO0 pull-down Issue](https://esp32.com/viewtopic.php?t=27916#p97524) indicating that this might be related to GPIO.0 being a strapping pin.

Sadly I'm not able to change the GPIO. Any advice or guidance is appreciated!

MitchBradley, 2024-09-25:
GPIO0 on ESP32 modules is used to control whether to start in bootloader mode - for reflashing the firmware - or run mode.  It has an external pullup resistor that is usually 10K or so, and a transistor pull-down that is controlled by a combination of the RTS and DTR lines from the USB-serial chip.  That 10K external pullup is stronger than ESP32 internal pulldown resistors, which tend to be in the 100K range.  If you must use GPIO0 as a control input, you would have to make it active low, and you could use the existing external pullup so adding :pu to the config line would be superfluous.

michue, 2024-09-25:
Hello @MitchBradley,
thank you very much for the explanation!
I will change my wiring accordingly.

closed 2024-09-25 by michue