bdring/FluidNC #1339: Problem: Homin issue on CoreXY

Kenabi31, 2024-09-28:
### Wiki Search Terms

Used these Pins http://wiki.fluidnc.com/en/hardware/3rd-party/MKS_DLC32

### Controller Board

MKS DLC 32 V2.1

### Machine Description

H-Bot System, two Nema23 Stepper Motors (X/Y), Servo Motor (Z)

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: Plottergeist
meta: (21.11.2022) by thedevleon

kinematics:
  corexy:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

start:
  must_home: false
  deactivate_parking: true
  check_limits: false
  

axes:
  shared_stepper_disable_pin: I2SO.0
  
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 5000
    max_travel_mm: 500.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 3
      feed_mm_per_min: 500
      seek_mm_per_min: 500
      settle_ms: 2
      seek_scaler: 2.100
      feed_scaler: 2.100
    motor0:
      limit_all_pin: gpio.36:high
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:high

  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 750.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 3
      feed_mm_per_min: 1000.000
      seek_mm_per_min: 1000.000
      settle_ms: 2
      seek_scaler: 2.1
      feed_scaler: 2.1
    motor0:
      limit_all_pin: gpio.35:high
      hard_limits: true
      pulloff_mm: 10
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 10
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 2000
    max_travel_mm: 10
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: true
    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.22
        min_pulse_us: 1000
        max_pulse_us: 2000

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.8.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Plottergeist]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (3.000,503.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.36]
[MSG:INFO: Axis Y (3.000,753.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.35]
[MSG:INFO: Axis Z (-10.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.22 Pulse Len(1000,2000 period:1048575)]
[MSG:INFO:     Update timer for rc_servo at 20 ms]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Connecting to STA SSID:Leokero 2.4GHz]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.178.76]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 8080]
```

### User Interface Software

WebUI

### What happened?

Hitting the Hard Limit of X Axis, the X Axis motor stops rotating as it should but the motor for the Y Axis still rotates; and the same for the Hard Limit of Y Axis, the X Axis motor keps rotating; making proper homing impossible.

Kenabi31, 2024-09-29:
solved

MitchBradley, 2024-09-29:
What was the solution, in case someone else encounters the issue?

Kenabi31, 2024-09-30:
> What was the solution, in case someone else encounters the issue?

im using h-not kinematics, I thought I had to pick corexy but it was midtbot kinematics

closed 2025-01-11 by bdring