bdring/FluidNC #1340: Problem: TMC5160 with MKS DLC32 V2.1

Kenabi31, 2024-09-30:
### Wiki Search Terms

http://wiki.fluidnc.com/en/hardware/3rd-party/MKS_DLC32

### Controller Board

MKS DLC32 V2.1

### Machine Description

H-Bot and Servo (Z)

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: Plottergeist
meta: (21.11.2022) by thedevleon

kinematics:
  midtbot:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

start:
  must_home: false
  deactivate_parking: true
  check_limits: false
  

axes:
  shared_stepper_disable_pin: I2SO.0
  
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 500.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 3
      feed_mm_per_min: 500.000
      seek_mm_per_min: 500.000
      settle_ms: 2
      seek_scaler: 2.100
      feed_scaler: 2.100
    motor0:
      limit_neg_pin: gpio.36:high
      hard_limits: true
      pulloff_mm: 5
      tmc_5160:
        cs_pin: gpio.15
        spi_index: -1
        r_sense_ohms: 0.075
        run_amps: 2.75
        hold_amps: 0.50
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.1
        direction_pin: I2SO.2:high

  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 750.000
    soft_limits: false
    homing:
      cycle: 3
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 3
      feed_mm_per_min: 500.000
      seek_mm_per_min: 500.000
      settle_ms: 2
      seek_scaler: 2.1
      feed_scaler: 2.1
    motor0:
      limit_neg_pin: gpio.35:high
      hard_limits: true
      pulloff_mm: 5
      tmc_5160:
        cs_pin: gpio.16
        spi_index: -1
        r_sense_ohms: 0.075
        run_amps: 2.75
        hold_amps: 0.50
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.5
        direction_pin: I2SO.6:high

  z:
    steps_per_mm: 10
    max_rate_mm_per_min: 6000
    acceleration_mm_per_sec2: 2000
    max_travel_mm: 10
    homing:
      cycle: 1
      mpos_mm: 1.3
      positive_direction: false
    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.22
        min_pulse_us: 1000
        max_pulse_us: 2000

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.8.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:ERR: gpio.16 - Pin is already used.]
[MSG:ERR: Configuration error at /machine/i2so: ]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:ERR: Validation error at /i2so: I2SO BCK pin should be configured once]
[MSG:INFO: Machine Plottergeist]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:NO_PIN WS:gpio.17 DATA:gpio.21]
[MSG:INFO: Not setting up I2SO: BCK pin has incorrect capabilities]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:DBG: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (3.000,503.000)]
[MSG:INFO:   Motor0]
[MSG:DBG: TMCStepper Library Ver. 0x703]
[MSG:INFO:     tmc_5160 Step:I2SO.1 Dir:I2SO.2 CS:gpio.15 Disable:NO_PIN Index:-1 R:0.075]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO: Axis Y (3.000,753.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:I2SO.5 Dir:I2SO.6 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.075]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO: Axis Z (1.300,11.300)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.22 Pulse Len(1000,2000 period:1048575)]
[MSG:INFO:     Update timer for rc_servo at 20 ms]
[MSG:ERR: X Axis TMC driver not detected - expected 0x30 got 0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x30 got 0x0]
[MSG:INFO: Kinematic system: midtbot]
[MSG:INFO: Connecting to STA SSID:Leokero 2.4GHz]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.178.76]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 8080]
```

### What happened?

I couldnt find any information on how to setup two tmc5160 drivers to a mks dlc32 v2.1.
GPT says i need to parallel connect the miso, mosi and sck of both driver to certain pins. Im not sure how to do that, i bought the controller and stepstick driver especially to not do something like that. 

can anyone provide a guide or information how to make it work? 
Thanks in advance.

MitchBradley, 2024-09-30:
Use standard_stepper: as the driver type instead of tmc5160.  The tmc5160 driver type is for SPI-configured 5160s.

MitchBradley, 2024-09-30:
I interpret your statement

> i bought the controller and stepstick driver especially to not do something like that.

to mean that the 5160 modules that you purchased are factory configured in "standalone mode", where all the configuration is done via fixed voltage levels or jumper settings on the MS pins.  My answer above depends on that assumption.

If my interpretation is incorrect, and the 5160 modules are setup at the factory for configuration using SPI, then you will have to change a jumper on the module, similar to what this says https://wiki.fysetc.com/S5160_V1.2/ , otherwise you will not be able to use them with DLC32.  The reason I say "similar" is because there are many vendors of modules like this and they all have different layouts.  Since you did not tell us exactly what you have, I have to give a handwaving answer instead of an exact one.

MitchBradley, 2024-09-30:
One more consideration: 5160s tend to draw a lot of surge current which can cause some power supplies to drop out, which then causes the 5160 chip to shut down.  If you have unexplained glitches or failures when you are using the device, that could be the cause.  The power level of 5160s is sometimes too much for the power delivery circuitry on stepstick boards.

Kenabi31, 2024-10-01:
> One more consideration: 5160s tend to draw a lot of surge current which can cause some power supplies to drop out, which then causes the 5160 chip to shut down. If you have unexplained glitches or [...]

Thanks for your replies, it sounds like the tmc5160 is a bit too complicated for me, i decided to use my tmc2209 and run my nema23 stepper motor with half its power. Luckily i have a good fan keeping my driver warm at max. even under full load.

closed 2024-10-01 by MitchBradley