bdring/FluidNC #1342: Edit yaml file with website

FMD14, 2024-10-01:
### Wiki Search Terms

Edit yaml file with website

### Controller Board

DLC MKS 32 V2.1

### Machine Description

Laser corexy Neje 30w

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: CoreXY Laser NJE 30W
meta: 02/11/2022 FMD14

kinematics:
  corexy:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 25000.000
    acceleration_mm_per_sec2: 3200.000
    max_travel_mm: 515.00
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 7000.000
      settle_ms: 525.00
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low

  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 25000.000
    acceleration_mm_per_sec2: 3200.000
    max_travel_mm: 555.00
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 7000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 1200.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0: 
  startup_line1: 
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000%  1000=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
Pas de message
```

### User Interface Software

LaserGRBL  LightBurn

### What happened?

How to do the yaml file with the web interface 
the old version we had access to the yaml file. Now when I click on the file it will download it
I would like to change the movement speed

### Other Information

[image]

breiler, 2024-10-01:
The config file editor is available at https://installer.fluidnc.com

FMD14, 2024-10-01:
I connect with the IP address 
However, I don't see the yaml file. 
my laser moves when I homing 
how to install install editor
I don't speak English

FMD14, 2024-10-02:
Good evening
I succeeded thanks to you The version has changed a lot 
It's more user-friendly 
Thanks again

FMD14, 2024-10-05:
hello, I modified values ​​in the yaml file with fluidnc web installer. I saved I did a restart. When I launched it asked me to create a config file. My question can we modify the file live. I reinstalled the version and the yaml file

yzheka, 2025-05-02:
+++ for this feature. I want to open file editor from the web interface instead of downloading the file, editing and uploading again

bdring, 2025-05-02:
@FMD14 I am not sure what happened with your file.  Maybe it was renamed or deleted.

You cannot edit a file live. Both the Web Installer and the WebUI will download the the config file, let you edit it and then upload it. You then need to restart FluidNC.

You can [tune some values live](http://wiki.fluidnc.com/en/config/overview#live-changes-tuning). This lets you test some values like speeds and accelerations. You must save those values to the file manually.

FWIW: I edit and test dozens of yaml files per day. I only use the Web Installer and FluidTerm. The WebUI workflow is too slow for me. You must wait for the wifi connection and reload the browser page. You also don't get to see the start messages right away. Editing the file on the PC and uploading also makes sure I always have a local copy of the config.

MitchBradley, 2025-05-02:
@yzheka The effective way to +++ a feature is to $$$ the developers.